espressif/arduino-esp32 #4436: spicommon_bus_initialize_io fails for ESP32-PICO-V3-02 where default VSPI pins unavailable ( GPIO 18 & 23) (IDFGH-4153)

starlino, 2020-10-23:
https://github.com/espressif/esp-idf/blob/0b71a0a46d67cce681ec55973b020d950d8596bd/components/driver/spi_common.c#L377

This fails for newer PICO chips when a non-existent pin is used (such as standard VSPI pins SCLK = 18, MISO = 19, MOSI = 23, SS = 5)

PICO does not have  GPIO  18  and 23 thus if user attempts to use VSPI (with default pins) in arduino they will get a

Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d6a7f  PS      : 0x00060530  A0      : 0x800d5484  A1      : 0x3ffb1e90  
A2      : 0x00000002  A3      : 0x3ffb1f2c  A4      : 0x00000001  A5      : 0x3f40a1bc  
A6      : 0x3ffb8354  A7      : 0x3f40a17c  A8      : 0x800d6a7f  A9      : 0x3ffb1e70  
A10     : 0x3ff49074  A11     : 0x00002a00  A12     : 0x00000a00  A13     : 0x00000001  
A14     : 0x00060320  A15     : 0x00000000  SAR     : 0x00000020  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x800d5484  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

ELF file SHA256: aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa00000000000aaaaaaaaaaaaaaaaaaaaa

Decoding 6 results
0x400d6a7f: spicommon_bus_initialize_io at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/driver/spi_common.c line 378
0x400d5481: spi_bus_initialize at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/driver/spi_master.c line 268
0x400d0ef0: SPI_Init(bool) at D:\PROJECTS\IC_V2\IC_FIRMWARE\tests\esp32_spi_idf/esp32_spi_idf.ino line 188
0x400d104d: setup() at D:\PROJECTS\IC_V2\IC_FIRMWARE\tests\esp32_spi_idf/esp32_spi_idf.ino line 203
0x400d15ca: loopTask(void*) at D:\DOCS\Arduino\hardware\espressif\esp32\cores\esp32/main.cpp line 14
0x40089401: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 355 (discriminator 1)

starlino, 2020-10-23:
I actually tried with latest IDF (without arduino) release/4.2 branch and SPI does not work with   VSPI  and remapped pins to 

#define pinSCLK_ALT  32 //for PICO
#define pinMOSI_ALT  33 //for PICO
#define pinMISO 19

my chip is  ESP32-PICO-V3-02  (8MB + PSRAM version)  is anybody currently working on this and is there an ETA to support /stabilize  SPI for these chips ?

felmue, 2020-10-24:
Hi guys

FWIW: In an Arduino environment on a M5Atom (which uses a slightly different ESP32 PICO variant, e.g. `ESP32-PICO-D4 (revision 1)`) I have SPI running successfully with the following pins:
```
#define SCK 21
#define MISO 19
#define MOSI 25
#define SLAVE_CS 22
```
Thanks
Felix

starlino, 2020-10-28:
Thanks, I ordered  ESP32-PICO-D4 dev boards and will test, I also ordered ESP32-PICO-V3 (4MB), will report back with results

closed 2021-01-11 by stale[bot]

hallard, 2021-06-02:
Hi Guys,

I'm designing a 2nd board with Pico-V3-02 any idea of what pins will be kind of 'official' for SPI with arduino (I now I can remap but best if I have the defaults one)

Thanks,