espressif/arduino-esp32 #4439: Onewire does not communicate with divices (ds18b20)

magicbug1, 2020-10-24:
### Hardware:
Board:							ESP32 NodeMCU Module WLANWiFi Development Board mit CP2102
                                                                (AZ-Delivery)
Core Installation version:			?1.0.0? ?1.0.1-rc4? ?1.0.1? ?1.0.1-git? ?1.0.2? ?1.0.3?
IDE name:						Arduino IDE
Flash Frequency:					?40Mhz?
PSRAM enabled:                                    ?no? ?yes?
Upload Speed:					?115200?
Computer OS:                                        Pi 4b

### Description:
I am trying to read the addresses of 2 ds18b20. If I read the pin directly with OneWire ow(pin); ow.search(address); no address is found. Playing around with  some code I found I started looping over some pins, including the pin I am using (5). I found that looping over pins 2 to 5 does not find the address of my ds18b20, but looping over pins 2 to 6 does, and it finds them correctly on pin 5. This demonstrates that my setup on the breadboard is correct. Searching on pin 6 causes an guru meditation error and resets. This seems necesarry to comunicate on the onewire. Note though that looping only over pin 5 and 6 does cause the reset, but the esp32 does not find the ds18b20.
Needless to say that using the reset button does not influence the behavior.
(superfluous) Note: the Guru Meditation Error is not the problem. My objective is to use pin 5 (or any other free pin for that matter). I do not need to scan pins

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

// find address of one-wire devices
#include <OneWire.h>

uint8_t findDevices(int pin)
{
  OneWire ow(pin);

  uint8_t address[8];
  uint8_t count = 0;

  Serial.print("\nEnter in findDevices - pin = ");
  Serial.println(pin);

  //ow.reset();
  //ow.reset_search();
  if (ow.search(address))
  {
    Serial.print("\nuint8_t pin");
    Serial.print(pin, DEC);
    Serial.println("[][8] = {");
    do {
      count++;
      Serial.println("  {");
      for (uint8_t i = 0; i < 8; i++)
      {
        Serial.print("0x");
        if (address[i] < 0x10) Serial.print("0");
        Serial.print(address[i], HEX);
        if (i < 7) Serial.print(", ");
      }
      Serial.println("  },");
    } while (ow.search(address));

    Serial.println("};");
    Serial.print("// nr devices found: ");
    Serial.println(count);
  } else {
    Serial.println("// No adress found");
  }

  return count;
}

void setup() {
  Serial.begin(115200);
  Serial.println("//\n// Start oneWireSearch.ino \n//");

// pin 6 causes guru meditation error and resets
// When this happens, the address is read correctly
// Works with i=2; i<7
// Does not work with i=2; i<6 or i=5; i<7
  for(int i=2;i<6;i++)  {
    findDevices(i);
  }
  Serial.println("\n//\n// End oneWireSearch.ino \n//");

}

void loop() {
  // put your main code here, to run repeatedly:
  //Serial.println("OneWire loop...");
  //delay(10000);
}

```
### Sample output
```
...
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400e9894: ffffffff ffffffff ffffffff
Core 1 register dump:
PC      : 0x400e9898  PS      : 0x00060d30  A0      : 0x80088218  A1      : 0x3ffb1fb0  
A2      : 0x3ffc0318  A3      : 0x00000000  A4      : 0x00060023  A5      : 0x3ffb8058  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x800d1dc8  A9      : 0x3ffb1f80  
A10     : 0x3ffc02dc  A11     : 0x00000007  A12     : 0x00000020  A13     : 0x80000020  
A14     : 0x00000008  A15     : 0x00000001  SAR     : 0x00000006  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

Backtrace: 0x400e9898:0x3ffb1fb0 0x40088215:0x3ffb1fd0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac
//
// Start oneWireSearch.ino 
//

Enter in findDevices - pin = 2
// No adress found

Enter in findDevices - pin = 3
// No adress found

Enter in findDevices - pin = 4
// No adress found

Enter in findDevices - pin = 5

uint8_t pin5[][8] = {
  {
0x28, 0xC0, 0xA8, 0xE1, 0x20, 0x19, 0x01, 0xDE  },
  {
0x28, 0x32, 0xE8, 0x38, 0x21, 0x19, 0x01, 0x12  },
};
// nr devices found: 2

Enter in findDevices - pin = 6
// No adress found

//
// End oneWireS
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

CelliesProjects, 2020-10-24:
The default Arduino OneWire does not work properly.
I use https://github.com/stickbreaker/OneWire and having no problems with DS18B20 sensors.

See https://github.com/CelliesProjects/FFatSensor and https://github.com/CelliesProjects/dallas-webserver

lbernstone, 2020-10-24:
Don't use pin 6-11. https://github.com/espressif/arduino-esp32#esp32dev-board-pinmap

magicbug1, 2020-10-24:
@CelliesProjects : That has solved my problem slpendidly
I have had a look at the two projects and find them very interesting. I will surely use some of it in my project.
Thanks!

closed 2020-10-24 by magicbug1