espressif/arduino-esp32 #4447: Move BLE memory to PSRAM

noukee, 2020-10-26:
Can anyone move BLE memory to PSRAM? I have an ESP32 Wrover-B and I am using PlatformIO but I am stuck getting bluetooth + Secure WiFi + Mqtt working at the same time.

These settings don't work?

CONFIG_BT_ALLOCATION_FROM_SPIRAM_FIRST=y   
CONFIG_BT_BLE_DYNAMIC_ENV_MEMORY=y

CONFIG_MBEDTLS_EXTERNAL_MEM_ALLOC 1

lbernstone, 2020-10-26:
You cannot change CONFIG_ settings within Arduino.  They are already compiled into the symbol libraries.  I would recommend using [NimBLE](https://github.com/h2zero/NimBLE-Arduino) instead.

noukee, 2020-10-26:
@lbernstone with PlatformIO, cant' change CONFIG_ settings from platformio.ini

```
build_flags =
;Debug level: 0 None, 1 Error, 2 Warn, 3 Info, 4 Debug, 5 Verbose
    -D CORE_DEBUG_LEVEL=5
    -D MQTT_MAX_PACKET_SIZE=256
    -D BOARD_HAS_PSRAM
    -mfix-esp32-psram-cache-issue

    -D CONFIG_BT_ALLOCATION_FROM_SPIRAM_FIRST=y
    -D CONFIG_BT_BLE_DYNAMIC_ENV_MEMORY=
```

Thanks!

closed 2021-01-09 by stale[bot]