espressif/arduino-esp32 #4452: ESP-IDF version clarity / update roadmap

elliotwoods, 2020-10-28:
The current ESP-IDF version used in arduino-esp32 is not clear. Tracking down commits it seems we're using commit hash
`a0468b2bd64c48d093309a4b3d623a7343c205c0` of ESP-IDF, which seems to be roughly ESP v3.1

It would be good to mention in the readme:

1. The current ESP-IDF version
2. Ehen this will be updated / why it is difficult to update to later versions
3. What users should do if they want to work with a later version of ESP-IDF 

e.g. i'd like to use the ESP-IDF implementation websockets, but it seems that is only added in later versions of ESP-IDF.

atanisoft, 2020-10-28:
@elliotwoods current arduino-esp32 1.0.x versions (which PIO picks currently) require IDF v3.x. You can build with IDF v4.0.x using the release/v4.0 branch of arduino-esp32, if PIO still allows you to use that version. There is no direct support for IDF v4.1 at this time in arduino-esp32 but there is for IDF v4.2+ on the esp32s2 branch, but again PIO likely doesn't support that version.

atanisoft, 2020-10-28:
One option you have is use only Arduino in PIO and override the package that it picks up and that will work in PIO. If you need Arduino and IDF for custom builds then you are restricted to what PIO supports.

closed 2021-01-11 by stale[bot]