espressif/arduino-esp32 #4453: Which Board to use for ESP32 DevKitC or ESP32U??

jhsa, 2020-10-28:
This might not be the right place to ask this question, but I can't find the info anywhere else. So, I do apologize in advance.
As the tittle say, I cannot find any information on which Arduino board I should select for the DevKitC board. I have one of those ESP32U boards with the external antenna connector.

Thank you for your answers.

lbernstone, 2020-10-28:
There is no need to use any board other than the "ESP32 Dev Module", ever.  The variants include pre-filled information about attached hardware.  Anything included in those [variants](https://github.com/espressif/arduino-esp32/tree/master/variants), like a builtin led, psram, LORA, etc., can be manually added just by putting the appropriate code in your sketch.
Specific to your question, there is no difference from the IDE standpoint between a on-board antenna and a IPEX connector.

jhsa, 2020-10-28:
Thanks for your answer. In that case, can I continue to use this one (DOIT ESP32 DEVKIT V1) that i already use for my other module? 
The reason I bought this one with the external antenna is that my project will be installed inside a metallic enclosure. I have noticed that the pinouts of the boards  are slightly different, hence my question.
Thank you very much again.

jhsa, 2020-10-28:
Sorry, if my questions are stupid.. I am a beginner with arduino and specially with WIFI modules

closed 2021-01-11 by stale[bot]