espressif/arduino-esp32 #4457: DNS failure when using WiFiClientSecure in idf-release/v4.0 branch

matthew798, 2020-10-29:
### Hardware:
Board:							ESP32 DoIt devkit 
Core Installation version:			idf-release/v4.0 
IDE name:						Platform.io / IDF component?
Flash Frequency:					40Mhz
Computer OS:                                        Manjaro (Arch)

### Description:
Using WiFiClientSecure to connect to anything results in a DNS failure, and subsequently an error in freertos/event_groups.c.

I tried with and without the calls to `dns_setserver`.

The line that is causing the failed `assert` in event_groups.c is:
```
	/* Check the user is not attempting to set the bits used by the kernel
	itself. */
	configASSERT( xEventGroup );
```

The CA cert being used is the one from the example file from this repo.

I tried manually patching in the changed from [this PR](https://github.com/espressif/arduino-esp32/pull/3731) but it made no difference.

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp
void WiFiManager::Init(){
    tcpip_adapter_init();
    wifiEvtGrp = xEventGroupCreate();
    ESP_ERROR_CHECK( esp_event_loop_init(ipEventHandler, NULL) );
    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_wifi_init(&cfg));
    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_STA));
    ESP_ERROR_CHECK(esp_wifi_start());
}

esp_err_t WiFiManager::ipEventHandler(void *ctx, system_event_t *event){
    
    if(event->event_id == SYSTEM_EVENT_STA_GOT_IP){
        WIFI_INFO("STA_GOT_IP");

        ip_addr_t dnsserver;
        IP_ADDR4( &dnsserver, 8,8,8,8);
        dns_setserver(0, &dnsserver);
        IP_ADDR4( &dnsserver, 8,8,4,4);
        dns_setserver(1, &dnsserver);

        WiFiClientSecure client;
        client.setCACert(caCert);

        Serial.println("\nStarting connection to server...");
        if (!client.connect("www.howsmyssl.com", 443)){
            Serial.println("Connection failed!");
            return ESP_OK;
        }
        else {
            Serial.println("Connected to server!");
            // Make a HTTP request:
            client.println("GET https://www.howsmyssl.com/a/check HTTP/1.0");
            client.println("Host: www.howsmyssl.com");
            client.println("Connection: close");
            client.println();
        }

        while (client.connected()) {
        String line = client.readStringUntil('\n');
            if (line == "\r") {
                WIFI_INFO("Headers received");
                break;
            }
        }

        while (client.available()) {
            char c = client.read();
            Serial.write(c);
        }

        client.stop();
    }

    return ESP_OK;
}
```

### Debug Messages:
```
Starting connection to server...
[E][WiFiGeneric.cpp:685] hostByName(): DNS Failed for www.howsmyssl.com
[E][WiFiClientSecure.cpp:132] connect(): start_ssl_client: -1
Connection failed!
/home/matthew/.platformio/packages/framework-espidf/components/freertos/event_groups.c:560 (xEventGroupSetBits)- assert failed!
abort() was called at PC 0x40097280 on core 1

ELF file SHA256: 54be485e51b651e7

Backtrace: 0x4009c8e9:0x3ffbb250 0x4009cc6d:0x3ffbb270 0x40097280:0x3ffbb290 0x400d8e88:0x3ffbb2b0 0x400e7bc6:0x3ffbb2d0 0x400e7ccb:0x3ffbb2f0 0x400e8a05:0x3ffbb310 0x400f006d:0x3ffbb360 0x400f38b1:0x3ffbb3a0 0x400f92e6:0x3ffbb3c0 0x400e76ed:0x3ffbb3e0 0x400e776c:0x3ffbb400 0x4009736d:0x3ffbb430
  #0  0x4009c8e9:0x3ffbb250 in invoke_abort at /home/matthew/.platformio/packages/framework-espidf/components/esp32/panic.c:155
  #1  0x4009cc6d:0x3ffbb270 in abort at /home/matthew/.platformio/packages/framework-espidf/components/esp32/panic.c:172
  #2  0x40097280:0x3ffbb290 in xEventGroupSetBits at /home/matthew/.platformio/packages/framework-espidf/components/freertos/event_groups.c:560 (discriminator 1)
  #3  0x400d8e88:0x3ffbb2b0 in wifi_dns_found_callback(char const*, ip_addr const*, void*) at /home/matthew/.platformio/packages/framework-arduinoespressif32-src-c69fc9322995db5694e2741d307d9723/libraries/WiFi/src/WiFiGeneric.cpp:660
  #4  0x400e7bc6:0x3ffbb2d0 in dns_call_found at /home/matthew/.platformio/packages/framework-espidf/components/lwip/lwip/src/core/dns.c:993
  #5  0x400e7ccb:0x3ffbb2f0 in dns_correct_response at /home/matthew/.platformio/packages/framework-espidf/components/lwip/lwip/src/core/dns.c:1191
  #6  0x400e8a05:0x3ffbb310 in dns_recv at /home/matthew/.platformio/packages/framework-espidf/components/lwip/lwip/src/core/dns.c:1338
  #7  0x400f006d:0x3ffbb360 in udp_input at /home/matthew/.platformio/packages/framework-espidf/components/lwip/lwip/src/core/udp.c:404
  #8  0x400f38b1:0x3ffbb3a0 in ip4_input at /home/matthew/.platformio/packages/framework-espidf/components/lwip/lwip/src/core/ipv4/ip4.c:778
  #9  0x400f92e6:0x3ffbb3c0 in ethernet_input at /home/matthew/.platformio/packages/framework-espidf/components/lwip/lwip/src/netif/ethernet.c:186
  #10 0x400e76ed:0x3ffbb3e0 in tcpip_thread_handle_msg at /home/matthew/.platformio/packages/framework-espidf/components/lwip/lwip/src/api/tcpip.c:180
  #11 0x400e776c:0x3ffbb400 in tcpip_thread at /home/matthew/.platformio/packages/framework-espidf/components/lwip/lwip/src/api/tcpip.c:154
  #12 0x4009736d:0x3ffbb430 in vPortTaskWrapper at /home/matthew/.platformio/packages/framework-espidf/components/freertos/port.c:143
```

lbernstone, 2020-10-29:
1) idf-release/v4.0 is not a release, and is not going to be supported.  There may be some backporting available from the platformIO group.  The esp32s2 branch is what will be the next major release.
2) I'm not sure why you think this is the fault of WFCS.  The error points to dns calls.  dns did get a bit of modification in IDF4 along with wide swaths of the network interface api.
3) We don't support 3rd party libraries here.  If you want WiFiManager to work, you will need to ask the fine folks managing that library to update it.

matthew798, 2020-10-29:
@lbernstone Sorry for not giving a specific version, but AFAIK PIO uses the idf-release/4.0 branch of this repo. 
Are you saying that there is no more development going on for `ifd-release/v4.0` at all?

I suspect it's a bug in this repo because it's caused by `WiFiClientSecure`. Or at least `WiFiClientSecure` is hitting a bug in IDF.

`WiFiManager` is my own class, the code I posted could very well all be in `app_main`.

I will try to run the example in the repo for `WiFiClientSecure` tonight. If it works, it's obviously something in my own code...

atanisoft, 2020-10-29:
@matthew798 you are making a blocking call from the event task. The event task is running at a higher priority than the dns task likely. You should do very limited API calls inside event callbacks and instead move your connection calls to loop() or similar task that is not the event task

closed 2021-01-14 by stale[bot]