espressif/arduino-esp32 #4461: WiFi reconnection fails permanently after connection attempt in low signal strength area fails

davefiddes, 2020-10-29:
### Hardware:
Board:							Wemos Lonlin32 clone and ESP32-WROOM-32D module
Core Installation version:			git master 1287c529330d0c11853b9f23ddf254e4a0bc9aaf
IDE name:						Arduino IDE 1.8.13
Flash Frequency:					80Mhz
PSRAM enabled:                                    no
Upload Speed:					921600
Computer OS:                                        Fedora Linux 32

### Description:

When using an ESP32 in an area of poor WiFi signal strength the connection is dropped as expected. When moving to a normal signal strength area the connection is not restored occasionally. I am seeing this behaviour with an AP hosted on a TP-Link TL-WDR3600 v1 using WPA2 PSK authentication. The only way to correct the situation is to restart the system which then connects immediately.

For low signal strength my AP is reporting -90dBm/-95dBm signal to noise (or out of contact completely). For normal signal strength the AP reports -32dBm/-95dBm signal to noise. Under normal signal strength the devices operate perfectly in high and low traffic operation.

It seems that there is something wrong in the WiFi connection state-machine somewhere and this behaviour is not as intended.

### Sketch: 

I am able to replicate with `libraries/WiFi/examples/WiFiClientEvents/WiFiClientEvents.ino` adding some additional logging in the main loop:

```cpp

void loop()
{
    delay(1000);

    Serial.print( "WiFi status: ");
    Serial.print( WiFi.status() );
    Serial.print( " Address: " );
    Serial.println( WiFi.localIP() );
}
```

### Debug Messages: (with Arduino debug level raised to Verbose)
```
...
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:390] _eventCallback(): Reason: 201 - NO_AP_FOUND
[WiFi-event] event: 5
Disconnected from WiFi access point
WiFi lost connection. Reason: 201
WiFi status: 1 Address: 0.0.0.0
WiFi status: 1 Address: 0.0.0.0
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:390] _eventCallback(): Reason: 4 - ASSOC_EXPIRE
WiFi status: 6 Address: 0.0.0.0
WiFi status: 6 Address: 0.0.0.0
WiFi status: 6 Address: 0.0.0.0
WiFi status: 6 Address: 0.0.0.0
WiFi status: 6 Address: 0.0.0.0
WiFi status: 6 Address: 0.0.0.0
WiFi status: 6 Address: 0.0.0.0
WiFi status: 6 Address: 0.0.0.0
WiFi status: 6 Address: 0.0.0.0
WiFi status: 6 Address: 0.0.0.0
WiFi status: 6 Address: 0.0.0.0
WiFi status: 6 Address: 0.0.0.0
WiFi status: 6 Address: 0.0.0.0
WiFi status: 6 Address: 0.0.0.0
WiFi status: 6 Address: 0.0.0.0
WiFi status: 6 Address: 0.0.0.0
WiFi status: 6 Address: 0.0.0.0
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 8 - STA_LOST_IP
[WiFi-event] event: 8
Lost IP address and IP address is reset to 0
WiFi status: 0 Address: 0.0.0.0

```
The final line repeats continuously even after moving to a high signal strength area.

The full log which includes several disconnect/reconnect events until the failure is exhibited can be found here:
[ESP32-wifi-disconnect-reconnection-failure.log](https://github.com/espressif/arduino-esp32/files/5460599/ESP32-wifi-disconnect-reconnection-failure.log)

### Hack Fix / Work around

I have found it is possible to fix the problem by extending the WiFiEvent handler to force a disconnect and then reconnect.

```cpp
        case SYSTEM_EVENT_STA_LOST_IP:
            Serial.println("Lost IP address and IP address is reset to 0");
            WiFi.disconnect();
            WiFi.reconnect();
            break;
```

davefiddes, 2020-11-01:
This is a complete version of the workaround above:
```cpp
    WiFi.onEvent( [ & ]( WiFiEvent_t event ) {
        if ( event == SYSTEM_EVENT_STA_LOST_IP )
        {
            Serial.println( "Forcing disconnect and reconnect of WiFi" );
            WiFi.disconnect();
            WiFi.reconnect();
        }
    } );
```

FrankBoesing, 2020-11-02:
It is completely useless to keep mentioning the WiFi connection problems.
For a very long time there has been no reaction to them.
I doubt that this will happen anytime soon.

davefiddes, 2021-01-02:
I have update to current git head (ef99cd7fe7778719c92d6f8df0f10d3f0f7aa35e) with the latest 3.3.3 IDF. Unfortunately I am still able to replicate this odd lockup in the WiFi reconnection logic when operating in poor signal strength areas.

closed 2021-07-08 by stale[bot]