espressif/arduino-esp32 #4462: Could not detects the BLE_iBeacon example with BLE_Beacon_Scanner Example

yuzuka4573, 2020-10-30:
### Hardware:
Board:							ESP32 Dev Module v1 ( and v4 )
Core Installation version:			master 1287c529330d0c11853b9f23ddf254e4a0bc9aaf
IDE name:						Arduino IDE
Flash Frequency:					40Mhz
PSRAM enabled:                                    no
Upload Speed:					115200
Computer OS:                                        Windows 10  
### Description:
I try to find the iBeacon with library example(BLE_iBeacon and BLE_Beacon_Scanner).  
But I couldn't detect the iBeacon made with BLE_iBeacon.  
Use nRF Connect mobile on Andriod, it detects that Beacon as iBeacon.

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
*Use "BLE_Beacon_Scanner" and "BLE_iBeacon"* ( no modifire )

### Debug Messages:  
LOG BLE_iBeacon  
```  
ets Jun  8 2016 00:22:57

rst:0x5 (DEEPSLEEP_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10864
load:0x40080400,len:6404
entry 0x400806b8
start ESP32 38
deep sleep (423s since last reset, 11s since last boot)
[I][BLEDevice.cpp:569] getAdvertising(): create advertising
[D][BLEDevice.cpp:571] getAdvertising(): get advertising
[V][BLEAdvertising.cpp:147] setAdvertisementData(): >> setAdvertisementData
[V][BLEAdvertising.cpp:155] setAdvertisementData(): << setAdvertisementData
[V][BLEUtils.cpp:1817] gapEventToString(): gapEventToString: Unknown event type 4 0x04
[V][BLEAdvertising.cpp:164] setScanResponseData(): >> setScanResponseData
[V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: Unknown event type
[E][BLEAdvertising.cpp:169] setScanResponseData(): esp_ble_gap_config_scan_rsp_data_raw: 258 Unknown ESP_ERR error
[V][BLEUtils.cpp:1264] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[V][BLEAdvertising.cpp:172] setScanResponseData(): << setScanResponseData
[D][BLEDevice.cpp:571] getAdvertising(): get advertising
[V][BLEAdvertising.cpp:181] start(): >> start: customAdvData: 1, customScanResponseData: 1
[D][BLEAdvertising.cpp:495] handleGAPEvent(): handleGAPEvent [event no: 4]
[D][BLEAdvertising.cpp:200] start(): - no services advertised
[V][BLEAdvertising.cpp:241] start(): << start
[V][BLEUtils.cpp:1817] gapEventToString(): gapEventToString: Unknown event type 6 0x06
[V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: Unknown event type
[V][BLEUtils.cpp:1264] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[D][BLEDevice.cpp:571] getAdvertising(): get advertising
[D][BLEAdvertising.cpp:495] handleGAPEvent(): handleGAPEvent [event no: 6]
[V][BLEAdvertising.cpp:251] stop(): >> stop
[V][BLEAdvertising.cpp:257] stop(): << stop
enter deep sleep
[V][BLEUtils.cpp:1817] gapEventToString(): gapEventToString: Unknown event type 17 0x11
[V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: Unknown event type
[V][BLEUtils.cpp:1264] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[D][BLEDevice.cpp:571] getAdvertising(): get advertising
[D][BLEAdvertising.cpp:495] handleGAPEvent(): handleGAPEvent [event no: 17]
[I][BLEAdvertising.cpp:511] handleGAPEvent(): STOP advertising

```
 
  
LOG BLE_Beacon_Scanner
```   
[V][BLEScan.cpp:197] start(): >> start(duration=5)
[V][FreeRTOS.cpp:189] take(): Semaphore taking: name: ScanEnd (0x3ffdd568), owner: <N/A> for start
[... 228 lines cut ...]
[V][BLEUtils.cpp:1817] gapEventToString(): gapEventToString: Unknown event type 3 0x03
[V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: Unknown event type
[V][BLEUtils.cpp:1264] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[D][BLEScan.cpp:99] handleGAPEvent(): Ignoring 1f:94:dc:cc:af:bc, already seen it.
[V][BLEUtils.cpp:1817] gapEventToString(): gapEventToString: Unknown event type 3 0x03
[V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: Unknown event type
[V][BLEUtils.cpp:1264] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[D][BLEScan.cpp:99] handleGAPEvent(): Ignoring 63:5c:94:45:88:e7, already seen it.
[V][BLEUtils.cpp:1817] gapEventToString(): gapEventToString: Unknown event type 3 0x03
[V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: Unknown event type
[V][BLEUtils.cpp:1264] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[D][BLEScan.cpp:99] handleGAPEvent(): Ignoring 1f:94:dc:cc:af:bc, already seen it.
[V][BLEUtils.cpp:1817] gapEventToString(): gapEventToString: Unknown event type 3 0x03
[V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: Unknown event type
[V][BLEUtils.cpp:1264] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[D][BLEScan.cpp:99] handleGAPEvent(): Ignoring 22:f3:6c:d0:4c:a5, already seen it.
[V][BLEUtils.cpp:1817] gapEventToString(): gapEventToString: Unknown event type 3 0x03
[V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: Unknown event type
[V][BLEUtils.cpp:1264] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[W][BLEScan.cpp:71] handleGAPEvent(): ESP_GAP_SEARCH_INQ_CMPL_EVT
[V][FreeRTOS.cpp:143] give(): Semaphore giving: name: ScanEnd (0x3ffdd568), owner: start
[V][FreeRTOS.cpp:77] wait(): << wait: Semaphore released: name: ScanEnd (0x3ffdd568), owner: <N/A>
Devices found: 5
Scan done!
```

chegewara, 2020-10-30:
`[E][BLEAdvertising.cpp:169] setScanResponseData(): esp_ble_gap_config_scan_rsp_data_raw: 258 Unknown ESP_ERR error`

yuzuka4573, 2020-10-30:
@chegewara 
Thanks for responding to the issue!
I just tried your advice, Scanner find the iBeacon examples!

chegewara, 2020-11-07:
If you know what can be improved feel free to make PR.

closed 2021-01-20 by stale[bot]