espressif/arduino-esp32 #4466: bitWrite macro does not enclose argument in ()

positron96, 2020-10-31:
Hello.

I've found a trivial bug in `bitWrite` macro. It's written as 
`#define bitWrite(value, bit, bitvalue) (bitvalue ? bitSet(value, bit) : bitClear(value, bit))` and its argument bitvalue is not enclosed in parentheses. This gives incorrect result for some arguments, for example I've triggered the bug with this code:
```
bool b=true;
bitWrite(dst, bit, b?1:0);
```

The macro is here: https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/Arduino.h#L91 and it affects all versions of ESP32 core since it was introduced.

In AVR Arduino core this macro is correct:
https://github.com/arduino/ArduinoCore-avr/blob/master/cores/arduino/Arduino.h#L115

lbernstone, 2020-10-31:
Please submit a PR.  You can do so right on that page by clicking the edit (pencil) icon.

positron96, 2020-10-31:
Huh, didn't know it was that easy.

lbernstone, 2020-10-31:
Well, you got the fork part done.  It should have had some option in there to create the pull request.  If you go to the [main repo page](https://github.com/espressif/arduino-esp32/) now, you should see a message at the very top showing that you have an edit ready for a pull request.  Click on that button and it should   take you to the PR page.  Enter `fixes #4466` in the comment box and create the pull request.

closed 2020-11-09 by me-no-dev