espressif/arduino-esp32 #4470: httpUpdateSecure doesn't work

workpage2, 2020-11-01:
I am using the standard httpUpdateSecure example. The update doesn't work. I get an error:

    HTTP_UPDATE_FAILED Error (-1): HTTP error: connection refused
I checked my https server. Here is the report:

    Starting Nmap 7.01 ( https://nmap.org ) at 2020-11-01 15:09 MSK
    Nmap scan report for myavrupdate.ru (188.126.46.147)
    Host is up (0.00079s latency).
    rDNS record for 188.126.46.147: mail.expertportal.ru
     PORT    STATE SERVICE  VERSION
     443/tcp open  ssl/http Apache httpd 2.4.18
     |_http-server-header: Apache/2.4.18 (Ubuntu)
     | ssl-enum-ciphers:
     |   TLSv1.0:
     |     ciphers:
     |       TLS_DHE_RSA_WITH_AES_128_CBC_SHA (dh 2048) - A
     |       TLS_DHE_RSA_WITH_AES_256_CBC_SHA (dh 2048) - A
     |       TLS_DHE_RSA_WITH_CAMELLIA_128_CBC_SHA (dh 2048) - A
     |       TLS_DHE_RSA_WITH_CAMELLIA_256_CBC_SHA (dh 2048) - A
     |       TLS_ECDHE_RSA_WITH_AES_128_CBC_SHA (secp256k1) - A
     |       TLS_ECDHE_RSA_WITH_AES_256_CBC_SHA (secp256k1) - A
     |       TLS_RSA_WITH_AES_128_CBC_SHA (rsa 2048) - A
     |       TLS_RSA_WITH_AES_256_CBC_SHA (rsa 2048) - A
     |       TLS_RSA_WITH_CAMELLIA_128_CBC_SHA (rsa 2048) - A
     |       TLS_RSA_WITH_CAMELLIA_256_CBC_SHA (rsa 2048) - A
     |     compressors:
     |       NULL
     |     cipher preference: client
     |   TLSv1.1:
     |     ciphers:
     |       TLS_DHE_RSA_WITH_AES_128_CBC_SHA (dh 2048) - A
     |       TLS_DHE_RSA_WITH_AES_256_CBC_SHA (dh 2048) - A
     |       TLS_DHE_RSA_WITH_CAMELLIA_128_CBC_SHA (dh 2048) - A
     |       TLS_DHE_RSA_WITH_CAMELLIA_256_CBC_SHA (dh 2048) - A
     |       TLS_ECDHE_RSA_WITH_AES_128_CBC_SHA (secp256k1) - A
     |       TLS_ECDHE_RSA_WITH_AES_256_CBC_SHA (secp256k1) - A
     |       TLS_RSA_WITH_AES_128_CBC_SHA (rsa 2048) - A
     |       TLS_RSA_WITH_AES_256_CBC_SHA (rsa 2048) - A
     |       TLS_RSA_WITH_CAMELLIA_128_CBC_SHA (rsa 2048) - A
     |       TLS_RSA_WITH_CAMELLIA_256_CBC_SHA (rsa 2048) - A
     |     compressors:
     |       NULL
     |     cipher preference: client
     |   TLSv1.2:
     |     ciphers:
     |       TLS_DHE_RSA_WITH_AES_128_CBC_SHA (dh 2048) - A
     |       TLS_DHE_RSA_WITH_AES_128_CBC_SHA256 (dh 2048) - A
     |       TLS_DHE_RSA_WITH_AES_128_GCM_SHA256 (dh 2048) - A
     |       TLS_DHE_RSA_WITH_AES_256_CBC_SHA (dh 2048) - A
     |       TLS_DHE_RSA_WITH_AES_256_CBC_SHA256 (dh 2048) - A
     |       TLS_DHE_RSA_WITH_AES_256_GCM_SHA384 (dh 2048) - A
     |       TLS_DHE_RSA_WITH_CAMELLIA_128_CBC_SHA (dh 2048) - A
     |       TLS_DHE_RSA_WITH_CAMELLIA_256_CBC_SHA (dh 2048) - A
     |       TLS_ECDHE_RSA_WITH_AES_128_CBC_SHA (secp256k1) - A
     |       TLS_ECDHE_RSA_WITH_AES_128_CBC_SHA256 (secp256k1) - A
     |       TLS_ECDHE_RSA_WITH_AES_128_GCM_SHA256 (secp256k1) - A
     |       TLS_ECDHE_RSA_WITH_AES_256_CBC_SHA (secp256k1) - A
     |       TLS_ECDHE_RSA_WITH_AES_256_CBC_SHA384 (secp256k1) - A
     |       TLS_ECDHE_RSA_WITH_AES_256_GCM_SHA384 (secp256k1) - A
     |       TLS_RSA_WITH_AES_128_CBC_SHA (rsa 2048) - A
     |       TLS_RSA_WITH_AES_128_CBC_SHA256 (rsa 2048) - A
     |       TLS_RSA_WITH_AES_128_GCM_SHA256 (rsa 2048) - A
     |       TLS_RSA_WITH_AES_256_CBC_SHA (rsa 2048) - A
     |       TLS_RSA_WITH_AES_256_CBC_SHA256 (rsa 2048) - A
     |       TLS_RSA_WITH_AES_256_GCM_SHA384 (rsa 2048) - A
     |       TLS_RSA_WITH_CAMELLIA_128_CBC_SHA (rsa 2048) - A
     |       TLS_RSA_WITH_CAMELLIA_256_CBC_SHA (rsa 2048) - A
     |     compressors:
     |       NULL
     |     cipher preference: client
     |_  least strength: A
I am using this code:

    #include <WiFi.h>
    #include <WiFiMulti.h>
    #include <HTTPClient.h>
    #include <HTTPUpdate.h>
    #include <time.h>
    WiFiMulti WiFiMulti;
    // Set time via NTP, as required for x.509 validation
    void setClock() {
      configTime(0, 0, "pool.ntp.org", "time.nist.gov");  // UTC
      Serial.print(F("Waiting for NTP time sync: "));
      time_t now = time(nullptr);
      while (now < 8 * 3600 * 2) {
        yield();
         delay(500);
        Serial.print(F("."));
        now = time(nullptr);
       }
      Serial.println(F(""));
      struct tm timeinfo;
       gmtime_r(&now, &timeinfo);
       Serial.print(F("Current time: "));
       Serial.print(asctime(&timeinfo));
     }
    /**
     * This is lets-encrypt-x3-cross-signed.pem
      */
     const char* rootCACertificate = \
     "-----BEGIN CERTIFICATE-----\n" \
     "MIIEkjCCA3qgAwIBAgIQCgFBQgAAAVOFc2oLheynCDANBgkqhkiG9w0BAQsFADA/\n" \
 "MSQwIgYDVQQKExtEaWdpdGFsIFNpZ25hdHVyZSBUcnVzdCBDby4xFzAVBgNVBAMT\n" \
 "DkRTVCBSb290IENBIFgzMB4XDTE2MDMxNzE2NDA0NloXDTIxMDMxNzE2NDA0Nlow\n" \
     "SjELMAkGA1UEBhMCVVMxFjAUBgNVBAoTDUxldCdzIEVuY3J5cHQxIzAhBgNVBAMT\n" \
     "GkxldCdzIEVuY3J5cHQgQXV0aG9yaXR5IFgzMIIBIjANBgkqhkiG9w0BAQEFAAOC\n" \
     "AQ8AMIIBCgKCAQEAnNMM8FrlLke3cl03g7NoYzDq1zUmGSXhvb418XCSL7e4S0EF\n" \
     "q6meNQhY7LEqxGiHC6PjdeTm86dicbp5gWAf15Gan/PQeGdxyGkOlZHP/uaZ6WA8\n" \
"SMx+yk13EiSdRxta67nsHjcAHJyse6cF6s5K671B5TaYucv9bTyWaN8jKkKQDIZ0\n" \
"Z8h/pZq4UmEUEz9l6YKHy9v6Dlb2honzhT+Xhq+w3Brvaw2VFn3EK6BlspkENnWA\n" \
"a6xK8xuQSXgvopZPKiAlKQTGdMDQMc2PMTiVFrqoM7hD8bEfwzB/onkxEz0tNvjj\n" \
"/PIzark5McWvxI0NHWQWM6r6hCm21AvA2H3DkwIDAQABo4IBfTCCAXkwEgYDVR0T\n" \
"AQH/BAgwBgEB/wIBADAOBgNVHQ8BAf8EBAMCAYYwfwYIKwYBBQUHAQEEczBxMDIG\n" \
"CCsGAQUFBzABhiZodHRwOi8vaXNyZy50cnVzdGlkLm9jc3AuaWRlbnRydXN0LmNv\n" \
"bTA7BggrBgEFBQcwAoYvaHR0cDovL2FwcHMuaWRlbnRydXN0LmNvbS9yb290cy9k\n" \
"c3Ryb290Y2F4My5wN2MwHwYDVR0jBBgwFoAUxKexpHsscfrb4UuQdf/EFWCFiRAw\n" \
"VAYDVR0gBE0wSzAIBgZngQwBAgEwPwYLKwYBBAGC3xMBAQEwMDAuBggrBgEFBQcC\n" \
"ARYiaHR0cDovL2Nwcy5yb290LXgxLmxldHNlbmNyeXB0Lm9yZzA8BgNVHR8ENTAz\n" \
"MDGgL6AthitodHRwOi8vY3JsLmlkZW50cnVzdC5jb20vRFNUUk9PVENBWDNDUkwu\n" \
"Y3JsMB0GA1UdDgQWBBSoSmpjBH3duubRObemRWXv86jsoTANBgkqhkiG9w0BAQsF\n" \
"AAOCAQEA3TPXEfNjWDjdGBX7CVW+dla5cEilaUcne8IkCJLxWh9KEik3JHRRHGJo\n" \
"uM2VcGfl96S8TihRzZvoroed6ti6WqEBmtzw3Wodatg+VyOeph4EYpr/1wXKtx8/\n" \
"wApIvJSwtmVi4MFU5aMqrSDE6ea73Mj2tcMyo5jMd6jmeWUHK8so/joWUoHOUgwu\n" \
"X4Po1QYz+3dszkDqMp4fklxBwXRsW10KXzPMTZ+sOPAveyxindmjkW8lGy+QsRlG\n" \
"PfZ+G6Z6h7mjem0Y+iWlkYcV4PIWL1iwBi8saCbGS5jN2p8M+X+Q7UNKEkROb3N6\n" \

    "KOqkqm57TH2H3eDJAkSnh6/DNFu0Qg==\n" \
      "-----END CERTIFICATE-----\n";
     void setup() {
      Serial.begin(115200);
       // Serial.setDebugOutput(true);
       Serial.println();
       Serial.println();
       Serial.println();
       for (uint8_t t = 4; t > 0; t--) {
         Serial.printf("[SETUP] WAIT %d...\n", t);
         Serial.flush();
         delay(1000);
       }
       WiFi.mode(WIFI_STA);
       WiFiMulti.addAP("SSID", "PASS");
     }
     void loop() {
        // wait for WiFi connection
        if ((WiFiMulti.run() == WL_CONNECTED)) {
         setClock();
         WiFiClientSecure client;
         client.setCACert(rootCACertificate);
         // Reading data over SSL may be slow, use an adequate timeout
         client.setTimeout(12000 / 3000); // timeout argument is defined in seconds for setTimeout
         t_httpUpdate_return ret = httpUpdate.update(client, "https://myavrupdate.ru/fw.bin");
         // Or:
         //t_httpUpdate_return ret = httpUpdate.update(client, "server", 443, "file.bin");
         switch (ret) {
            case HTTP_UPDATE_FAILED:
             Serial.printf("HTTP_UPDATE_FAILED Error (%d): %s\n", httpUpdate.getLastError(), httpUpdate.getLastErrorString().c_str());
             break;
            case HTTP_UPDATE_NO_UPDATES:
             Serial.println("HTTP_UPDATE_NO_UPDATES");
             break;
            case HTTP_UPDATE_OK:
             Serial.println("HTTP_UPDATE_OK");
             break;
         }
       }
     }

I've been trying to get it to work for two days now. Tell me what is the mistake?

felmue, 2020-11-01:
Hi @workpage2 

the root CA certificate included in the example code is from Let's Encrypt and therefore only works for servers which use a certificate issued by Let's Encrypt. Your server however seems to use a certificate from 'Sectigo RSA Domain Validation Secure Server CA' therefore you'll need to use their root CA certificate.

Cheers
Felix

chegewara, 2020-11-02:
https://esp32.com/viewtopic.php?f=2&t=15788&hilit=0x2700#p61358

workpage2, 2020-11-02:
> https://esp32.com/viewtopic.php?f=2&t=15788&hilit=0x2700#p61358

Yes! It worked!

The certificate chain from firefox worked!
Another question. How do I make my esp32 certificate chain perpetual? Or for 100 years? My certificate will be outdated in a year, and with it the certificate in esp32. And I won't be able to update it.

workpage2, 2020-11-02:
Success!
Next, there will be instructions on how to follow the certificate for 100 years.
All information is taken from here: https://www.opennet.ru/base/sec/ssl_cert.txt.html
1. You need to create your own certificate. To do this, in the ubuntu console:

    sudo openssl req -new -newkey rsa:1024 -nodes -keyout ca.key -x509 -days 36500 -subj /C=RU/ST=Msk/L=Msk/O=YOUDOMEN.COM/OU=Sale/CN=YOUDOMEN.COM/emailAddress=YOU@EMAIL.ru -out ca.crt

2. Place this certificate on your server and register it in the sites-available settings.
3. Run Firefox. Open you site. 
4. In the address bar, click on the lock. 
-Click "Show connection information". 
-Click "More". 
-Click "View certificate".
-Click PEM (certificate chain).
5. A file with certificates will be downloaded to your computer. ALL contents of the file must be transferred to your esp32.
Profit!

felmue, 2020-11-02:
Hello @workpage2 

glad to hear you got it working and thanks for the additional information.

Cheers
Felix

workpage2, 2020-11-03:
There are some observations and additions, without which nothing will work.
Firstly, HTTPS_OTA_Update and similar applications do not work with only one certificate. I was only able to get it to work when I put together a construction like this:
and upload in esp32

    -----BEGIN CERTIFICATE-----
    your certificate
    -----END CERTIFICATE-----
    -----BEGIN CERTIFICATE-----
    root certificate
    -----END CERTIFICATE-----
And my root certificate is not working! But a commercial certificate works, for example Sectigo. Ok.
I added my root certificate to the trusted ones on my server:

    sudo cp ./MYCRT.crt /usr/local/share/ca-certificates/
    sudo update-ca-certificates

But it didn’t work.

workpage2, 2020-11-05:
In a recent update, the developers introduced enabling / disabling domain name validation. To cancel checking a domain name, just specify:

    HttpsOTA.begin(url, server_certificate, false); 
And in esp32 it is enough to use only one self-signed certificate for any validity period.

Eiyvor, 2020-11-12:
@workpage2  There are three certificates in this pem(chain) file, which one should i use?

workpage2, 2020-11-12:
all three certificates must be used. One by one in a row. Into one variable.

Eiyvor, 2020-11-12:
can you please share an example?

workpage2, 2020-11-12:
const char* rootCACertificate = \
     "-----BEGIN CERTIFICATE-----\n" \
     "Bla-bla-bla\n" \
     "-----END CERTIFICATE-----\n" \
     "-----BEGIN CERTIFICATE-----\n" \
     "Bla-bla-bla\n" \
     "-----END CERTIFICATE-----\n" \
     "-----BEGIN CERTIFICATE-----\n" \
     "Bla-bla-bla\n" \
     "-----END CERTIFICATE-----";

Eiyvor, 2020-11-12:
[SETUP] WAIT 4...
[SETUP] WAIT 3...
[SETUP] WAIT 2...
[SETUP] WAIT 1...
Waiting for NTP time sync: .
Current time: Thu Nov 12 09:04:08 2020
HTTP_UPDATE_FAILED Error (-106): Verify Bin Header Failed
Waiting for NTP time sync: 
Current time: Thu Nov 12 09:04:17 2020
HTTP_UPDATE_FAILED Error (-106): Verify Bin Header Failed
Waiting for NTP time sync: 
Current time: Thu Nov 12 09:04:23 2020
HTTP_UPDATE_FAILED Error (6): Update error: Stream Read Timeout
Waiting for NTP time sync: 
Current time: Thu Nov 12 09:04:30 2020
HTTP_UPDATE_FAILED Error (-106): Verify Bin Header Failed
Waiting for NTP time sync: 
Current time: Thu Nov 12 09:04:36 2020
HTTP_UPDATE_FAILED Error (-106): Verify Bin Header Failed
Waiting for NTP time sync: 
Current time: Thu Nov 12 09:04:42 2020
HTTP_UPDATE_FAILED Error (-106): Verify Bin Header Failed
Waiting for NTP time sync: 
Current time: Thu Nov 12 09:04:47 2020
 I'm getting this error now.

closed 2021-01-25 by stale[bot]