espressif/arduino-esp32 #4472: BLERemoteCharacteristic writeValue fails to write std::strings containing null character '\0'

snosrap, 2020-11-01:
Unlike C-style strings, `std::string`s can contain the `null` character. For example `std::string("hello\0world")`.

However, in [BLERemoteCharacteristic.cpp](https://github.com/espressif/arduino-esp32/blob/f7fb00632e04d74a7890a77fa7dbbb8ae572e866/libraries/BLE/src/BLERemoteCharacteristic.cpp#L538), the `std::string newValue` is converted to a C-style string before being sent to [writeValue](https://github.com/espressif/arduino-esp32/blob/f7fb00632e04d74a7890a77fa7dbbb8ae572e866/libraries/BLE/src/BLERemoteCharacteristic.cpp#L561). This means that any characters after a null character in `newValue` will *not* be sent to `writeValue`. For example, `std::string("hello\0world")` will be converted into the C-style/null-terminated string `"hello\0"`.

https://github.com/espressif/arduino-esp32/blob/f7fb00632e04d74a7890a77fa7dbbb8ae572e866/libraries/BLE/src/BLERemoteCharacteristic.cpp#L538

A fix would be to use the `data()` method on `std::string` and, more importantly, the `length()` method on `std::string`. `newValue.length()` will give the total length of the string (including any null characters), however `strlen` will only return the length up to the first null character.

```c++
void BLERemoteCharacteristic::writeValue(std::string newValue, bool response) {
	writeValue((uint8_t*)newValue.data(), newValue.length(), response);
} // writeValue
```

chegewara, 2020-11-01:
https://github.com/espressif/arduino-esp32/blob/master/libraries/BLE/src/BLERemoteCharacteristic.h#L50

snosrap, 2020-11-01:
@chegewara, I'm not sure I understand your comment. Are you suggesting I not use `writeValue(std::string newValue, bool response = false)` in cases where `newValue` may contain null characters?

chegewara, 2020-11-02:
Sorry, i didnt read carefully your OP. Yes, this change is good.

commit f3902467 (2020-11-02): Don't convert to null-terminated string prior to writeValue (#4473)

fixes: #4472
closed 2020-11-02 by me-no-dev