espressif/arduino-esp32 #4476: unable to upload a program

vamsikrishnaA9, 2020-11-02:
### Hardware:
Chip							ESP32D0WDQ5 (revision 1)
Core Installation version:			?1.0.0? ?1.0.1-rc4? ?1.0.1? ?1.0.1-git? ?1.0.2? ?1.0.3?
IDE name:						Arduino IDE
Flash Frequency:			        80Mhz
PSRAM enabled:                               no
Upload Speed:					115200
Computer OS:                                   Windows 8

### Description:
i cant able to uplaod a basic blink program into esp32 chip. can anybody please guide me to solve this issue. Error showing MD5 of file does not match data in flash.

### Debug Messages:

esptool.py v2.6
Serial port COM8
Connecting........_____....._____....._____.....__
Chip is ESP32D0WDQ5 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: 98:f4:ab:63:04:9c
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 8MB
Compressed 8192 bytes to 47...

Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 4096.0 kbit/s)...
Hash of data verified.
Flash params set to 0x023f
Compressed 15856 bytes to 10276...

Writing at 0x00001000... (100 %)
Wrote 15856 bytes (10276 compressed) at 0x00001000 in 0.9 seconds (effective 139.5 kbit/s)...
File  md5: 05366bf69a52e01f1c3f632ec089ec40
Flash md5: c0ccf7855d15a3b91c4f79e238b6e043
MD5 of 0xFF is 5712c0fba5d6d98e8c49b568ff9fdc83

A fatal error occurred: MD5 of file does not match data in flash!
A fatal error occurred: MD5 of file does not match data in flash!

Serial monitor response:

this is what i am getting from esp32 through serial.

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5656
load:0x40078000,len:0
ho 12 tail 0 room 4
load:0x40078000,len:13844
entry 0x40078fc4
[0;32mI (30) boot: ESP-IDF v3.0.7 2nd stage bootloader[0m
[0;32mI (30) boot: compile time 09:04:31[0m
[0;32mI (30) boot: Enabling RNG early entropy source...[0m
[0;32mI (35) boot: SPI Speed      : 40MHz[0m
[0;32mI (39) boot: SPI Mode       : DIO[0m
[0;32mI (43) boot: SPI Flash Size : 4MB[0m
[0;32mI (47) boot: Partition Table:[0m
[0;32mI (51) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (58) boot:  0 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (65) boot:  1 otadata          OTA data         01 00 00010000 00002000[0m
[0;32mI (73) boot:  2 nvs              WiFi data        01 02 00012000 0000e000[0m
[0;32mI (80) boot:  3 at_customize     unknown          40 00 00020000 000e0000[0m
[0;32mI (88) boot:  4 ota_0            OTA app          00 10 00100000 00180000[0m
[0;32mI (95) boot:  5 ota_1            OTA app          00 11 00280000 00180000[0m
[0;32mI (103) boot: End of partition table[0m
[0;32mI (107) boot: No factory image, trying OTA 0[0m
[0;32mI (112) esp_image: segment 0: paddr=0x00100020 vaddr=0x3f400020 size=0x20614 (132628) map[0m
[0;32mI (167) esp_image: segment 1: paddr=0x0012063c vaddr=0x3ffc0000 size=0x02d7c ( 11644) load[0m
[0;32mI (172) esp_image: segment 2: paddr=0x001233c0 vaddr=0x40080000 size=0x00400 (  1024) load[0m
[0;32mI (175) esp_image: segment 3: paddr=0x001237c8 vaddr=0x40080400 size=0x0c848 ( 51272) load[0m
[0;32mI (204) esp_image: segment 4: paddr=0x00130018 vaddr=0x400d0018 size=0xdfc80 (916608) map[0m
[0;32mI (526) esp_image: segment 5: paddr=0x0020fca0 vaddr=0x4008cc48 size=0x02504 (  9476) load[0m
[0;32mI (530) esp_image: segment 6: paddr=0x002121ac vaddr=0x400c0000 size=0x00064 (   100) load[0m
[0;32mI (541) boot: Loaded app from partition at offset 0x100000[0m
[0;32mI (541) boot: Disabling RNG early entropy source...[0m
1.1.3

I (642) wifi: wifi firmware version: 703e53b
I (643) wifi: config NVS flash: enabled
I (643) wifi: config nano formating: disabled
I (652) wifi: Init dynamic tx buffer num: 32
I (653) wifi: Init data frame dynamic rx buffer num: 32
I (653) wifi: Init management frame dynamic rx buffer num: 32
I (659) wifi: wifi driver task: 3ffdeeb8, prio:23, stack:3584
I (664) wifi: Init static rx buffer num: 10
I (667) wifi: Init dynamic rx buffer num: 32
I (672) wifi: wifi power manager task: 0x3ffe369c prio: 21 stack: 2560
[0;31mE (678) phy_init: failed to validate PHY data partition[0m
[0;31mE (683) phy_init: failed to obtain PHY init data[0m
abort() was called at PC 0x40182703 on core 0

CelliesProjects, 2020-11-02:
Happens to me all the time with M5Stack boards.

I use `~/Arduino/hardware/espressif/esp32/tools/esptool.py --port /dev/ttyUSB0 write_flash_status --non-volatile 0`

I have to confess I don't know what it does but it works for me.

And found it here, 
https://forum.m5stack.com/topic/838/a-fatal-error-occurred-md5-of-file-does-not-match-data-in-flash

I also wipe the complete flash after such an event.

` ~/Arduino/hardware/espressif/esp32/tools/esptool.py --port /dev/ttyUSB0 erase_flash`

And flash the sketch after that.

CelliesProjects, 2020-11-02:
>  �[0;32mI (80) boot: 3 at_customize unknown 40 00 00020000 000e0000�[0m

Seems like the partition table is borked somehow.

closed 2021-01-16 by stale[bot]