espressif/arduino-esp32 #4477: Issue with the XMIT process of SerialHardware (TX buffering)

AldoLodos, 2020-11-02:
Hello

I have an issue with the XMIT process of SerialHardware on an ESP32 TTGO T-Display
On a 1200 bauds com I receive 10 to 15 byte  messages separated by 20 to 60ms of silence 
The skecth is simple : read bytes on port 1 and write them on port 2 as soon as as they arrive

Each time I receive the first byte of a new message, the writing of this byte is delayed of about 200 ms
Obviouly during this delay the RX continues to receive bytes that are buffered and when the transmit starts after the  200 ms delay, its a long set of bytes that are ransmitted on a row.  My receiving device doesn't like that.
It seems there is a buffering of the TX process

To verify this I made an analysis with an oscilloscope with the Pin 2 trigger signal generated by the sketch on chanel 1 and the TX port on chanel 2
During all the time  of Pin 2 HIGH , there is no data transmitted on TX. The data start to transmit as soon as the Pin 2 signel is LOW

Here is below my sketch and the result on the debug console

Any idea how to avoid this behaviour and have the byte really sent in real time after the Serial.write ?

Thanks for your help

void loop() {    

  if (MySerial2.available()) {
      x = MySerial2.read();
     
      Serial.print(StringInt(millis(),6)+"  ");     // absolute time 
      Serial.print(StringInt(millis()-m,4));        // delay since the last receive
      
      Serial.print(" RECV port2: ");;
      SERIAL_PRINTHEX(x);
      Serial.print(" -> XMIT port1 "); 
      if (((millis()-m) > 20) && ((millis()-m) < 100)) { // trigger on the message begining
              Serial.print("*Trigger*");
              digitalWrite(Sync_Pin,HIGH);   // for Oscilloscope analisys (Sync)
      }
      m = millis(); 
      
      int TXBufSize = MySerial1.availableForWrite();
      mmm = millis();
      MySerial1.write(x);     
      while (MySerial1.availableForWrite() < TXBufSize) { }
      
      digitalWrite(Sync_Pin,LOW);   // for oscilloscope analysis
      
      Serial.print("** XMIT delay:");
      Serial.print(StringInt(millis() - mmm,4));
      Serial.print(" ** TXBufSize:");
      Serial.println(TXBufSize);
  } 

}

String StringInt(int val, int siz) {
  String S = String(val);
  if (S.length() >= siz) return S;
  S  = "                        "+S;
  return S.substring(S.length()-siz);
}

372347     8 RECV port2: 1A -> XMIT port1 ** XMIT delay:   9 ** TXBufSize:127
372356     9 RECV port2: F0 -> XMIT port1 ** XMIT delay:   8 ** TXBufSize:127
372364     8 RECV port2: 11 -> XMIT port1 ** XMIT delay:   8 ** TXBufSize:127
372372     8 RECV port2: 21 -> XMIT port1 ** XMIT delay:   9 ** TXBufSize:127
372381     9 RECV port2: 32 -> XMIT port1 ** XMIT delay:   8 ** TXBufSize:127
372389     8 RECV port2: 11 -> XMIT port1 ** XMIT delay:   8 ** TXBufSize:127
372397     8 RECV port2: 20 -> XMIT port1 ** XMIT delay:   9 ** TXBufSize:127
372406     9 RECV port2: 11 -> XMIT port1 ** XMIT delay:   8 ** TXBufSize:127
372414     8 RECV port2: 11 -> XMIT port1 ** XMIT delay:   8 ** TXBufSize:127
372426    12 RECV port2: 16 -> XMIT port1 ** XMIT delay:   4 ** TXBufSize:127
372431     4 RECV port2: 0B -> XMIT port1 ** XMIT delay:   8 ** TXBufSize:127
372439     8 RECV port2: 01 -> XMIT port1 ** XMIT delay:   8 ** TXBufSize:127
372447     8 RECV port2: 22 -> XMIT port1 ** XMIT delay:   9 ** TXBufSize:127
372489    42 RECV port2: 11 -> XMIT port1 *Trigger*** XMIT delay: 192 ** TXBufSize:127   <<<<<<< Message begining
372682   193 RECV port2: 25 -> XMIT port1 ** XMIT delay:   4 ** TXBufSize:127
372686     4 RECV port2: 36 -> XMIT port1 ** XMIT delay:   8 ** TXBufSize:127
372694     8 RECV port2: 11 -> XMIT port1 ** XMIT delay:   8 ** TXBufSize:127
372702     8 RECV port2: 20 -> XMIT port1 ** XMIT delay:   9 ** TXBufSize:127
372711     9 RECV port2: 11 -> XMIT port1 ** XMIT delay:   8 ** TXBufSize:127
372719     8 RECV port2: 31 -> XMIT port1 ** XMIT delay:   8 ** TXBufSize:127
372727     8 RECV port2: 16 -> XMIT port1 ** XMIT delay:   9 ** TXBufSize:127
372736     9 RECV port2: 1F -> XMIT port1 ** XMIT delay:   8 ** TXBufSize:127
372744     8 RECV port2: 00 -> XMIT port1 ** XMIT delay:   8 ** TXBufSize:127
372752     8 RECV port2: BE -> XMIT port1 ** XMIT delay:   9 ** TXBufSize:127
372761     9 RECV port2: F3 -> XMIT port1 ** XMIT delay:   8 ** TXBufSize:127
372769     8 RECV port2: 11 -> XMIT port1 ** XMIT delay:   8 ** TXBufSize:127
372777     8 RECV port2: 24 -> XMIT port1 ** XMIT delay:   9 ** TXBufSize:127
372786     9 RECV port2: 35 -> XMIT port1 ** XMIT delay:   8 ** TXBufSize:127
372794     8 RECV port2: 11 -> XMIT port1 ** XMIT delay:   8 ** TXBufSize:127
372802     8 RECV port2: 20 -> XMIT port1 ** XMIT delay:   9 ** TXBufSize:127
372832    30 RECV port2: 11 -> XMIT port1 *Trigger*** XMIT delay: 204 ** TXBufSize:127  <<<<<<< Message begining
373037   205 RECV port2: 08 -> XMIT port1 ** XMIT delay:   4 ** TXBufSize:127
373041     4 RECV port2: 16 -> XMIT port1 ** XMIT delay:   8 ** TXBufSize:127
373049     8 RECV port2: 1A -> XMIT port1 ** XMIT delay:   8 ** TXBufSize:127
373057     8 RECV port2: F0 -> XMIT port1 ** XMIT delay:   9 ** TXBufSize:127
373066     9 RECV port2: 11 -> XMIT port1 ** XMIT delay:   8 ** TXBufSize:127
373074     8 RECV port2: 21 -> XMIT port1 ** XMIT delay:   8 ** TXBufSize:127
373082     8 RECV port2: 32 -> XMIT port1 ** XMIT delay:   9 ** TXBufSize:127
373091     9 RECV port2: 11 -> XMIT port1 ** XMIT delay:   8 ** TXBufSize:127
373099     8 RECV port2: 20 -> XMIT port1 ** XMIT delay:   8 ** TXBufSize:127

hreintke, 2020-11-02:
I expect this is solved by PR's #3664 and #3713 
Both are already merged in git master.

AldoLodos, 2020-11-03:
I managed to install the git master and it actually solved my problem
I don't know if I have to close this issue ?
Thanks for your help

closed 2020-11-03 by AldoLodos