espressif/arduino-esp32 #4478: Problem with key length in preferences

Schwenkner, 2020-11-02:
I have figured out that there must be some kind of maximum key length for preferences.
Unfortunately I did not read it somewhere.
Can this be confirmed?

Here is my test code:

```
#include <Preferences.h>

class Settings
{
private:
	Preferences preferences;
public:
	String Get(const String key, String defaultValue)
	{
		preferences.begin("tests", true);
		String res = preferences.getString(key.c_str(), defaultValue);
		preferences.end();
		return res;
	}

	void Set(const String key, String value)
	{
		preferences.begin("tests", false);
		preferences.putString(key.c_str(), value);
		preferences.end();
	}
};

Settings settings;

// the setup function runs once when you press reset or power the board
void setup() {
	Serial.println();
	Serial.println("Started");

	String t1("A_12345678901234");
	String t2("B_1234567890123");

	settings.Set(t1, "12345");
	String value = settings.Get(t1, "default");
	Serial.printf("expected is '12345', and was '%s'\r\n", value.c_str());

	settings.Set(t2, "12345");
	value = settings.Get(t2, "default");
	Serial.printf("expected is '12345', and was '%s'\r\n", value.c_str());
}

// the loop function runs over and over again until power down or reset
void loop() {
  
}

```
And this is the serial output:
```
Started
expected is '12345', and was 'default'
expected is '12345', and was '12345'
```

With the longer key (16 chars), the default value is got on reading, with the 15chars-key I've got the formerly written value.

chegewara, 2020-11-02:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/nvs_flash.html#namespaces

lbernstone, 2020-11-03:
Note that the return value from all the putXXX functions is a size_t value that indicates the bytes allocated in nvs.  If you are going to make wrapper functions, you should check the result.

closed 2021-01-17 by stale[bot]