espressif/arduino-esp32 #4479: Number to pin mapping per board

tannewt, 2020-11-02:
In SAMD Arduino it is possible to map an arduino pin number to a specific pin on a per-board basis? Is the same possible with the ESP32 arduino version? I only see extra variables like `D1 = 7;` in variant headers.

I'd like to see the number to pin mapping within the API so that we can use standard pin numbering between arduino implementations.

tannewt, 2020-11-02:
For example, here is the variant file for the QTPy board: https://github.com/adafruit/ArduinoCore-samd/blob/master/variants/qtpy_m0/variant.cpp#L25

The pin number is the index into this array and allows the number to remain the same across boards with different routing.

It's also possible to have empty entries for unsupported pin numbers: https://github.com/adafruit/ArduinoCore-samd/blob/master/variants/zero_radio/variant.cpp#L193

atanisoft, 2020-11-02:
The pin number (0-39) is common to all esp32 regardless of what "names" are given in the variant files. The variants are very inconsistent as each maker has defined them however they decided and not following any set standards.

UnexpectedMaker, 2020-11-03:
@atanisoft I don't think the ESP32-Sx CPU pin numbering to GPIO  is consistent to the previous ESP32's.  With the Sx series, pins were re-mapped to make IO run in sequence to PIN numbers. That is not how previous gen ESP32s were pin matched.

UnexpectedMaker, 2020-11-03:
But regardless, what @tannewt is looking for is the ability to make a cpu pin that is an IO, be whatever IO number he want's. So instead of GPIO5 being physically mapped to cpu pin 10, he wants to make it virtually mapped to cpu pin 13 etc.

atanisoft, 2020-11-03:
@UnexpectedMaker that is true, there is no direct mapping of the hardware from esp32 to esp32s2 etc.

If we look at the SAMD example, it is defining a large array containing each pin that exists and what peripheral it is tied to. The esp32 (all variants) allow remapping of the peripherals to almost any pin, I'm not certain that a 40+ entry array is needed on the esp32.

UnexpectedMaker, 2020-11-03:
> @UnexpectedMaker that is true, there is no direct mapping of the hardware from esp32 to esp32s2 etc.
>
> If we look at the SAMD example, it is defining a large array containing each pin that exists and [...]

The main bit of that array that is important here is the order it's created defines the GPIO number it's matched to. I don't think @tannewt is really after the "peripheral creation part" but more the "GPIO number ordering part" so any pre-existing hardware can have their GPIO numbers virtually changed to any position on the header pins.

atanisoft, 2020-11-03:
Hmm, I'm not certain that it can be done genetically in a variant file. It's a bit different than AVR where A0 is always analog, D1 is always pin 1, etc..

tannewt, 2020-11-03:
Yup, my main concern is flexibility in mapping the number given in the Arduino API calls to any GPIO on the ESP. It is coincidence that the SAMD arduino code also uses it for pin capabilities. This mapping ability will also help for boards being updated from ESP32 to ESP32-S2 where the pin functionality may change.

me-no-dev, 2020-11-06:
given that some pins have specific functions, how will that work if then those pins have different numbers across the boards? FYI this topic has crossed my mind a few times before.

Another thing... why not use the IO numbering that matches the chip? Uno and Leonardo do not match. Given the amount of boards and layouts out there, this needs to be well thought of. Much code exists already that uses actual GPIO numbers

tannewt, 2020-11-10:
> given that some pins have specific functions, how will that work if then those pins have different numbers across the boards? FYI this topic has crossed my mind a few times before.

It's the board designer's job to ensure that a specifically designated pin (like the SCL and SDA positions) can actually do SCL or SDA accordingly. This is easy to do with the ESP32-S2's gpio mux. So, remapping numbers will make it easier to route boards while maintaining code.

> Another thing... why not use the IO numbering that matches the chip? Uno and Leonardo do not match. Given the amount of boards and layouts out there, this needs to be well thought of. Much code exists already that uses actual GPIO numbers

Mapping based on the chip's number assumes that your code is only running on one chip. Allowing the number to be chosen on a per-board basis rather than per-chip basis means that code can run across chips on boards made in the same form factor.

I think allowing a number to number mapping on a per board basis is backwards compatible. The default mapping can be the chip numbers. Only new boards and their example code will use it to match other existing boards.

tannewt, 2020-12-22:
@me-no-dev Can we do this please? The longer this exists, the more boards will be designed with silkscreen that doesn't match other boards of the same form factor.

lbernstone, 2020-12-22:
How does anything that happens in this project prevent random manufacturers from printing whatever they want on their boards?  Idiot proofing only ensures more creative idiots.

tannewt, 2021-02-23:
> How does anything that happens in this project prevent random manufacturers from printing whatever they want on their boards? Idiot proofing only ensures more creative idiots.

It's important to add the mapping to this project so that this code can be easily used by boards that number their pins differently. Not having that functionality makes it impossible to design multiple boards with the same form factor that run ESP chips.

closed 2021-07-09 by stale[bot]