espressif/arduino-esp32 #4490: ESP32 CANbus receiving task reads transmitted messages

ColinKr, 2020-11-04:
Hi all,

### Description:
I'm working on a CANbus project using the ESP32 Wroom board in combination with an sn65HVD230, to be able of transmitting/receiving messages.

While transmitting data, which properly arrives at a Kvaser CAN to USB interface, the receiving task also stores/reads this message.

I expect the CAN receive task to remain unused since no CANbus data is provided from the Kvaser interface to the ESP32
For some reason, this is not the case. Please see attached code:

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>
#include <driver/can.h>
#include <driver/gpio.h>
#include <string.h>
#include <stdio.h>
#include <Ticker.h>

// Global objects
Ticker canBusTimer;

// Global variables
uint8_t canBusFLAG = 0;

// Prototype functions
void CAN_begin(void);
void CAN_receive(void);
void CAN_transmit_stdData(uint32_t id, uint8_t len);
void can_flag(void);

void setup() {
  CAN_begin();
  delay(1000);
  canBusTimer.attach_ms(1000, can_flag);
}

void loop() {
  CAN_receive();
  delay(5000);
  CAN_transmit_stdData(30, 8);
}

// FUNCTIONS
void can_flag(void)
{
  canBusFLAG = 1;
}

void CAN_begin(void)
{
  //Initialize configuration structures using macro initializers (changed GPIO type in esp idf library)
  can_general_config_t g_config = CAN_GENERAL_CONFIG_DEFAULT((gpio_num_t)GPIO_NUM_4, (gpio_num_t)GPIO_NUM_5, CAN_MODE_NO_ACK);
  can_timing_config_t t_config = CAN_TIMING_CONFIG_500KBITS();
  can_filter_config_t f_config = CAN_FILTER_CONFIG_ACCEPT_ALL();

  //Install CAN driver
  if (can_driver_install(&g_config, &t_config, &f_config) == ESP_OK)
    printf("Driver installed\n");
  else
  {
    printf("Failed to install driver\n");
    return;
  }

  //Start CAN driver
  if (can_start() == ESP_OK)
    printf("Driver started\n");
  else
  {
    printf("Failed to start driver\n");
    return;
  }
}

void CAN_receive(void)
{
  can_status_info_t can_status;
  can_message_t message_rx;

  //Check for received messages
  can_get_status_info(&can_status);
  if (can_status.msgs_to_rx >= 1)
  {
    printf("RX Queue: %d\n", can_status.msgs_to_rx);
    if (can_receive(&message_rx, pdMS_TO_TICKS(10)) == ESP_OK)
      printf("Message received\n");
    else
    {
      printf("Failed to receive message\n");
      return;
    }
  } else {
    return;
  }
}

void CAN_transmit_stdData(uint32_t id, uint8_t len)
{
  //Configure message to transmit
  can_message_t message_tx;
  message_tx.flags = CAN_MSG_FLAG_SELF;
  message_tx.identifier = id;
  message_tx.data_length_code = len;
  message_tx.data[7] = 1;
  /*for (uint8_t i = 0; i < len; i++){
    message.data[i] = data[i];
  }*/

  can_status_info_t info;
  can_get_status_info(&info);

  if (info.state == CAN_STATE_RUNNING)
    printf("CAN running\n");
  else if (info.state == CAN_STATE_STOPPED)
    printf("CAN stopped\n");
  else if (info.state == CAN_STATE_RECOVERING)
    printf("CAN recovering\n");
  else if (info.state == CAN_STATE_BUS_OFF)
    printf("CAN bus-off\n");

  ESP_ERROR_CHECK_WITHOUT_ABORT(can_transmit(&message_tx, pdMS_TO_TICKS(100)));
}
```

### Debug Messages:
```
CAN running
RX Queue: 1
Message received
```

igrr, 2020-11-05:
I think you need to disable CAN_MSG_FLAG_SELF, according to the docs it enables the behavior you describe: https://docs.espressif.com/projects/esp-idf/en/stable/api-reference/peripherals/can.html#message-flags.

By the way, since you aren't setting all files of the structure, i suggest zero-initializing it first. Otherwise it will contain garbage values from the stack.
```
can_message_t message_tx = {};
```

ColinKr, 2020-11-05:
Indeed, the flag setting caused the described behavior. I found out about it this morning but did not share this here as of yet.

Thank you for the tip regarding the initialization!

closed 2020-11-05 by ColinKr