espressif/arduino-esp32 #4494: HID Telephony User Page Not Working

ajar-99, 2020-11-05:
Hi,

I wanted to get caller ID, of an incoming call. I found a USAGE in TEEPHONY user page 0x30 in usb hid document. So i made an entire new report like this

USAGE_PAGE(1), 0x0B, // USAGE_PAGE (Telephony)
USAGE(1), 0x01, // USAGE (Phone)
COLLECTION(1), 0x01, // COLLECTION (Application)
REPORT_ID(1), PHONE_KEYS_ID, // REPORT_ID (3)
USAGE_PAGE(1), 0x0B, // USAGE_PAGE (Telephony)
LOGICAL_MINIMUM(1), 0x01, // LOGICAL_MINIMUM (0)
LOGICAL_MAXIMUM(1), 0x07, // LOGICAL_MAXIMUM (1)
REPORT_SIZE(1), 0x03, // REPORT_SIZE (1)
REPORT_COUNT(1), 0x01, // REPORT_COUNT (16)
USAGE(1), 0x30, // USAGE (Caller ID)
USAGE(1), 0x26, // USAGE (Drop)
USAGE(1), 0x2B, // USAGE (Speaker Phone)
USAGE(1), 0x2F, // USAGE (Phone Mute)
USAGE(1), 0x20, // USAGE (Hook Switch)
HIDINPUT(1), 0x06, // INPUT (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
END_COLLECTION(0)

But this didn't work, so i found some other ways to get caller ID, like using ANCS service to get all apple notifications, but my issue is with getting caller ID in android devices. Can anyone tell me if there is a way to get notifications from android devices. I heard that there is a service called Alert Notification Service, Did any one implement ANS on ESP32. (or) made HID telephony user page worked. Thanks in advance.

sansillusion, 2020-11-05:
I can only guess there is a way since windows 10 "my phone" app can. I dont
know if there is a need for an app installed on the phone with proper
permission for it to work. Windows requies me to install an app but it may
be for other features, I am not a real programmer and know next to nothing
about Bluetooth.

ajar-99, 2020-11-06:
Hi,
Thanks for the reply. I don't want windows to read my notifications, i want **ESP32's BLE** to read my **android phones** notifications. and so that i can later display them on some other display unit(which i will do later). But for now i would like to know if there is a way to read android phones **CALLER ID** with ESP32 without installing any mobile application on android side. I found that there is a **service with ID: 0x1811 (Alert Notification Service)** with which we can implement this it seems. I want to know that if it works and if anyone implement it.

closed 2021-01-20 by stale[bot]