espressif/arduino-esp32 #4495: Connection to WiFI Access Point happen once of two in a network with two AP

pberna67, 2020-11-05:
Dear Developer,

I've encountered Wifi connection problems using the library network with a specific SSID. In some home network can happen that two or more AP with the same SSID are in place. There are different scenarios: the two access points (used to extend the communication range) operate in the same frequency band and are connected together with an Ethernet cable (scenario 1), but can happen, more frequently, that they operate in the same frequency band (scenario 2)

Scenario1:
It is important that the library selects the best AP (the one with the strongest signal). I've noticed, however that using the https://github.com/espressif/arduino-esp32/archive/master.zip ES32 library and WiFiClient example, it doesn't happen, and to connect to two Access point with the same SSID, the below example works once out of two (if the Access point operate in two different frequency band (100% reproducibility), After the first hw reset of the NINA-W102 EVK (ESP 32) the board connects the AP, after the second no, after the third reset, yes and so on (two access point which operate in two different frequency band)

Scenario 2:
If the two Access point operate in the same frequency band, , sometime it is not be able to connect to the Wifi network.
See nelow, for the same example of the Scenario 1, the test report

I think the problem is in the native Wifi.begin method, but I'm not sure.
If there is only one access point in the network no connectivity problem is detected.

Setup
https://github.com/espressif/arduino-esp32/archive/master.zip ES32 HAL
WiFiClient example

Hw
NINA-W101 EVK, but also ESP6832 board with related HAL

Same situation is seen in the ESP8266 hw

/*
 *  This sketch sends data via HTTP GET requests to data.sparkfun.com service.
 *
 *  You need to get streamId and privateKey at data.sparkfun.com and paste them
 *  below. Or just customize this script to talk to other HTTP servers.
 *
 */

#include <WiFi.h>

const char* ssid     = "Vodafone-27586605";
const char* password = "My password"                                                                        ";

const char* host = "data.sparkfun.com";
const char* streamId   = "....................";
const char* privateKey = "....................";

void setup()
{
    Serial.begin(115200);
    delay(10);

    // We start by connecting to a WiFi network

    Serial.println();
    Serial.println();
    Serial.print("Connecting to ");
    Serial.println(ssid);

    WiFi.begin(ssid, password);

    while (WiFi.status() != WL_CONNECTED) {
        delay(500);
        Serial.print(".");
    }

    Serial.println("");
    Serial.println("WiFi connected");
    Serial.println("IP address: ");
    Serial.println(WiFi.localIP());
}

int value = 0;

void loop()
{
    delay(5000);
[... 67 lines cut ...]
21:24:57.326 -> Requesting URL: /input/....................?private_key=....................&value=1
21:24:57.526 -> HTTP/1.1 410 Gone
21:24:57.526 -> Server: awselb/2.0
21:24:57.526 -> Date: Thu, 05 Nov 2020 20:24:57 GMT
21:24:57.526 -> Content-Type: text/html; charset=utf-8
21:24:57.526 -> Content-Length: 700
21:24:57.526 -> Connection: close
21:24:58.526 -> 
21:24:58.526 -> <html> <head></head> <body> <h3>Phant is No Longer in Operation</h3> <p>Unfortunately Phant, our data-streaming service, is no longer in service. The system has reached capacity and, like a less-adventurous Cassini, has plunged conclusively into a fiery and permanent retirement. There are several other maker-friendly, data-streaming services and/or IoT platforms available as alternatives. The three we recommend are Blynk, ThingSpeak, and Cayenne. You can read our <a href="https://www.sparkfun.com/news/2413">blog post on the topic</a> for an overview and helpful links for each platform. The code in this tutorial will need to be adjusted to work with the other data streams.</p> </body> </html>
21:24:58.606 -> closing connection
21:24:59.996 -> ets Jun  8 2016 00:22:57
21:24:59.996 -> 
21:24:59.996 -> rst:0x1 (POWERON_RESET),boot:0x37 (SPI_FAST_FLASH_BOOT)
21:24:59.996 -> configsip: 0, SPIWP:0xee
21:24:59.996 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
21:24:59.996 -> mode:DIO, clock div:2
21:24:59.996 -> load:0x3fff0018,len:4
21:24:59.996 -> load:0x3fff001c,len:1044
21:25:00.036 -> load:0x40078000,len:10044
21:25:00.036 -> load:0x40080400,len:5872
21:25:00.036 -> entry 0x400806ac
21:25:00.316 -> ⸮
21:25:00.316 -> 
21:25:00.316 -> Connecting to Vodafone-27586605
21:25:00.956 -> ...................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................

vito45h, 2020-11-09:
I have had a similar issue with a large Mesh setup, the default scan method when you call Wifi.begin is to use WIFI_FAST_SCAN as seen in this Enum
`
typedef enum {
    WIFI_FAST_SCAN = 0,                   /**< Do fast scan, scan will end after find SSID match AP */
    WIFI_ALL_CHANNEL_SCAN,                /**< All channel scan, scan will end after scan all the channel */
}wifi_scan_method_t;
`

my work around was to add the following line in the WiFi.begin
`conf.sta.scan_method = WIFI_ALL_CHANNEL_SCAN;`

[image]

I imagine a long term solution would be to add a boolean uint8_t scanMode to the WiFi.begin, this would allow the program to select which scan mode is used at run time.

pberna67, 2020-12-30:
It seems the problem is solved from ESP 1.04 version

closed 2020-12-30 by pberna67