espressif/arduino-esp32 #4499: DHCP Renew using 0.0.0.0 as destination incorrectly

vito45h, 2020-11-06:
I am seeing that my ESP32 DHCP renewal is being sent to 0.0.0.0, below is a wireshark.
My DHCP server is Windows Server 2019.

[image]
 
In the wireshark I see the renewal starting at 30 minutes (1800 seconds, time shown is duration between packets), but it is sent to 0.0.0.0 and therefore isn't making it to the DHCP server, then the retries happen until finally there is a request sent to 255.255.255.255, which gets an immediate ACK and renewal timer goes another 30 minutes and process repeats.

In the DHCP ACK I see the DHCP Server Identifier Option 54 is set correctly to 192.168.15.254.

I am using VS code and PlatformIO with an platformio.ini as follows:
[env:wifi]
platform = espressif32@2.0.0
;platform = espressif32@1.9.0
board = esp32dev
framework = arduino
lib_deps =
  ArduinoJson
  Time

Is there a way to review the dhcp.c code file that is being used here, and perhaps point me in the direction of a guide to recompile it(I'm thinking it's deep in the LWIP library), when I get it using the server IP?

lbernstone, 2020-11-06:
This should be pretty close. https://github.com/espressif/esp-lwip/blob/2.1.2-esp/src/core/ipv4/dhcp.c
I'm not sure if pio works exactly the same, but in arduino, project files are preferred over the compiled libraries, so if you put that code and the matching header in your project, it should use them instead.

vito45h, 2020-11-06:
yes this dhcp.c file at first pass the code looks good
I see it's checking for the option to be available and if it is it reads it in and says it is remembering the offered server IP

from the dhcp.c
`
  if (dhcp_option_given(dhcp, DHCP_OPTION_IDX_SERVER_ID)) 
`
and
`
    ip_addr_set_ip4_u32(&dhcp->server_ip_addr, lwip_htonl(dhcp_get_option_value(dhcp, DHCP_OPTION_IDX_SERVER_ID)));
`
Then backtracking to verify that it's reading in the right option 
`      case (DHCP_OPTION_SERVER_ID):
        LWIP_ERROR("len == 4", len == 4, return ERR_VAL;);
        decode_idx = DHCP_OPTION_IDX_SERVER_ID;`
which shows option 54 and is correct
`#define DHCP_OPTION_SERVER_ID       54`

So this is all correct, when an offer is received, ah light bulb moment, now i see.

When the ESP32 boots it's starting off with a request for it's previous address, and the logic to remember the dhcp server IP is only available in an offer response.
Fortunately I did see a work around from another thread to do a discover on bootup:

https://github.com/espressif/arduino-esp32/issues/2537#issuecomment-508558849

preforming 
`WiFi.config(INADDR_NONE, INADDR_NONE, INADDR_NONE);`
before
`WiFi.begin();`

For me I can see a couple options, but maybe these are out of the scope for this repo:

1. Make ESP32 always reset the dhcp stack on power up so it preforms the discover
2. Store DHCP server ip where last IP is stored so the DHCP module can be initialized fully
3. Update esp-lwip to set server on DHCP ack if the server_ip_addr is set to 0.0.0.0
4. Just use the work around

lbernstone, 2020-11-06:
Why don't you try posting your issue in that repo?  The authors of the code will have a much better understanding of their flow than anybody here.

vito45h, 2020-11-17:
I'll go make an issue over there, but through using your quick guide at https://github.com/espressif/arduino-esp32/blob/master/docs/lib_builder.md

I was able to disable 
menuconfig ->
Component config ->
LWIP->
Disable DHCP: Restore last IP obtained from DHCP server

By disabling this the device will always start with a discover and learn the dhcp server ip
And subsequent renews should go to the right ip address.
Thanks

closed 2020-11-17 by vito45h