espressif/arduino-esp32 #4501: ESP32 BLE - non static notifyCallback?

libasoles, 2020-11-06:
Can I pass a member function of an object as a `notifyCallback`?

Issue is: I have state and utils in my BLE client object, but [notifyCallback ](https://github.com/espressif/arduino-esp32/blob/master/libraries/BLE/examples/BLE_client/BLE_client.ino)is a static function and I see no possible way to pass it muy utils and state. Any clue?

```cpp
static void notifyCallback(
  BLERemoteCharacteristic* pBLERemoteCharacteristic,
  uint8_t* pData,
  size_t length,
  bool isNotify) {
    
    // al this is useful for me if I cannot broadcast it using my utils.
    // can I make notifyCallback part of an object?
    Serial.print("Notify callback for characteristic ");
    Serial.print(pBLERemoteCharacteristic->getUUID().toString().c_str());
    Serial.print(" of data length ");
    Serial.println(length);
    Serial.print("data: ");
    Serial.println((char*)pData);
}
```

chegewara, 2020-11-06:
No, you cant, but maybe there is a way to do what you need. Try to elaborate what exactly you need.

libasoles, 2020-11-06:
Ok, thanks.

I'm not pulling data, I just want to get notified of the data. And when I receive a notification, I want to react to it using some leds.

Now, my software is object oriented. I have a **BluetoothService** that creates a connection to a BLE service. And that object knows how to notify the rest of the object about events. So, for me, it would be great to receive BLE notifications inside **BluetoothService** and broadcast them from there.

See, I don't have global variables, singletons, and nothing like that. So i'm wondering how would the static function **notifyCallback** fit into my design. Because it should be able to broadcast what it receives.

chegewara, 2020-11-06:
Is there any reason you dont have global `BluetoothService`?
In callback notification you could call function on that object and pass arguments you want if any.

libasoles, 2020-11-06:
Reasons was just a design choice. Using objects.
```

#pragma once

#include <Arduino.h>
#include <iostream>
#include "Ring.h"
#include "AnimationStateHandler.h"
#include "SerialPortCommunicationHandler.h"
#include "BluetoothService.h"

class App {
public:
    void setup() {
        delay(3000); // power-up safety delay

        animationsHandler.reactTo(Event::BOOT);

        subscribeToBluetooth();
        subscribeToSerialPort();
    }

    void subscribeToBluetooth() {
        bluetooth.init();
        bluetooth.subscribe(&animationsHandler);
    }

    void subscribeToSerialPort() {
        keyboard.subscribe(&animationsHandler);
    }

    void loop() {
        ring.draw(animationsHandler.currentAnimation());
        keyboard.checkInput();
        bluetooth.keepConnectionAlive();
    }

private:
    Ring ring{RING_BRIGHTNESS};
    AnimationStateHandler animationsHandler{ring};
    SerialPortCommunicationHandler keyboard{SERIAL_PORT};
    Bluetooth bluetooth{}; <-- HERE
};

```

But if the only way to communicate the notificationCallback with the rest of the system is by using global vars, I guess I will.

Thanks for the feedback.

chegewara, 2020-11-07:
I have to say i never saw code that wraps arduino setup() and loop() into class object, but i am guessing you have something like that in your code:
```
App app;
void setup()
{
   app.setup();
}

void loop()
{
   app.loop();
}
```
Similarly you can have notification callback (pseudocode):
```
void notifCallback()
{
   app.callback();
}
```

closed 2021-01-20 by stale[bot]

DiscoMan18, 2023-05-16:
> How about this? Don't know if it works at runtime, but does compile.

Won't work. It will overflow the stack.