espressif/arduino-esp32 #4502: Program not appearing to be flashed even though flash was succesful

jack-davidson, 2020-11-06:
Hello, I have encountered a problem where my program compiles and flashes but the program does not seem to run because my LED is not turning on. I am using a MELIFE esp32 from amazon. I am also using esptool.py to flash *.bin files to the esp32 for execution. My flash freq. is the default 40Mhz and write speed is the default speed for esptool.py which is 115200 baud. The core I am using is the latest stable core. I am using the arduino-cli for development. My operating system is GNU + Linux kernel 5.9.

command in my Makefile for compiling my sketch:
```bash
arduino-cli compile --fqbn esp32:esp32:esp32 esp32-example # esp32-example is the name of my sketch
```
command in my Makefile for flashing the compiled binary to the esp32:
```bash
esptool.py write_flash 0x1000 build/esp32.esp32.esp32/esp32-example.ino.bin --flash-mode qio
```

and then here is my sketch:
```cpp
void setup()
{
    pinMode(2, OUTPUT);
}

void loop()
{
    digitalWrite(2, HIGH);
}
```
this is very simple but `D2` does not turn on and does not provide `3.3v` output to power my led. The led *does* turn on when connected to `3.3v` and `GND` as it should but when I move `3.3v` to `D2`, it does not work even though I have programmed `D2` to be set to `HIGH`. I am getting no error messages but the light does not turn on, this seems like the answer could be very simple and I am not seeing it. I am new to ESP32 development but have developed for arduino platform in the past.

I will put the output of the compilation with arduino-cli and the flashing with esptool.py below:

output of `arduino-cli compile --fqbn esp32:esp32:esp32 esp32-example # esp32-example is the name of my sketch`:
```
arduino-cli compile --fqbn esp32:esp32:esp32 esp32-example
Sketch uses 207547 bytes (15%) of program storage space. Maximum is 1310720 bytes.
Global variables use 15220 bytes (4%) of dynamic memory, leaving 312460 bytes for local variables. Maximum is 327680 bytes.
```

output of `esptool.py write_flash 0x1000 build/esp32.esp32.esp32/esp32-example.ino.bin --flash-mode qio`:
```
esptool.py write_flash 0x1000 build/esp32.esp32.esp32/esp32-example.ino.bin --flash_mode qio
esptool.py v2.8
Found 2 serial ports
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP32
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 10:52:1c:67:ac:94
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Flash params set to 0x002f
Compressed 207664 bytes to 105309...
Wrote 207664 bytes (105309 compressed) at 0x00001000 in 9.3 seconds (effective 178.6 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

Thank you so much!

chegewara, 2020-11-07:
I dont know how to use arduinocli, but did you use OTA update before with that board?
Maybe board flash to 0x10000, but is running code from second ota partition?

Another question is, did you really try to flash to 0x1000 or it is just typo?

lbernstone, 2020-11-07:
Set verbose output in Arduino IDE to see the correct format for a full upload.  You will need to know what your partition table looks like if you want to upload just a new firmware app.
https://github.com/espressif/esptool#missing-bootloader

jack-davidson, 2020-11-07:
> I dont know how to use arduinocli, but did you use OTA update before with that board?
> Maybe board flash to 0x10000, but is running code from second ota partition?
> 
> Another question is, did you really try to flash to 0x1000 or it is just typo?

I have been flashing to 0x1000, is that where the boot loader is located? I have not OTA updated the board and I don't know how to do that, I will look it up.

jack-davidson, 2020-11-07:
I looked at the docs and found the boot loader is located at 0x1000. I now need to figure out how to re-flash the boot loader right?

lbernstone, 2020-11-07:
Yes, there is even an app that can do all this for you! https://www.arduino.cc/en/software

jack-davidson, 2020-11-07:
> Yes, there is even an app that can do all this for you! https://www.arduino.cc/en/software

The arduino-ide includes utilities to flash the bootloader? That would be great. Maybe I should switch to the arduino-ide as it has more support and that is the whole point of using arduino platform. Does what I just said sound sane?

lbernstone, 2020-11-07:
This repo includes bootloaders.  The Arduino IDE knows how to use them.  If you turn on verbose logging you can see exactly what it is doing when it uploads.  Walk before you run.

jack-davidson, 2020-11-07:
I will update you once I flash a bootloader and get arduino-ide set up

jack-davidson, 2020-11-09:
I haven't gotten a chance to set it up but I will later today. If I can 1: set up arduino-ide to use the esp32 core, and 2: flash the bootloaders to the esp32, and 3: get a blink sketch compiled and flashed to the esp32 I will close this issue.

- [ ] set up arduino-ide to use esp32 core
- [ ] flash bootloader to esp32
- [ ] compile and flash a test 'blink' sketch to the esp32

jack-davidson, 2020-11-16:
UPDATE: I have since switched to esp-idf. Esp-idf is so much better than trying to use arduino and I got some networking stuff up on the esp32. I love esp-idf and everything works great.

closed 2020-11-16 by jack-davidson