espressif/arduino-esp32 #4504: wifigeneric

mehrdad987, 2020-11-07:
is it normal to reset wifi by rebooting my esp\

15:52:42.658 -> [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
15:52:42.658 -> [W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 200 - BEACON_TIMEOUT
15:52:42.658 -> Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
15:52:42.692 -> Core 1 register dump:
15:52:42.692 -> PC      : 0x40180444  PS      : 0x00060034  A0      : 0x8008145c  A1      : 0x3ffbe790  
15:52:42.692 -> A2      : 0x3ffbec34  A3      : 0x00000000  A4      : 0x3ffc34f4  A5      : 0x00000000  
15:52:42.692 -> A6      : 0x3ffc34f0  A7      : 0xffffffff  A8      : 0x80081020  A9      : 0x3ffbe760  
15:52:42.692 -> A10     : 0x3ffbee60  A11     : 0x3ffba264  A12     : 0x00000001  A13     : 0x00000001  
15:52:42.727 -> A14     : 0x00060021  A15     : 0x00000000  SAR     : 0x0000001d  EXCCAUSE: 0x00000007  
15:52:42.727 -> EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
15:52:42.727 -> Core 1 was running in ISR context:
15:52:42.727 -> EPC1    : 0x400879ac  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40180444
15:52:42.727 -> 
15:52:42.727 -> Backtrace: 0x40180444:0x3ffbe790 0x40081459:0x3ffbe7b0 0x400856f1:0x3ffbe7d0 0x400879a9:0x3ffba190 0x40083aef:0x3ffba1b0 0x40089afd:0x3ffba1d0
15:52:42.762 -> 
15:52:42.762 -> Rebooting...

lbernstone, 2020-11-07:
https://github.com/espressif/arduino-esp32/blob/master/docs/ISSUE_TEMPLATE.md

mehrdad987, 2020-12-08:
hey
anyone knows about the cause of this error I appreciate if it can help me

closed 2021-06-23 by mehrdad987