espressif/arduino-esp32 #4505: I2C - max 800kHz

FrankBoesing, 2020-11-07:
I2C can be set to 800kHz max.
Ist there a workaround?
The ESp32 supports more, I think

felmue, 2020-11-08:
Hi @FrankBoesing 

there is a check for min / max I2C frequency in place in function [i2cSetFrequency](https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-i2c.c). I admit I don't fully understand the boundaries check, but it seems to be dependent on APB clock (which is about 80MHz for ESP32) and a minimal ratio between APB and I2C clock of 100, hence the 800kHz maximum.

That said for testing purposes I've commented out the boundaries check and that allowed me to increase the I2C clock. In my test setup (M5Stack device) I2C clocks higher than 1 MHz didn't work reliably and started to produce errors. I guess a reason being too weak pull-up resistors and / or I2C devices not supporting higher clocks.

Thanks
Felix

lbernstone, 2020-11-08:
1 MHz is the stated max.  https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/i2c.html#_CPPv4N12i2c_config_t9clk_speedE
The divider math for I2C was put in a long time ago.  There may be older hardware revs with lower limits.

felmue, 2020-11-09:
Hi @lbernstone 

thank you for the link and the helpful information. I appreciate it.

Cheers
Felix

closed 2020-12-10 by FrankBoesing