espressif/arduino-esp32 #4508: ESP32 IP address

zhouypau, 2020-11-09:
Why does My ESP32 Dev Module, when run Serial.println(WiFi.localIP()), the IP address will change between two address ?
(1) Correct subnet IP address:      192.168.0.22
(2) different Subnet address:        192.168.1.2 
I do have WiFi extender in my workshop which is ESP32 board connect to.

lbernstone, 2020-11-09:
https://en.wikipedia.org/wiki/Dynamic_Host_Configuration_Protocol

vito45h, 2020-11-09:
The extender is using the other ip range, compared to your main network.

To see which AP you are connected to also print Serial.println(WiFi.BSSIDstr()); 

Keep in mind this is the MAC address of the Wireless side of your router, some home routers make this hard to find, as they only show the WAN Mac address.

Once you know the BSSID, you can force the device to connect to your main router or the extender, by using the BSSID parameter in the WiFi.begin()

zhouypau, 2020-11-10:
Thank you ! the WiFi.BSSIDstr() is great, I got MAC address which ESP32 connected to.
for some reasons, I now only got 192.168.0.22 which is my main router sub-lan address.
I will keep eye, the MAC address if it goes 192.168.1.1 and then do the further tracing, thanks again, now I can continue 
my work. by the way, the wifi extender has same SSID as main router,

closed 2021-01-24 by stale[bot]