espressif/arduino-esp32 #4509: Unable to writeValue or readValue on BLE characteristics

whc2001, 2020-11-09:
### Hardware:
Board:							ESP32 DevKit V1
Core Installation version:			1.0.4
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                     no
Upload Speed:					921600
Computer OS:                                        Windows 10

### Description:
I tried to use the BLEClient to communicate with my camera. The camera requires pairing before doing anything, so I tried to use nRFConnect app on my Android phone. When not paired, you can discover all the services and characteristics, but if you try to access (read/write) any characteristic, the pairing prompt will appear on the camera screen and you would need to accept it to successfully read/write the value. I can discover the services and characteristics using BLEClient, but absolutely nothing happens when I tried to read or write the characteristics. There is no pairing prompt on camera screen, reading always return empty string and writting just nothing at all.

### Sketch:  (removed the searching and doing other actions part)
```cpp
...
/* BLE device found handler */
class BleAdvertisedDeviceHandler: public BLEAdvertisedDeviceCallbacks
{
        void onResult(BLEAdvertisedDevice advertisedDevice)
        {
            Serial.print("BLE Advertised Device found: ");
            Serial.println(advertisedDevice.toString().c_str());

            if(advertisedDevice.getAddress().equals(cameraMacAddress))
            {
                bleScanner->stop();
                cameraDevice = new BLEAdvertisedDevice(advertisedDevice);
                connectState = FOUND_BLE_DEVICE;
            }
        }
};

bool Connect()
{
    cameraConnection = BLEDevice::createClient();
    cameraConnection->setClientCallbacks(new BleConnectionHandler());
    cameraConnection->connect(cameraDevice);
    Serial.println("Connected");

    // Get location service
    BLERemoteService* locationService = cameraConnection->getService(locationServiceUuid);
    if (locationService == nullptr)
    {
        Serial.println("ERROR: No specified service");
        cameraConnection->disconnect();
        return false;
    }
    Serial.println("Location service found");

    // Get location read parameter characteristic
    locationCharaRead = locationService->getCharacteristic(locationCharaReadUuid);
    if (locationCharaRead == nullptr)
    {
        Serial.println("ERROR: No specified characteristic");
        cameraConnection->disconnect();
        return false;
    }
    Serial.println("Location read parameter characteristic found");
    Serial.print("Handle=0x");
    Serial.print(locationCharaRead->getHandle(), HEX);
    Serial.print(" CanRead=");
    Serial.print(locationCharaRead->canRead());
    Serial.print(" CanWrite=");
    Serial.println(locationCharaRead->canWrite());

    // Get location write coordinate characteristic
    locationCharaWrite = locationService->getCharacteristic(locationCharaWriteUuid);
    if (locationCharaWrite == nullptr)
    {
        Serial.println("ERROR: No specified characteristic");
        cameraConnection->disconnect();
        return false;
    }
    Serial.println("Location write coordinate characteristic found");
    Serial.print("Handle=0x");
    Serial.print(locationCharaWrite->getHandle(), HEX);
    Serial.print(" CanRead=");
    Serial.print(locationCharaWrite->canRead());
    Serial.print(" CanWrite=");
    Serial.println(locationCharaWrite->canWrite());

    // Try to read the characteristic
    Serial.print(locationCharaRead->readValue().c_str());

    return true;
}
...
```

### Debug Messages:
```
[V][BLEUtils.cpp:951] gattClientEventTypeToString(): Unknown GATT Client event type: 18
[V][BLEUtils.cpp:1283] dumpGattClientEvent(): GATT Event: Unknown
[V][BLERemoteService.cpp:162] retrieveCharacteristics(): >> getCharacteristics() for service: 8000dd00-dd00-ffff-ffff-ffffffffffff
[V][BLEUtils.cpp:951] gattClientEventTypeToString(): Unknown GATT Client event type: 18
[D][BLERemoteService.cpp:193] retrieveCharacteristics(): Found a characteristic: Handle: 42, UUID: 0000dd01-0000-1000-8000-00805f9b34fb
[D][BLEClient.cpp:158] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[V][BLERemoteCharacteristic.cpp:36] BLERemoteCharacteristic(): >> BLERemoteCharacteristic: handle: 42 0x42, uuid: 0000dd01-0000-1000-8000-00805f9b34fb
[V][BLERemoteCharacteristic.cpp:250] retrieveDescriptors(): >> retrieveDescriptors() for characteristic: 0000dd01-0000-1000-8000-00805f9b34fb
[D][BLERemoteCharacteristic.cpp:280] retrieveDescriptors(): Found a descriptor: Handle: 43, UUID: 00002902-0000-1000-8000-00805f9b34fb
[V][BLERemoteCharacteristic.cpp:294] retrieveDescriptors(): << retrieveDescriptors(): Found 1 descriptors.
[V][BLERemoteCharacteristic.cpp:45] BLERemoteCharacteristic(): << BLERemoteCharacteristic
[D][BLERemoteService.cpp:193] retrieveCharacteristics(): Found a characteristic: Handle: 45, UUID: 0000dd11-0000-1000-8000-00805f9b34fb
[V][BLERemoteCharacteristic.cpp:36] BLERemoteCharacteristic(): >> BLERemoteCharacteristic: handle: 45 0x45, uuid: 0000dd11-0000-1000-8000-00805f9b34fb
[V][BLERemoteCharacteristic.cpp:250] retrieveDescriptors(): >> retrieveDescriptors() for characteristic: 0000dd11-0000-1000-8000-00805f9b34fb
[E][BLERemoteCharacteristic.cpp:274] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[V][BLERemoteCharacteristic.cpp:294] retrieveDescriptors(): << retrieveDescriptors(): Found 0 descriptors.
[V][BLERemoteCharacteristic.cpp:45] BLERemoteCharacteristic(): << BLERemoteCharacteristic
[D][BLERemoteService.cpp:193] retrieveCharacteristics(): Found a characteristic: Handle: 47, UUID: 0000dd21-0000-1000-8000-00805f9b34fb
[V][BLERemoteCharacteristic.cpp:36] BLERemoteCharacteristic(): >> BLERemoteCharacteristic: handle: 47 0x47, uuid: 0000dd21-0000-1000-8000-00805f9b34fb
[V][BLERemoteCharacteristic.cpp:250] retrieveDescriptors(): >> retrieveDescriptors() for characteristic: 0000dd21-0000-1000-8000-00805f9b34fb
[E][BLERemoteCharacteristic.cpp:274] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[V][BLERemoteCharacteristic.cpp:294] retrieveDescriptors(): << retrieveDescriptors(): Found 0 descriptors.
[V][BLERemoteCharacteristic.cpp:45] BLERemoteCharacteristic(): << BLERemoteCharacteristic
[V][BLERemoteService.cpp:209] retrieveCharacteristics(): << getCharacteristics()
Location read parameter characteristic found
Handle=0x2F CanRead=1 CanWrite=0
Location write coordinate characteristic found
Handle=0x2D CanRead=0 CanWrite=1
[V][BLERemoteCharacteristic.cpp:398] readValue(): >> readValue(): uuid: 0000dd21-0000-1000-8000-00805f9b34fb, handle: 47 0x2f
[D][FreeRTOS.cpp:189] take(): Semaphore taking: name: ReadCharEvt (0x3ffdfab0), owner: <N/A> for readValue
[D][FreeRTOS.cpp:198] take(): Semaphore taken:  name: ReadCharEvt (0x3ffdfab0), owner: readValue
[V][FreeRTOS.cpp:63] wait(): >> wait: Semaphore waiting: name: ReadCharEvt (0x3ffdfab0), owner: readValue for readValue
[V][BLEUtils.cpp:951] gattClientEventTypeToString(): Unknown GATT Client event type: 3
[D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[V][BLEUtils.cpp:951] gattClientEventTypeToString(): Unknown GATT Client event type: 3
[V][BLEUtils.cpp:1283] dumpGattClientEvent(): GATT Event: Unknown
[V][BLEUtils.cpp:951] gattClientEventTypeToString(): Unknown GATT Client event type: 3
[D][BLEClient.cpp:158] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[V][FreeRTOS.cpp:143] give(): Semaphore giving: name: ReadCharEvt (0x3ffdfab0), owner: readValue
[V][FreeRTOS.cpp:77] wait(): << wait: Semaphore released: name: ReadCharEvt (0x3ffdfab0), owner: <N/A>
[V][BLERemoteCharacteristic.cpp:426] readValue(): << readValue(): length: 0

```

chegewara, 2020-11-10:
You have to check examples with security.

whc2001, 2020-11-11:
Thanks, after checking the example sketch I added these codes before connecting to device, and implemented the callback:

    BLEDevice::setEncryptionLevel(ESP_BLE_SEC_ENCRYPT);
    BLEDevice::setSecurityCallbacks(new BleSecurityHandler());
    BLESecurity *bleSecurity = new BLESecurity();
    bleSecurity->setAuthenticationMode(ESP_LE_AUTH_REQ_SC_BOND);
    bleSecurity->setCapability(ESP_IO_CAP_NONE);
    bleSecurity->setRespEncryptionKey(ESP_BLE_ENC_KEY_MASK | ESP_BLE_ID_KEY_MASK);

Now I will got the pairing prompt on my camera on the first read/write characterstic attempt, which is what I want. Now I just read the characterstic once after connect, and if it successed directly then no need to pair, otherwise do the reading repeatedly in a loop, prompt user to accept pairing and check a flag that is set inside callback indicating pairing accepted. All working as intended just like the mobile phone application. Now the only question is, since it seems like after accepting pairing, the pairing process will not be triggered again even if I power cycle the ESP32 (which is preferable considering how things should work). But I still wanted a way to forcefully remove the pairing (i.e. next time I connect, it will be refused by the camera and pairing prompt pop again), just like clicking "forget this pairing" on the mobile phone?

chegewara, 2020-11-12:
You can use esp-idf code to delete paired device.

whc2001, 2020-11-12:
Thanks!

    #include "esp_gap_bt_api.h"

    esp_ble_remove_bond_device((uint8_t*)cameraMacAddress.getNative())

closed 2020-11-12 by whc2001