espressif/arduino-esp32 #4510: Example for using classic bluetooth for searching for devices

redplane, 2020-11-09:
**1. What you are trying to do**

Currently, I'm trying to implement bluetooth scanner function for my esp32 device.
I have been searching on internet for tutorial but mostly, they are about:
- Make connection **FROM** app to ESP
- Using ESP Bluetooth Low energy.

However, I have many devices which only support Bluetooth legacy (Bluetooth classic)
I would like to use ESP32 for **scanning for bluetooth classic devices** and make a connection to them

**2. Describe your system (Hardware, computer, O/S, core version, environment)**
- Windows 10
- x64
- IDE Clion

**3. Describe what is failing**
There is no tutorial about using Bluetooth classic for scanning for devices.
I have found [this example about classic bluetooth discovery](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/classic_bt/bt_discovery).
From my view, I'm just a beginner to embedded and just started from arduino framework.
I don't know how to convert this code to Arduino code structure, therefore, not so sure whether that example meets my expectation or not.

**4. Show the shortest possible code that will duplicate the error **
- N/A

Can anyone provide an example about using **Classic Bluetooth (Bluetooth legacy)** for scanning devices please ?

Thank you,

redplane, 2021-01-13:
I think Bluetooth classic is still being used by many developers.
This issue shouldn't be stale.

ArcticSnowSky, 2021-02-02:
Raising my hand here too!

Wanted to connect my esp32 to a Variometer which would give me NMEA data through BT Classic SPP then

ArcticSnowSky, 2021-02-15:
Heya,
I was searching for the same functionality and didn't find anything.
If you're still in need, check #4811 out

aleferrero98, 2021-04-09:
I'm just like you redplane! I'd like to know if you found how to do that because I need to work with Classic Bluetooth in my thesis. All information will be welcome. Greetings

redplane, 2021-04-24:
@ArcticSnowSky , sorry for my late reply.
Have you tried the solution you found ? I have not tried it yet.

lbernstone, 2021-04-24:
His solution is included as part of v2.0.0, which is currently available as an alpha release.

closed 2021-07-11 by stale[bot]

weinerschizel, 2022-01-14:
I have same question. I need to connect ESP32 to a HFP headset or A2DP speaker. There's examples on how to setup the protocol and do this, but they don't show how to find and connect to the headset. Go figure! So I have same question as @redplane redplane