espressif/arduino-esp32 #4511: [ESP32 (S2?)] cannot set pin direction when pin > 32

ladyada, 2020-11-09:
hihi i made hardware with an indicator LED on pin 42 (TMS) and am not using JTAG debugging on this board. I'd like to blink the LED but cannot set the output mux with `pinMode` - instead i use a helper function i found on the forums. 

----------
```
int led = LED_BUILTIN;

// the setup routine runs once when you press reset:
void setup() {
  // initialize the digital pin as an output.
  aPinMode(led, OUTPUT); // CANNOT use pinMode here, will not actually set mux
}

// the loop routine runs over and over again forever:
void loop() {
  digitalWrite(led, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);               // wait for a second
  digitalWrite(led, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);               // wait for a second
}

void aPinMode(int pinNum, int pinDir) {
  if (pinNum >= 32) {
    uint64_t gpioBitMask = 1ULL << pinNum;
    gpio_mode_t gpioMode = (pinDir == OUTPUT) ? GPIO_MODE_OUTPUT : GPIO_MODE_INPUT;
    gpio_config_t io_conf;
    io_conf.intr_type = GPIO_INTR_DISABLE;
    io_conf.mode = gpioMode;
    io_conf.pin_bit_mask = gpioBitMask;
    io_conf.pull_down_en = GPIO_PULLDOWN_DISABLE;
    io_conf.pull_up_en = GPIO_PULLUP_DISABLE;
    gpio_config(&io_conf);
  } else pinMode(pinNum, pinDir);
}
```

lbernstone, 2020-11-09:
There has been a commit related to this.  Please make sure you have the latest code downloaded.  I just tested and a simple pinMode works for me on pin 42.

ladyada, 2020-11-09:
oh yeah rad!!!

closed 2020-11-09 by ladyada