espressif/arduino-esp32 #4516: mini D1 ESP32 need to reboot 2 times to get a wifi connection

Slayess, 2020-11-10:
### Hardware:
Board:							mini DI ESP32
Core Installation version:			?1.0.0? ?1.0.1-rc4? ?1.0.1? ?1.0.1-git? ?1.0.2? ?1.0.3?
IDE name:						Arduino IDE 1.8.13
Flash Frequency:					80Mhz
PSRAM enabled:                                     no 
Upload Speed:					921600
Computer OS:                                        Windows 10

### Description:
I must restart 2 times the ESP to get a wifi connection, on my test there is only a simple wifi connection. I tried to connect the ESP to 2 differents AP (internet provider box and mobil sharing)
First boot >> Ok ... ESP has an IP address
Next boot >> NOK ... no connection to the AP
Next boot >> Ok ... ESP has an IP address
....

### Sketch
```#include <WiFi.h>

const char* ssid     = "Slayes";
const char* password = "xxxxxxx";

void setup() {
  esp_log_level_set("*", ESP_LOG_VERBOSE);
  Serial.begin(115200);
  Serial.print("Connecting to "); Serial.println(ssid);
  WiFi.disconnect(true,true);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("WiFi connected.");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
}

void loop() {
}
```

### Debug Messages:
```
//////////////////////////FIRST AP >> WIFI CONNECTION OK ///////////////////////////////
19:26:33.863 -> Connecting to Slayess
19:26:33.863 -> I (35) wifi: wifi driver task: 3ffb5784, prio:23, stack:3584, core=0
19:26:33.910 -> I (104) wifi: wifi firmware version: 7997e4b
19:26:33.910 -> I (105) wifi: config NVS flash: enabled
19:26:33.910 -> I (105) wifi: config nano formating: disabled
19:26:33.910 -> I (105) wifi: Init dynamic tx buffer num: 32
19:26:33.910 -> I (109) wifi: Init data frame dynamic rx buffer num: 32
19:26:33.910 -> I (114) wifi: Init management frame dynamic rx buffer num: 32
19:26:33.910 -> I (119) wifi: Init management short buffer num: 32
19:26:33.910 -> I (124) wifi: Init static rx buffer size: 1600
19:26:33.910 -> I (129) wifi: Init static rx buffer num: 16
19:26:33.910 -> I (132) wifi: Init dynamic rx buffer num: 32
19:26:33.957 -> I (187) wifi: mode : sta (fc:f5:c4:01:a4:40)
19:26:33.957 -> [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
19:26:33.957 -> [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
19:26:34.098 -> I (313) wifi: n:4 1, o:1 0, ap:255 255, sta:4 1, prof:4
19:26:34.098 -> I (313) wifi: state: init -> auth (b0)
19:26:34.098 -> I (315) wifi: state: auth -> assoc (0)
19:26:34.098 -> I (325) wifi: state: assoc -> run (10)
19:26:34.429 -> .I (742) wifi: connected with Slayess, channel 4, 40U
19:26:34.524 -> I (786) wifi: pm start, type: 1
19:26:34.524 -> 
19:26:34.572 -> [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 4 - STA_CONNECTED
19:26:34.572 -> [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 7 - STA_GOT_IP
19:26:34.572 -> [D][WiFiGeneric.cpp:381] _eventCallback(): STA IP: 192.168.0.16, MASK: 255.255.255.0, GW: 192.168.0.254
19:26:34.950 -> .
19:26:34.950 -> WiFi connected.
19:26:34.950 -> IP address: 
19:26:34.950 -> 192.168.0.16

//////////////////////////FIRST AP + REBOOT >> WIFI CONNECTION NOK ///////////////////////////////
19:25:26.863 -> Connecting to Slayess
19:25:26.910 -> I (35) wifi: wifi driver task: 3ffb5784, prio:23, stack:3584, core=0
19:25:26.958 -> I (104) wifi: wifi firmware version: 7997e4b
19:25:26.958 -> I (104) wifi: config NVS flash: enabled
19:25:26.958 -> I (105) wifi: config nano formating: disabled
19:25:26.958 -> I (105) wifi: Init dynamic tx buffer num: 32
19:25:26.958 -> I (109) wifi: Init data frame dynamic rx buffer num: 32
19:25:26.958 -> I (114) wifi: Init management frame dynamic rx buffer num: 32
19:25:26.958 -> I (119) wifi: Init management short buffer num: 32
19:25:26.958 -> I (124) wifi: Init static rx buffer size: 1600
19:25:26.958 -> I (128) wifi: Init static rx buffer num: 16
19:25:26.958 -> I (132) wifi: Init dynamic rx buffer num: 32
19:25:27.005 -> I (187) wifi: mode : sta (fc:f5:c4:01:a4:40)
19:25:27.005 -> [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
19:25:27.005 -> [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
19:25:27.100 -> I (313) wifi: n:4 1, o:1 0, ap:255 255, sta:4 1, prof:4
19:25:27.100 -> I (313) wifi: state: init -> auth (b0)
19:25:27.100 -> I (317) wifi: state: auth -> assoc (0)
19:25:27.147 -> I (326) wifi: state: assoc -> run (10)
19:25:27.480 -> ............I (6630) wifi: state: run -> init (fc0)
19:25:33.473 -> I (6630) wifi: n:4 0, o:4 1, ap:255 255, sta:4 1, prof:4
19:25:33.473 -> I (6630) wifi: n:4 0, o:4 0, ap:255 255, sta:4 1, prof:4
19:25:33.473 -> [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
19:25:33.473 -> [W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 15 - 4WAY_HANDSHAKE_TIMEOUT
19:25:33.473 -> ..................................................................

//////////////////////////SECOND AP >> WIFI CONNECTION OK ///////////////////////////////
I (35) wifi: wifi driver task: 3ffb5784, prio:23, stack:3584, core=0
I (104) wifi: wifi firmware version: 7997e4b
I (105) wifi: config NVS flash: enabled
I (105) wifi: config nano formating: disabled
I (105) wifi: Init dynamic tx buffer num: 32
I (109) wifi: Init data frame dynamic rx buffer num: 32
I (114) wifi: Init management frame dynamic rx buffer num: 32
I (119) wifi: Init management short buffer num: 32
I (124) wifi: Init static rx buffer size: 1600
I (129) wifi: Init static rx buffer num: 16
I (132) wifi: Init dynamic rx buffer num: 32
I (187) wifi: mode : sta (fc:f5:c4:01:a4:40)
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
..I (1530) wifi: n:11 0, o:1 0, ap:255 255, sta:11 0, prof:4
..I (2303) wifi: state: init -> auth (b0)
I (2306) wifi: state: auth -> assoc (0)
I (2316) wifi: state: assoc -> run (10)
I (2345) wifi: connected with Slayes, channel 11, BW20
I (2349) wifi: pm start, type: 1

[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:381] _eventCallback(): STA IP: 192.168.43.154, MASK: 255.255.255.0, GW: 192.168.43.183
.
WiFi connected.
IP address: 
192.168.43.154

//////////////////////////SECOND AP + REBOOT >> WIFI CONNECTION OK ///////////////////////////////
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
Connecting to Slayes
I (35) wifi: wifi driver task: 3ffb5784, prio:23, stack:3584, core=0
I (104) wifi: wifi firmware version: 7997e4b
I (105) wifi: config NVS flash: enabled
I (105) wifi: config nano formating: disabled
I (105) wifi: Init dynamic tx buffer num: 32
I (109) wifi: Init data frame dynamic rx buffer num: 32
I (114) wifi: Init management frame dynamic rx buffer num: 32
I (119) wifi: Init management short buffer num: 32
I (124) wifi: Init static rx buffer size: 1600
I (128) wifi: Init static rx buffer num: 16
I (132) wifi: Init dynamic rx buffer num: 32
I (188) wifi: mode : sta (fc:f5:c4:01:a4:40)
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
I (314) wifi: n:11 0, o:1 0, ap:255 255, sta:11 0, prof:4
I (315) wifi: state: init -> auth (b0)
..I (1315) wifi: state: auth -> init (200)
I (1315) wifi: n:11 0, o:11 0, ap:255 255, sta:11 0, prof:4
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 2 - AUTH_EXPIRE
I (1458) wifi: n:11 0, o:11 0, ap:255 255, sta:11 0, prof:4
..I (2230) wifi: state: init -> auth (b0)
I (2234) wifi: state: auth -> init (1c0)
I (2234) wifi: n:11 0, o:11 0, ap:255 255, sta:11 0, prof:4
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 202 - AUTH_FAIL
......................```
```

Adminius, 2020-11-23:
same issue with ESP32 wemos D1 mini...

CelliesProjects, 2021-01-23:
Probably a shoddy board see https://github.com/espressif/arduino-esp32/issues/3899.

Boils down to: 
Use a 3v3 regulator and don't trust the onboard one.

closed 2021-07-08 by stale[bot]