espressif/arduino-esp32 #4521: Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.

yogeshwaran774, 2020-11-11:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.4
IDE name:						Arduino IDE
Flash Frequency:					40Mhz
PSRAM enabled:                                      no
Upload Speed:					115200
Computer OS:                                        Windows 10

### Description:
I am using timer 0 and one task which listen queue and read from the queue then post data to the cloud via mqtt.
Things are working fine for sometime then i am getting Exception at rtc_isr. I am not able to understand what is happening by checking at the exception decoder. I have attached the Exception below can someone guide me to handle this?

### Debug Messages:
```
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40142f3c: f01d020c 2d004136 20c04591
Core 0 register dump:
PC      : 0x40142f43  PS      : 0x00050031  A0      : 0x40084fd8  A1      : 0x3ffbe1b0  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00060023  A5      : 0x4008bb78  
A6      : 0x00000000  A7      : 0x00000001  A8      : 0x00000000  A9      : 0x3ffb5240  
A10     : 0x00000000  A11     : 0x3ffbbe00  A12     : 0x00060023  A13     : 0x3ffbc784  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000018  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
Core 0 was running in ISR context:
EPC1    : 0x40142f43  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x00000000

Backtrace: 0x40142f43:0x3ffbe1b0 0x40084fd5:0x3ffbe1d0 0x40158033:0x3ffbc100 0x400e919b:0x3ffbc120 0x4008abd5:0x3ffbc140 0x400893e1:0x3ffbc160
```

**Exception Decoder**
PC: 0x40142f43: rtc_isr at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/driver/rtc_module.c line 1965
EXCVADDR: 0x00000000

Decoding stack results
0x40142f43: rtc_isr at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/driver/rtc_module.c line 1965
0x4008a55d: vTaskExitCritical at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c line 4274
0x400891a5: xQueueGenericReceive at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c line 1546
0x400d2937: post_to_cloud(void*) at C:\Users\91953\Desktop\Hacklab-master\Hacklab-master\Products\P019-Ble_Tag\Firmware\Esp32_Nrf_Uart/Esp32_Nrf_Uart.ino line 1174
0x400893e1: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

lbernstone, 2020-11-11:
Your code is throwing the error at line 1174.  Most likely, you are doing something in an ISR which is not allowed.  ISRs must run quickly and uninterrupted.  Set a flag variable or semaphore, and do your processing after the ISR.

yogeshwaran774, 2020-11-11:
> Your code is throwing the error at line 1174. Most likely, you are doing something in an ISR which is not allowed. ISRs must run quickly and uninterrupted. Set a flag variable or semaphore, and do your processing after the interrupt.

I am using gpio interrupt from external rtc module and the interrupt handler was just incrementing one variable and timer 0 was running with 15 sec timeout and i am feeding it each 5 sec once. Including with this mqtt subscribe callback was attached. 

I monitored while esp32 restarts like there were no subscribe callback call. There is something with rtc_isr which is associated with timer in esp32. Is that correct?

closed 2021-01-24 by stale[bot]