espressif/arduino-esp32 #4525: how to read i2c with esp32 NodeMCU scd30 co2 sensor

mourazo, 2020-11-12:
Make your question, not a Statement, inclusive. Include all pertinent information:

What you are trying to do?

I'm trying to read a scd30 sensor with a esp32 nodeMCU

Describe your system( Hardware, computer, O/S, core version, environment).

Esp32 nodeMCU programmed with ArduinoIDE and sensirion scd30 co2 sensor

Describe what is failing.

When I run the sketch in the board it doesn't work, the board doesn't do anything and the serial monitor is empty. I run the same skecth in arduino without problem.

Show the shortest possible code that will duplicate the error.

The error is general, the board doesn't work.

### Hardware:
Board:							ESP32 nodeMCU
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                      no
Upload Speed:					115200
Computer OS:                                        Windows 10

### Sketch:  

#include <Wire.h>

#include "SparkFun_SCD30_Arduino_Library.h" //Click here to get the library: http://librarymanager/All#SparkFun_SCD30
SCD30 airSensor;

```
void setup()
{
   
  Serial.begin(115200);
 
  Serial.println("SCD30 Example");
  Wire.begin(35,34, 1000000);

  if (airSensor.begin() == false)
  {
    Serial.println("Air sensor not detected. Please check wiring. Freezing...");
    while (1)
      ;
  }

  //The SCD30 has data ready every two seconds
}

void loop()
{
  if (airSensor.dataAvailable())
  {
    Serial.print("co2(ppm):");
    Serial.print(airSensor.getCO2());

    Serial.print(" temp(C):");
    Serial.print(airSensor.getTemperature(), 1);

    Serial.print(" humidity(%):");
    Serial.print(airSensor.getHumidity(), 1);

    Serial.println();
  }
  else
    Serial.println("Waiting for new data");

  delay(500);
}
```

chegewara, 2020-11-12:
Hi,
try to use different pins. GPIO35 is I/O.

mourazo, 2020-11-12:
> Hi,
> try to use different pins. GPIO35 is I/O.

Hi, thanks for response, I tried with the defaults ports scl/sda 22/21 with the same results. Thanks.

chegewara, 2020-11-12:
Then please check connection. This sensor is nice, but is not easy to work with on breadboard.
Next thing to do is to use examples from library and maybe 1MHz is too fast for it, try with 100-400kHz.
In my case it was sequence to init it:
```
      Wire.begin();
      delay(1000);
      scd30.initialize();
      delay(100);
      scd30.setAutoSelfCalibration(1);
      scd30.setMeasurementInterval(5);
```

mourazo, 2020-11-12:
Thanks for your support, I have managed to make it work in the GPIO12 and GPIO14 but I need to make it work between the next pins: A0, GPIO35, GPIO34, GPIO39, GPIO33, GPIO26, GPIO25 or GPIO32 because I have one [az-touch](https://www.az-delivery.de/es/products/az-touch-wandgehauseset-mit-touchscreen-fur-esp8266-und-esp32) and this are the pins availables for connect to the board. All combinations that I try doesn't work, why? it is posible make it work in this pins? Thanks for all.

mourazo, 2020-11-12:
Now its working in the GPIO25 _EMAC_RXD0 and GPIO26_EMAC_RXD1. Thanks for your support

lbernstone, 2020-11-12:
pins >33 are input only. https://github.com/espressif/arduino-esp32#esp32dev-board-pinmap

mourazo, 2020-11-16:
> pins >33 are input only. https://github.com/espressif/arduino-esp32#esp32dev-board-pinmap

Thanks!

closed 2020-11-16 by mourazo