espressif/arduino-esp32 #4526: how more partitions

hzhh110, 2020-11-12:
[env:nodemcu-32s]
platform = espressif32
; board = esp32cam
board = nodemcu-32s

; board = ttgo-t-watch
framework = arduino
lib_deps  = AliyunMqttArduino,ArduinoJson,NTPClient,ESPiLight,lsatan/SmartRC-CC1101-Driver-Lib @ ^2.5.2
; lib_deps  = AliyunMqttArduino
; lib_deps  = 551,64,6100
; upload_flags =
;     --port=3232
build_flags =
  -D MQTT_MAX_PACKET_SIZE=1024
  -D MQTT_KEEPALIVE=60

board_build.partitions = esp32cam2.csv

# Name | Type | SubType | Offset | Size | Flags
-- | -- | -- | -- | -- | --
nvs | data | nvs | 0x9000 | 0x5000 |  
otadata | data | ota | 0xe000 | 0x2000 |  
app0 | app | ota_0 | 0x10000 | 0x1E0000 |  
app1 | app | ota_1 | 0x1F0000 | 0x1E0000 |  
spiffs | data | spiffs | 0x3D0000 | 0x30000 |  

if I user 

# Name | Type | SubType | Offset | Size | Flags
-- | -- | -- | -- | -- | --
nvs | data | nvs | 0x2000 | 0xc000 |  
otadata | data | ota | 0xe000 | 0x2000 |  
app0 | app | ota_0 | 0x10000 | 0x1E0000 |  
app1 | app | ota_1 | 0x1F0000 | 0x1E0000 |  
spiffs | data | spiffs | 0x3D0000 | 0x30000 |  

CSV Error: First partition offset 0x2000 overlaps end of partition table 0x9000

chegewara, 2020-11-12:
Partition table ends at offset 0x9000, so you cant put any partition in there.

hzhh110, 2020-11-12:
Can nvs be bigger? How to set it?

chegewara, 2020-11-12:
Few ways to do it:
- move it last partition,
- make it bigger and start next partitions with empty offset, it will be filled auto-magically.

hzhh110, 2020-11-12:
# Name | Type | SubType | Offset | Size | Flags
-- | -- | -- | -- | -- | --
nvs | data | nvs | 0x9000 | 0x10000 |  
otadata | data | ota | 0x19000 | 0x2000 |  
app0 | app | ota_0 | 0x1b000 | 0x1E0000 |  
app1 | app | ota_1 | 0x1Fb000 | 0x1E0000 |  
spiffs | data | spiffs | 0x3Db000 | 0x25000 |

hzhh110, 2020-11-12:
Partition app0 invalid: Offset 0x1b000 is not aligned to 0x10000
Compiling .pio/build/nodemcu-32s/libc6c/Update/Updater.cpp.o
Compiling .pio/build/nodemcu-32s/lib7cd/WiFi/ETH.cpp.o
*** [.pio/build/nodemcu-32s/partitions.bin] Error 2

chegewara, 2020-11-12:
`Partition app0 invalid: Offset 0x1b000 is not aligned to 0x10000`
app0 offset = 0x20000 or 0x30000 or 0x40000 etc, but try to leave all offsets empty (space).

hzhh110, 2020-11-12:
Ask a lower-level question: nvs size: how many bytes of data can 0x5000 store?

lbernstone, 2020-11-12:
lmgtfy: https://www.google.com/?q=convert+hexadecimal+decimal

hzhh110, 2020-11-13:
struct MyRemoteData
{

       unsigned long long longLongValue;
      unsigned short MHZType;
	unsigned short bitLenght;
	unsigned short protocol;
       unsigned short switchMode;
      unsigned short encryptionMode;
      unsigned short syncValue;
};
#define kChannelNum  5
#define kBtnNum  4
struct DeviceDetails
{
  bool isHaveWiFi;
  bool isHaveDetails;
  char wifissid[32];
  char wifipassword[32];
  char porductKey[20];
  char myDeviceName[20];
  char detailsId[20];
  char region[20];
  char deviceSecret[50];
  char serialNum[20];
  char verifyCode[20];
  bool isUpgrade;
  bool upgradeSuccess; 
  char userId[20];
  char oldVersion[20];
  char admin[20];
  unsigned int userIds[MaxUsers];//MaxUsers = 11
  MyRemoteData remoteData[kChannelNum][kBtnNum];
}; 

I don't know what's going on, every time I save the structure, there will be the problem of subsequent storage failure.
MyRemoteData remoteData[kChannelNum][kBtnNum];  remoteData[1][4]     remoteData[2][4]    no problem   //  remoteData[4][4]     remoteData[5][4]    something wrong

hzhh110, 2020-11-13:
void saveDetailsConfig(DeviceDetails details)
{
  int size =   sizeof(details);
  EEPROM.begin(size);
  uint8_t *p = (uint8_t *)(&details);
  for (int i = 0; i < size; i++)
  {
    EEPROM.write(i, *(p + i));
  }
  EEPROM.commit();
  EEPROM.end();
}

hzhh110, 2020-11-13:
void saveDetailsConfig(DeviceDetails details)
{

  size_t size =   sizeof(details);
  nvs_write_wifi(&details,size);
}
esp_err_t nvs_write_wifi(struct DeviceDetails *details, size_t len)
{
    nvs_handle write_handle;
    esp_err_t err;

    // printfln("NVS_WRITE：存储wifi信息\n");
    // Open
    err = nvs_open(STORAGE_NAMESPACE, NVS_READWRITE, &write_handle);
    if (err != ESP_OK)
    {
        // printf("NVS_WRITE：存储空间打开失败\n");
        return err;
    }
    err = nvs_set_blob(write_handle, "details", details, len);
    if (err != ESP_OK)
    {
        // printf("NVS_WRITE：存储空间存储失败\n");
        return err;
    }
    err = nvs_commit(write_handle);
    if (err != ESP_OK)
    {
        // printf("NVS_WRITE：存储空间提交失败\n");
        return err;
    }

    nvs_close(write_handle);
    return ESP_OK;
}

hzhh110, 2020-11-13:
esp_err_t nvs_write_wifi(struct DeviceDetails *details, size_t len)
{
    nvs_handle write_handle;
    esp_err_t err;
    err = nvs_open(STORAGE_NAMESPACE, NVS_READWRITE, &write_handle);
    if (err != ESP_OK)
    {
        return err;
    }
    err = nvs_set_blob(write_handle, "details", details, len);
    if (err != ESP_OK)
    {
        return err;
    }
    err = nvs_commit(write_handle);
    if (err != ESP_OK)
    {
        return err;
    }

    nvs_close(write_handle);
    return ESP_OK;
}

hzhh110, 2020-11-13:
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d219d  PS      : 0x00060230  A0      : 0x800d6ab1  A1      : 0x3ffce3a0  
A2      : 0x00000005  A3      : 0x00000004  A4      : 0x3ffc6330  A5      : 0x3ffc61c0  
A6      : 0x00000010  A7      : 0xe4dd8968  A8      : 0x000003c0  A9      : 0x3f4025e0  
A10     : 0x00000001  A11     : 0x3ffceb10  A12     : 0x3ffcea5c  A13     : 0x3ffc6324  
A14     : 0x00000170  A15     : 0xff000000  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xe4dd8968  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000

hzhh110, 2020-11-13:
struct MyRemoteData
{
         unsigned long long longLongValue;
         unsigned int MHZType;
	unsigned int bitLenght;
        unsigned int delay;
	unsigned int protocol;
        unsigned int switchMode;
      unsigned int encryptionMode;
      unsigned int syncValue;

};

closed 2021-01-30 by stale[bot]