espressif/arduino-esp32 #4527: [v4.2 branch] enabling PSRAM on ESP32-S2 board does not boot anymore

ladyada, 2020-11-12:
reproduce by uploading any sketch to ESP32-S2 board using WROVER module with v4.2 branch - when PSRAM is enabled and the hardware usart output is

```
17:37:21.428 -> ESP-ROM:esp32s2-rc4-20191025
17:37:21.428 -> Build:Oct 25 2019
17:37:21.428 -> rst:0x3 (RTC_SW_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
17:37:21.428 -> Saved PC:0x40025eb9
17:37:21.428 -> SPIWP:0xee
17:37:21.428 -> mode:DIO, clock div:1
17:37:21.428 -> load:0x3ffe6100,len:0x8
17:37:21.428 -> load:0x3ffe6108,len:0x608
17:37:21.428 -> load:0x4004c000,len:0xa38
17:37:21.428 -> load:0x40050000,len:0x2848
17:37:21.428 -> entry 0x4004c190
17:37:21.568 -> E (148) spiram: SPI RAM not initialized
17:37:21.568 -> 
17:37:21.568 -> abort() was called at PC 0x40087972 on core 0
17:37:21.568 -> 
17:37:21.568 -> Backtrace:0x40027666:0x3ffc4df0 0x40027f35:0x3ffc4e10 0x4002c8c9:0x3ffc4e30 0x40087972:0x3ffc4eb0 0x40087993:0x3ffc4ed0 0x4008140d:0x3ffc4f00 0x400813a9:0x3ffc4f20 0x4008186a:0x3ffc4f40 0x4009c3a8:0x3ffc4f70
17:37:21.568 -> 
17:37:21.568 -> 
17:37:21.568 -> ELF file SHA256: 0000000000000000
17:37:21.568 -> 
17:37:21.568 -> Rebooting...
```

Miraculix200, 2020-11-13:
Enabling PSRAM works for me with the version cloned on September 10, 2020 (2656452d044215d5a49ba6e909f4ea19c089baa9) on the ESP32-S2 WROVER. Maybe try that?

My code uses so much memory it crashes without PSRAM. Did not try any later versions yet.

me-no-dev, 2020-11-16:
fixed here: https://github.com/espressif/arduino-esp32/commit/cbafe67ee1fd0bc31a4c98e2ae52cb322ab0ef81

closed 2020-11-16 by me-no-dev

ladyada, 2020-11-17:
yes this is no longer crashing thanx!