espressif/arduino-esp32 #4528: [v4.2 branch] Auto-reset ESP32-S2 USB UART to bootloader times out on packet header or on changing mode

ladyada, 2020-11-12:
Using windows 10 and Arduino 1.8.13 with ESP32-S2 WROVER, uploading 'blink sketch'

Uploading from ROM bootloader works great. Uploading from "user USB-serial" mode gets close to working!

Sometimes can switch mode and get into the bootloader and then times out later in the process

```
C:\Users\ladyada\Desktop\arduino-1.8.13\arduino-builder -dump-prefs -logger=machine -hardware C:\Users\ladyada\Desktop\arduino-1.8.13\hardware -hardware C:\Users\ladyada\AppData\Local\Arduino15\packages -hardware C:\Users\ladyada\Dropbox\ArduinoSketches\hardware -tools C:\Users\ladyada\Desktop\arduino-1.8.13\tools-builder -tools C:\Users\ladyada\Desktop\arduino-1.8.13\hardware\tools\avr -tools C:\Users\ladyada\AppData\Local\Arduino15\packages -built-in-libraries C:\Users\ladyada\Desktop\arduino-1.8.13\libraries -libraries C:\Users\ladyada\Dropbox\ArduinoSketches\libraries -fqbn=espressif:arduino-esp32:adafruit_metro_esp32s2:SerialMode=cdc,PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none -vid-pid=239A_80DF -ide-version=10813 -build-path C:\Users\ladyada\AppData\Local\Temp\arduino_build_163958 -warnings=none -build-cache C:\Users\ladyada\AppData\Local\Temp\arduino_cache_257336 -prefs=build.warn_data_percentage=75 -verbose C:\Users\ladyada\Dropbox\ArduinoSketches\BSP\MetroESP32S2\Blink\Blink.ino
C:\Users\ladyada\Desktop\arduino-1.8.13\arduino-builder -compile -logger=machine -hardware C:\Users\ladyada\Desktop\arduino-1.8.13\hardware -hardware C:\Users\ladyada\AppData\Local\Arduino15\packages -hardware C:\Users\ladyada\Dropbox\ArduinoSketches\hardware -tools C:\Users\ladyada\Desktop\arduino-1.8.13\tools-builder -tools C:\Users\ladyada\Desktop\arduino-1.8.13\hardware\tools\avr -tools C:\Users\ladyada\AppData\Local\Arduino15\packages -built-in-libraries C:\Users\ladyada\Desktop\arduino-1.8.13\libraries -libraries C:\Users\ladyada\Dropbox\ArduinoSketches\libraries -fqbn=espressif:arduino-esp32:adafruit_metro_esp32s2:SerialMode=cdc,PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none -vid-pid=239A_80DF -ide-version=10813 -build-path C:\Users\ladyada\AppData\Local\Temp\arduino_build_163958 -warnings=none -build-cache C:\Users\ladyada\AppData\Local\Temp\arduino_cache_257336 -prefs=build.warn_data_percentage=75 -verbose C:\Users\ladyada\Dropbox\ArduinoSketches\BSP\MetroESP32S2\Blink\Blink.ino
Using board 'adafruit_metro_esp32s2' from platform in folder: C:\Users\ladyada\Dropbox (Personal)\ArduinoSketches\hardware\espressif\arduino-esp32
Using core 'esp32' from platform in folder: C:\Users\ladyada\Dropbox (Personal)\ArduinoSketches\hardware\espressif\arduino-esp32
cmd /c if exist "C:\\Users\\ladyada\\Dropbox\\ArduinoSketches\\BSP\\MetroESP32S2\\Blink\\partitions.csv" COPY /y "C:\\Users\\ladyada\\Dropbox\\ArduinoSketches\\BSP\\MetroESP32S2\\Blink\\partitions.csv" "C:\\Users\\ladyada\\AppData\\Local\\Temp\\arduino_build_163958\\partitions.csv"
cmd /c if not exist "C:\\Users\\ladyada\\AppData\\Local\\Temp\\arduino_build_163958\\partitions.csv" COPY "C:\\Users\\ladyada\\Dropbox (Personal)\\ArduinoSketches\\hardware\\espressif\\arduino-esp32\\tools\\partitions\\\\default.csv" "C:\\Users\\ladyada\\AppData\\Local\\Temp\\arduino_build_163958\\partitions.csv"
Detecting libraries used...
"C:\\Users\\ladyada\\Dropbox (Personal)\\ArduinoSketches\\hardware\\espressif\\arduino-esp32/tools/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-g++" -DHAVE_CONFIG_H "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DUNITY_INCLUDE_CONFIG_H -DWITH_POSIX -D_GNU_SOURCE "-DIDF_VER=\"v4.3-dev-1561-g357a27760-dirty\"" -DESP_PLATFORM "-IC:\\Users\\ladyada\\Dropbox (Personal)\\ArduinoSketches\\hardware\\espressif\\arduino-esp32/tools/sdk/esp32s2/include/config" "-IC:\\Users\\ladyada\\Dropbox (Personal)\\ArduinoSketches\\hardware\\espressif\\arduino-esp32/tools/sdk/esp32s2/include/newlib/platform_include" "-IC:\\Users\\ladyada\\Dropbox (Personal)\\ArduinoSketches\\hardware\\espressif\\arduino-esp32/tools/sdk/esp32s2/include/freertos/include" "-IC:\\Users\\ladyada\\Dropbox (Personal)\\ArduinoSketches\\hardware\\espressif\\arduino-esp32/tools/sdk/esp32s2/include/freertos/xtensa/include" "-IC:\\Users\\ladyada\\Dropbox (Personal)\\ArduinoSketches\\hardware\\espressif\\arduino-esp32/tools/sdk/esp32s2/include/heap/include" "-IC:\\Users\\ladyada\\Dropbox (Personal)\\ArduinoSketches\\hardware\\espressif\\arduino-esp32/tools/sdk/esp32s2/include/log/include" "-IC:\\Users\\ladyada\\Dropbox (Personal)\\ArduinoSketches\\hardware\\espressif\\arduino-esp32/tools/sdk/esp32s2/include/lwip/include/apps" "-IC:\\Users\\ladyada\\Dropbox (Personal)\\ArduinoSketches\\hardware\\espressif\\arduino-esp32/tools/sdk/esp32s2/include/lwip/include/apps/sntp" "-IC:\\Users\\ladyada\\Dropbox (Personal)\\ArduinoSketches\\hardware\\espressif\\arduino-esp32/tools/sdk/esp32s2/include/lwip/lwip/src/include" "-IC:\\Users\\ladyada\\Dropbox (Personal)\\ArduinoSketches\\hardware\\espressif\\arduino-esp32/tools/sdk/esp32s2/include/lwip/port/esp32/include" "-IC:\\Users\\ladyada\\Dropbox (Personal)\\ArduinoSketches\\hardware\\espressif\\arduino-esp32/tools/sdk/esp32s2/include/lwip/port/esp32/include/arch" "-IC:\\Users\\ladyada\\Dropbox (Personal)\\ArduinoSketches\\hardware\\espressif\\arduino-esp32/tools/sdk/esp32s2/include/soc/include" "-IC:\\Users\\ladyada\\Dropbox (Personal)\\ArduinoSketches\\hardware\\espressif\\arduino-esp32/tools/sdk/esp32s2/include/soc/src/esp32s2" "-IC:\\Users\\ladyada\\Dropbox (Personal)\\ArduinoSketches\\hardware\\espressif\\arduino-esp32/tools/sdk/esp32s2/include/soc/src/esp32s2/include" "-IC:\\Users\\ladyada\\Dropbox (Personal)\\ArduinoSketches\\hardware\\espressif\\arduino-esp32/tools/sdk/esp32s2/include/hal/esp32s2/include" "-IC:\\Users\\ladyada\\Dropbox (Personal)\\ArduinoSketches\\hardware\\espressif\\arduino-esp32/tools/sdk/esp32s2/include/hal/include" "-IC:\\Users\\ladyada\\Dropbox (Personal)\\ArduinoSketches\\hardware\\espressif\\arduino-esp32/tools/sdk/esp32s2/include/esp_rom/include" "-IC:\\Users\\ladyada\\Dropbox (Personal)\\ArduinoSketches\\hardware\\espressif\\arduino-esp32/tools/sdk/esp32s2/include/esp_common/include" "-IC:\\Users\\ladyada\\Dropbox (Personal)\\ArduinoSketches\\hardware\\espressif\\arduino-esp32/tools/sdk/esp32s2/include/esp_system/include" "-IC:\\Users\\ladyada\\Dropbox (Personal)\\ArduinoSketches\\hardware\\espressif\\arduino-esp32/tools/sdk/esp32s2/include/xtensa/include" "-IC:\\Users\\ladyada\\Dropbox (Personal)\\ArduinoSketches\\hardware\\espressif\\arduino-esp32/tools/sdk/esp32s2/include/xtensa/esp32s2/include" "-IC:\\Users\\ladyada\\Dropbox (Personal)\\ArduinoSketches\\hardware\\espressif\\arduino-esp32/tools/sdk/esp32s2/include/esp32s2/include" "-IC:\\Users\\ladyada\\Dropbox (Personal)\\ArduinoSketches\\hardware\\espressif\\arduino-esp32/tools/sdk/esp32s2/include/driver/include" "-IC:\\Users\\ladyada\\Dropbox (Personal)\\ArduinoSketches\\hardware\\espressif\\arduino-esp32/tools/sdk/esp32s2/include/driver/esp32s2/include" "-IC:\\Users\\ladyada\\Dropbox (Personal)\\ArduinoSketches\\hardware\\espressif\\arduino-esp32/tools/sdk/esp32s2/include/esp_ringbuf/include" "-IC:\\Users\\ladyada\\Dropbox (Personal)\\ArduinoSketches\\hardware\\espressif\\arduino-esp32/tools/sdk/esp32s2/include/efuse/include" "-IC:\\Users\\ladyada\\Dropbox (Personal)\\ArduinoSketches\\hardware\\espressif\\arduino-esp32/tools/sdk/esp32s2/include/efuse/esp32s2/include" "-IC:\\Users\\ladyada\\Dropbox (Personal)\\ArduinoSketches\\hardware\\espressif\\arduino-esp32/tools/sdk/esp32s2/include/espcoredump/include" "-IC:\\Users\\ladyada\\Dropbox (Personal)\\ArduinoSketches\\hardware\\espressif\\arduino-esp32/tools/sdk/esp32s2/include/esp_timer/include" "-IC:\\Users\\ladyada\\Dropbox (Personal)\\ArduinoSketches\\hardware\\espressif\\arduino-esp32/tools/sdk/esp32s2/include/esp_ipc/include" "-IC:\\Users\\ladyada\\Dropbox (Personal)\\ArduinoSketches\\hardware\\espressif\\arduino-esp32/tools/sdk/esp32s2/include/esp_pm/include" "-IC:\\Users\\ladyada\\Dropbox (Personal)\\ArduinoSketches\\hardware\\espressif\\arduino-esp32/tools/sdk/esp32s2/include/soc/soc/esp32s2/include" "-IC:\\Users\\ladyada\\Dropbox (Personal)\\ArduinoSketches\\hardware\\espressif\\arduino-esp32/tools/sdk/esp32s2/include/soc/soc/include" "-IC:\\Users\\ladyada\\Dropbox (Personal)\\ArduinoSketches\\hardware\\espressif\\arduino-esp32/tools/sdk/esp32s2/include/soc/soc/esp32s2/private_include" "-IC:\\Users\\ladyada\\Dropbox [... cut ...]
Writing at 0x00022000... (63 %)
Writing at 0x00022800... (65 %)
Writing at 0x00023000... (67 %)
Writing at 0x00023800... (68 %)
Writing at 0x00024000... (70 %)
Writing at 0x00024800... (72 %)
Writing at 0x00025000... (74 %)
Writing at 0x00025800... (75 %)
Writing at 0x00026000... (77 %)
Writing at 0x00026800... (79 %)
Writing at 0x00027000... (81 %)
Writing at 0x00027800... (82 %)

A fatal error occurred: Timed out waiting for packet header
A fatal error occurred: Timed out waiting for packet header
```

me-no-dev, 2020-11-13:
yes I am aware, work on USB has not completed yet :) System timer is resetting after some time needs to be prepared properly before rebooting into Download. I'm thinking of turning off persistence in the branch for now. Will fix the issue, but the device will reenumerate. If default config is used, it will come up on the same port.

You can try by commenting out the following lines: https://github.com/espressif/arduino-esp32/blob/idf-release/v4.2/cores/esp32/esp32-hal-tinyusb.c#L534-L537

ladyada, 2020-11-13:
hmm i commented those out and it makes windows 10 upset, will manually reset for now

```
Global variables use 16320 bytes (4%) of dynamic memory, leaving 311360 bytes for local variables. Maximum is 327680 bytes.
esptool.py v3.0-dev
Serial port COM38
Connecting........
Traceback (most recent call last):
  File "esptool.py", line 3682, in <module>
  File "esptool.py", line 3675, in _main
  File "esptool.py", line 3330, in main
  File "esptool.py", line 515, in connect
  File "esptool.py", line 492, in _connect_attempt
  File "esptool.py", line 431, in sync
  File "esptool.py", line 362, in command
  File "site-packages\serial\serialutil.py", line 352, in timeout
  File "site-packages\serial\serialwin32.py", line 222, in _reconfigure_port
serial.serialutil.SerialException: Cannot configure port, something went wrong. Original message: WindowsError(433, 'A device which does not exist was specified.')
Failed to execute script esptool
the selected serial port Failed to execute script esptool
 does not exist or your board is not connected
```

me-no-dev, 2020-11-17:
could you please try with the latest branch code?

ladyada, 2020-11-17:
the 4.2 branch or master?

ladyada, 2020-11-17:
ok on the latest 4.2 commit i can reset into bootloader mode but it does not seem to be able to get out of bootloader mode. i have to press the reset button to launch the new code

ladyada, 2020-11-17:
about half the time it isnt able to get into bootloader mode, hangs at the Connecting... stage until i press reset then fails

```
c:/users/ladyada/dropbox (personal)/arduinosketches/hardware/espressif/esp32/tools/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: C:\Users\ladyada\Dropbox (Personal)\ArduinoSketches\hardware\espressif\esp32/tools/sdk/esp32s2/lib\librtc.a(rtc.o)(.text.rtc_pad_gpio_wakeup+0x50b7fb000000a9): could not decode instruction; possible configuration mismatch
"C:\\Users\\ladyada\\Dropbox (Personal)\\ArduinoSketches\\hardware\\espressif\\esp32/tools/esptool/esptool.exe" --chip esp32s2 elf2image --flash_mode dio --flash_freq 80m --flash_size 4MB -o "C:\\Users\\ladyada\\AppData\\Local\\Temp\\arduino_build_915917/Blink.ino.bin" "C:\\Users\\ladyada\\AppData\\Local\\Temp\\arduino_build_915917/Blink.ino.elf"
esptool.py v3.0-dev
"C:\\Users\\ladyada\\Dropbox (Personal)\\ArduinoSketches\\hardware\\espressif\\esp32/tools/gen_esp32part.exe" -q "C:\\Users\\ladyada\\AppData\\Local\\Temp\\arduino_build_915917/partitions.csv" "C:\\Users\\ladyada\\AppData\\Local\\Temp\\arduino_build_915917/Blink.ino.partitions.bin"
"C:\\Users\\ladyada\\Dropbox (Personal)\\ArduinoSketches\\hardware\\espressif\\esp32/tools/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-size" -A "C:\\Users\\ladyada\\AppData\\Local\\Temp\\arduino_build_915917/Blink.ino.elf"
Sketch uses 213598 bytes (16%) of program storage space. Maximum is 1310720 bytes.
Global variables use 16408 bytes (5%) of dynamic memory, leaving 311272 bytes for local variables. Maximum is 327680 bytes.
C:\Users\ladyada\Dropbox\ArduinoSketches\hardware\espressif\esp32/tools/esptool/esptool.exe --chip esp32s2 --port COM55 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0xe000 C:\Users\ladyada\Dropbox\ArduinoSketches\hardware\espressif\esp32/tools/partitions/boot_app0.bin 0x1000 C:\Users\ladyada\Dropbox\ArduinoSketches\hardware\espressif\esp32/tools/sdk/esp32s2/bin/bootloader_qio_80m.bin 0x10000 C:\Users\ladyada\AppData\Local\Temp\arduino_build_915917/Blink.ino.bin 0x8000 C:\Users\ladyada\AppData\Local\Temp\arduino_build_915917/Blink.ino.partitions.bin 
esptool.py v3.0-dev
Serial port COM55
Connecting........___
Traceback (most recent call last):
  File "esptool.py", line 3682, in <module>
  File "esptool.py", line 3675, in _main
  File "esptool.py", line 3330, in main
  File "esptool.py", line 515, in connect
  File "esptool.py", line 492, in _connect_attempt
  File "esptool.py", line 433, in sync
  File "esptool.py", line 362, in command
  File "site-packages\serial\serialutil.py", line 352, in timeout
  File "site-packages\serial\serialwin32.py", line 222, in _reconfigure_port
serial.serialutil.SerialException: Cannot configure port, something went wrong. Original message: WindowsError(31, 'A device attached to the system is not functioning.')
Failed to execute script esptool
the selected serial port Failed to execute script esptool
 does not exist or your board is not connected
```

me-no-dev, 2020-11-17:
I'll look into it further. I hope that one of my boards will fail. TinyUSB is not yet done deal. AMF what is in Arduino is different from what is currently in IDF :) Did it stop resetting the download mid way though? When you enter download mode manually, bootloader can not reset into the app by itself (USB only). You should get a message about it:
```
Wrote 3072 bytes (128 compressed) at 0x00008000 in 0.0 seconds (effective 5260.5 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
ERROR: ESP32-S2 chip was placed into download mode using GPIO0.
esptool.py can not exit the download mode over USB. To run the app, reset the chip manually.
To suppress this error, set --after option to 'no_reset'.
```

me-no-dev, 2020-11-17:
The other issue is that esptool expects the port to be available when it resets the device, which is not the case once persistence is disabled and that caused the earlier error you were getting. If you want a few seconds and rerun esptool it would go.

me-no-dev, 2020-11-17:
Commenting [this line](https://github.com/espressif/arduino-esp32/blob/idf-release/v4.2/cores/esp32/esp32-hal-tinyusb.c#L388) disables any tweaks about persistence, with the exception of the ability to reboot into download from software.

ladyada, 2020-11-17:
im no longer getting intermittant errors - once it starts the auto-reset and download it completes (altho it does not auto-reset into the loaded firmware until i press reset)

me-no-dev, 2020-11-17:
esptool will autoreset if you do not enter download manually. You do not need to press the buttons once you flash the latest 4.2. It should reboot fine on it's own

me-no-dev, 2020-11-17:
here is my test code:
```cpp
USBCDC USBSerial;

static void usbEventCallback(void* arg, esp_event_base_t event_base, int32_t event_id, void* event_data){
    if(event_base == ARDUINO_USB_EVENTS){
        arduino_usb_event_data_t * data = (arduino_usb_event_data_t*)event_data;
        switch (event_id){
            case ARDUINO_USB_STARTED_EVENT:
                Serial.println("USB PLUGGED");
                break;
            case ARDUINO_USB_STOPPED_EVENT:
                Serial.println("USB UNPLUGGED");
                break;
            case ARDUINO_USB_SUSPEND_EVENT:
                Serial.printf("USB SUSPENDED: remote_wakeup_en: %u\n", data->suspend.remote_wakeup_en);
                break;
            case ARDUINO_USB_RESUME_EVENT:
                Serial.println("USB RESUMED");
                break;
            
            default:
                break;
        }
    } else if(event_base == ARDUINO_USB_CDC_EVENTS){
        arduino_usb_cdc_event_data_t * data = (arduino_usb_cdc_event_data_t*)event_data;
        switch (event_id){
            case ARDUINO_USB_CDC_CONNECTED_EVENT:
                Serial.println("CDC CONNECTED");
                break;
            case ARDUINO_USB_CDC_DISCONNECTED_EVENT:
                Serial.println("CDC DISCONNECTED");
                break;
            case ARDUINO_USB_CDC_LINE_STATE_EVENT:
                Serial.printf("CDC LINE STATE: dtr: %u, rts: %u\n", data->line_state.dtr, data->line_state.rts);
                break;
            case ARDUINO_USB_CDC_LINE_CODING_EVENT:
                Serial.printf("CDC LINE CODING: bit_rate: %u, data_bits: %u, stop_bits: %u, parity: %u\n", data->line_coding.bit_rate, data->line_coding.data_bits, data->line_coding.stop_bits, data->line_coding.parity);
                break;
            case ARDUINO_USB_CDC_RX_EVENT:
                Serial.printf("CDC RX: %u\n", data->rx.len);
                {
                    uint8_t buf[data->rx.len];
                    size_t len = USBSerial.read(buf, data->rx.len);
                    Serial.write(buf, len);
                }
                break;
            
            default:
                break;
        }
    }
}

void setup() {
    Serial.begin(115200);
    Serial.setDebugOutput(true);
    USB.onEvent(usbEventCallback);
    USBSerial.onEvent(usbEventCallback);
    USBSerial.begin();
    USB.begin();
}
 
void loop() {
    while(Serial.available()){
        size_t l = Serial.available();
        uint8_t b[l];
        l = Serial.read(b, l);
        USBSerial.write(b, l);
    }
}
```

ladyada, 2020-11-17:
I'm just trying to upload simple blink sketch:

```
void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
  Serial.begin();
}

void loop() {
  digitalWrite(LED_BUILTIN, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(100);              // wait for a second
  digitalWrite(LED_BUILTIN, LOW);    // turn the LED off by making the voltage LOW
  delay(100);              // wait for a second
}
```

me-no-dev, 2020-11-17:
but you do want to upload over USB, which is not the default. USB does require specifically to be turned on (or Serial selected to go to USB CDC). Else USB is not on and you can't just upload to the board over it.

ladyada, 2020-11-18:
yeah i have USB CDC selected by default for my variants. it works...most of the time. just not always :)

me-no-dev, 2020-11-18:
I have a couple S2 boards here, one is Saola and the other has only native USB. Both boards work fine, I have them connected and trying upload from time to time hoping for fail :) so far no fail :( I tried random pick of not-so-great cables, those made no difference... still flashing fine. I am on a Mac though...

ladyada, 2020-11-18:
do you have access to windows 10 computer?

me-no-dev, 2020-11-18:
I have VM that I can try, though it's so bogged down that I doubt it has anything to do with reality :) I'll think of something

ladyada, 2020-11-18:
ok - a windows computer/laptop can be pretty inexpensive - a VM will not work well i think, USB never quite passes through the same way

me-no-dev, 2020-11-19:
Yeah I'm worried about the same thing. Will probably get some Intel Nuk type of small PC. It's good to have anyway :)

me-no-dev, 2020-11-19:
Bought an i3 windows thing. Can replicate the problem :) Will let you know when we solve it and I'll look for a workaround/non-persistent mode

ladyada, 2020-11-21:
yay thank you so much! a lot of people use windows and this will be really helpful

alantchochiev, 2021-01-06:
Hi there.  Any luck on getting the auto reset fixed.  It's pretty painful having to put the board in upload mode and then reset it every time.  

Thank you,
Alan

cezer-io, 2021-02-10:
Hi All,

Is there any update related to this issue, the idf version is working properly with a simple "USB Console" configuration in menuconfig. However, on Ardunio side there is a problem even which crashes the configuration made by esp-idf.

ozgurbostan, 2021-03-07:
The TinyUSB library is not working, there is a problem in Windows. As an alternative solution, is it possible to implement "USB Console" feature of ESP-IDF to Arduino ESP32 Core? @me-no-dev

ozgurbostan, 2021-03-11:
Any progress or any estimated finish time?

ladyada, 2021-06-20:
bumping stalebot

me-no-dev, 2021-07-20:
Can you please retry with [this](https://github.com/espressif/arduino-esp32/blob/master/libraries/USB/examples/USBSerial/USBSerial.ino) sketch and current master?

ladyada, 2021-07-22:
on windows 10, it can kick into the bootloader, but it doesnt recognize that the COM port changes. its COM3 using arduino runtime, and COM66 in the bootloader mode

[image]

```
C:\Users\ladyada\Dropbox\ArduinoSketches\hardware\espressif\esp32/tools/esptool/esptool.exe --chip esp32s2 --port COM3 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0xe000 C:\Users\ladyada\Dropbox\ArduinoSketches\hardware\espressif\esp32/tools/partitions/boot_app0.bin 0x1000 C:\Users\ladyada\AppData\Local\Temp\arduino_build_238156/Blink.ino.bootloader.bin 0x10000 C:\Users\ladyada\AppData\Local\Temp\arduino_build_238156/Blink.ino.bin 0x8000 C:\Users\ladyada\AppData\Local\Temp\arduino_build_238156/Blink.ino.partitions.bin 0x2d0000 C:\Users\ladyada\Dropbox\ArduinoSketches\hardware\espressif\esp32/variants/adafruit_metro_esp32s2/tinyuf2.bin 
esptool.py v3.1
Serial port COM3
Connecting........Traceback (most recent call last):

  File "esptool.py", line 4582, in <module>
  File "esptool.py", line 4575, in _main
  File "esptool.py", line 4074, in main
  File "esptool.py", line 121, in get_default_connected_device
  File "esptool.py", line 632, in connect
  File "esptool.py", line 596, in _connect_attempt
  File "esptool.py", line 486, in sync
  File "esptool.py", line 418, in command
  File "serial\serialutil.py", line 372, in timeout
  File "serial\serialwin32.py", line 222, in _reconfigure_port
serial.serialutil.SerialException: Cannot configure port, something went wrong. Original message: PermissionError(13, 'A device attached to the system is not functioning.', None, 31)
[2024] Failed to execute script esptool
the selected serial port [2024] Failed to execute script esptool
 does not exist or your board is not connected
```

me-no-dev, 2021-07-23:
hi @ladyada which board did you select? Can you try with the generic S2 module, select Upload through USB CDC and try again? The upload should have probed for the changed port and I do not see it in the log above

ladyada, 2021-09-14:
i retried this today while testing https://github.com/espressif/arduino-esp32/issues/4729 and on the metro esp32-s2, at least, auto-reset is much more stable so this seems to be ok!

FlorianBaumgartner, 2021-09-20:
@ladyada How did you achive that?
I‘ve installed the latest ESP32 toolchain (2.0.0) on my Windows 10 machine.
I can download code with the Arduino IDE to my UM FeatherS2 but I still need to press the reset button everytime to exit the programming mode. Is there a fix for that?

me-no-dev, 2021-09-20:
you should not need to press the reset to enter the bootloader, but if you do, you always need to press reset to exit it after flashing. This is not Arduino related, but how the USB download mode works on ESP32-S2

VojtechBartoska, 2021-09-22:
@me-no-dev seems this issue could be closed? Or do more testing is needed?

me-no-dev, 2021-09-22:
It should be OK to close. No more issues have been reported

closed 2021-09-22 by me-no-dev