espressif/arduino-esp32 #4530: [v4.2 branch] ESP32S2 Wire.begin() div zero in i2c_hal_set_bus_timing

ladyada, 2020-11-13:
example code to replicate:
```
#include <Wire.h>

void setup() {
  Wire.begin();
}

void loop() {
}
```

selected ESP32S2 dev module and upload with IDE 1.8.3

```
PC: 0x4008cb1a: i2c_hal_set_bus_timing at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/hal/esp32s2/include/hal/i2c_ll.h line 96
EXCVADDR: 0x00000000

Decoding stack results
0x4008cb17: i2c_hal_set_bus_timing at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/hal/i2c_hal.c line 179
0x400870f8: i2c_param_config at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/driver/i2c.c line 621
0x400812dc: i2cInit (C:\Users\ladyada\Dropbox at Personal)\ArduinoSketches\hardware\espressif\esp32\cores\esp32\esp32-hal-i2c.c line 1822
0x400811b7: TwoWire::begin(int, int, unsigned int) (C:\Users\ladyada\Dropbox at Personal)\ArduinoSketches\hardware\espressif\esp32\libraries\Wire\src\Wire.cpp line 110
0x400810c7: setup() at C:\Users\ladyada\Dropbox\ArduinoSketches\BSP\MetroESP32S2\i2ccrash/i2ccrash.ino line 4
0x40081856: loopTask(void*) (C:\Users\ladyada\Dropbox at Personal)\ArduinoSketches\hardware\espressif\esp32\cores\esp32\main.cpp line 27
```

me-no-dev, 2020-11-17:
done :)

closed 2020-11-17 by me-no-dev

ladyada, 2020-11-17:
thanks - will try soon!