espressif/arduino-esp32 #4531: vTaskList() and vTaskGetRunTimeStats() not (fully) supported

drws, 2020-11-13:
### Description:

I would like to use vTaskList() and vTaskGetRunTimeStats() in arduino-esp32, but compiling (linking) fails with "undefined reference" (error message below). The functions are [supported in esp-idf](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/freertos.html) and declared in [freertos/task.h](https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/include/freertos/freertos/task.h) and also configured in [freertos/FreeRTOSConfig.h](https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/include/freertos/freertos/FreeRTOSConfig.h).

I'm compiling for "ESP32 Dev Module" board running updated Linux system with Arduino v1.8.13, arduino-builder v1.5.4 and arduino-esp32 v1.0.4.

### Sketch: 
```cpp
#include <freertos/task.h>

void setup() {
  Serial.begin(115200);

  // test vTaskList()
  char buffer1[2048] = {0};
  vTaskList(buffer1);
  Serial.println(buffer1);

  // test vTaskGetRunTimeStats()
  char buffer2[2048] = {0};
  vTaskGetRunTimeStats(buffer2);
  Serial.println(buffer2);

  // end
  Serial.flush();
  while (1);
}

void IRAM_ATTR loop() {
  vTaskDelete(NULL);
}
```

And fails with:

    .../sketch.ino.cpp.o:(.literal._Z5setupv+0x24): undefined reference to `vTaskList'
    .../sketch.ino.cpp.o:(.literal._Z5setupv+0x28): undefined reference to `vTaskGetRunTimeStats'
    .../sketch/sketch.ino.cpp.o: In function `setup()':
    .../sketch.ino:8: undefined reference to `vTaskList'
    .../sketch.ino:12: undefined reference to `vTaskGetRunTimeStats'
    collect2: error: ld returned 1 exit status
    exit status 1
    Error compiling for board ESP32 Dev Module.

lbernstone, 2020-11-13:
arduino-esp32 contains compiled libraries from esp-idf.  These functions are not available unless you [recompile the libraries](https://github.com/espressif/arduino-esp32/blob/master/docs/lib_builder.md)

drws, 2020-11-17:
Do you maybe know the reason for this? The required support is enabled in [FreeRTOSConfig.h](https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/include/freertos/freertos/FreeRTOSConfig.h).

rpannekoek, 2021-01-26:
I would also like to use this functionality without having to build custom libraries. Indeed, there seems to be inconsistency between the defines in FreeRTOSConfig.h (which claim the functionality is enabled) and the ESP IDF libraries, which apparently have it disabled.

FrozenNick, 2021-02-05:
Are there any solutions to this? I am currently trying to solve this issue as well

chegewara, 2021-02-05:
It requires to enable this in sdkconfig:
`CONFIG_FREERTOS_GENERATE_RUN_TIME_STATS`

SalsaMicha, 2021-02-17:
Is it possible to add the config with the next release or has anyone a step by step how to build the components newly with the features enabled?

drws, 2021-06-26:
Since it appears this will not be a supported feature without manual [library rebuilding](https://github.com/espressif/arduino-esp32/blob/master/docs/lib_builder.md), I'm closing the issue.

closed 2021-06-26 by drws

positron48, 2025-02-02:
If anyone wants to enable these functions for the Arduino framework in PlatformIO, I have written a guide on how to do it.
https://positroid.tech/en/post/platformio-esp32-stats