espressif/arduino-esp32 #4532: Wrong definition of t-beam.build.board in board.txt

vsupacha, 2020-11-14:
### Hardware:
Board:							TTGO T-Beam
Core Installation version:			1.0.4
IDE name:						Arduino IDE
Computer OS:                                         Windows 10

### Description:
Board definition in board.txt is used as a compile-time macro ARDUINO_{build.board} as defined in [Arduino Platform Specification](https://arduino.github.io/arduino-cli/latest/platform-specification/). But the definition of T-Beam in board.txt is not valid.
```
t-beam.build.board=T-Beam
```

### Sketch: 
Try to use preprocessor to check board type. Almost boards are ok, except T-Beam with - (dash) in its build.board setting.
```c
#ifdef ARDUINO_T-Beam
#warning OK
#endif

#if defined(ARDUINO_T) 
#error Wrong board name
#endif
```

```
error: #error Wrong board name
 #error Wrong board name
  ^
exit status 1
```

### Debug Messages:
Compiler messages show that - (dash) appeared in compiler directive.
```
-DARDUINO_T-Beam -DARDUINO_ARCH_ESP32 "-DARDUINO_BOARD=\"T-Beam\"" "-DARDUINO_VARIANT=\"t-beam\"" 
warning: extra tokens at end of #ifdef directive
```

lbernstone, 2020-11-14:
Already fixed
https://github.com/espressif/arduino-esp32/commit/6f237a84152d59537a012e7bfc3edbabd6adbd9d

vsupacha, 2020-11-14:
Thank @lbernstone for pointing.

closed 2020-11-14 by vsupacha