espressif/arduino-esp32 #4533: WiFi.begin() should provide an option for "all channel scan"

drws, 2020-11-14:
### Description:
There are [two channel scanning methods](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_wifi.html#_CPPv418wifi_scan_method_t) in ESP32 STA modes. [WiFi.begin()](https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/src/WiFiSTA.cpp) uses ESP-IDF's default configuration, which yields WIFI_FAST_SCAN. In this mode, the ESP will start scanning from channel 1 up and connect to the first AP found with matching SSID. This always works in single-AP WLANs (which use one channel anyway). In a multiple-AP environment, this scanning method could select a non-optimal AP or even a problematic one just above the RSSI threshold. Alternatively, the WIFI_ALL_CHANNEL_SCAN method scans all channels and then selects the most appropriate AP. I think WiFi.begin() should provide an option to set the WIFI_ALL_CHANNEL_SCAN method (or even use it as a default?).

Currently, there are two partial solutions. First of all, specific APs can be selected via the BSSID parameter, but using it defeats the purpose of SSID and disables automatic roaming. Secondly, the channel parameter to WiFi.begin() selects the starting channel in the scan. This provides user an option to manually select the channel of the optimal AP, but again introduces many problems. AP channels need to be manually input and in the case of auto-reconfigurable Wi-Fi networks even that fails. It also fails if optimal AP is only currently offline and the second best one needs to be selected. There is a logic to use WIFI_ALL_CHANNEL_SCAN that enables roaming, seamless AP selection, and better failover.

Currently, the channel parameter is limited between 1 and 13. A quick, simple and compatible solution for implementing the above would be to use channel=0 to force WIFI_ALL_CHANNEL_SCAN method.

me-no-dev, 2020-11-14:
Already done for the upcoming 2.0.0 😄 https://github.com/espressif/arduino-esp32/blob/idf-release/v4.2/libraries/WiFi/src/WiFiSTA.cpp#L71

drws, 2020-11-17:
Great! Thanks for the info.

closed 2020-11-17 by drws