espressif/arduino-esp32 #4536: [ESP32-S2 / IDFv4.2] Wifi connection lost shortly after connect (ESP32-S2)

BavariaESP32, 2020-11-15:
Using a ESP32-S2-WROVER devboard (Olimex Devkit-Lipo or Saola-1R) - and arduino-esp32 from branch esp32s2 (or IDFv4.2) and trying to run my old programs - which are working fine on "normal" ESP32 boards -  which dealing with MQTT-message via Wifi.
I am using the Arduino IDE 1.8.13 – where I have replaced – for ESP32-S2 support – the 1.0.4 directory (..\Arduino15\packages\esp32\hardware\esp32\1.0.4) by the files from github esp32s2 branch (but also tried the IDFv4.2 branch with the same result).

 I observed that the Wifi-connection disappears shortly after it was established - when trying to connect the MQTT-broker.
(Same happens with other examples - when something in "addition" to Wifi should be done - e.g. Webserver Example - under File->Examples->Wifi->SimpleWebServer - shows the same effect).

Please find below an example for Wifi connected MQTT-Client with debug output from ESP32-S2 (not working) and ESP32 (Node32s – working).

### Hardware:
Board:							ESP32-S2 Dev Module WROVER-DevKit-Lipo/Saola-1R
Core Installation version:			1.0.4
IDE name:						Arduino IDE 
Flash Frequency:					40Mhz
PSRAM enabled:                                   	no
Upload Speed:				      	115200
Computer OS:                                        	Windows 10

### Sketch: 
```cpp

#include <WiFi.h>
#include <PubSubClient.h>
#include <Wire.h>
#include "esp32-hal-log.h"

// Replace the next variables with your SSID/Password combination
const char* ssid = "REPLACE_WITH_YOUR_SSID";
const char* password = "REPLACE_WITH_YOUR_PASSWORD";

// Add your MQTT Broker IP address, example:
//const char* mqtt_server = "YOUR_MQTT_BROKER_IP_ADDRESS";
const char* mqtt_server = "192.168.178.77";

// Replace the next variables by your topic for MQTT subscribe
#define CO2_TOPIC  "infodisplay/co2"

WiFiClient espClient;
PubSubClient client(espClient);
long lastMsg = 0;
char msg[50];
int value = 0;

void setup() {
  Serial.begin(115200);
  delay(500);

  esp_log_level_set("wifi", ESP_LOG_VERBOSE);      // enable WARN logs from WiFi stack

  setup_wifi();
  client.setServer(mqtt_server, 1883);
  client.setCallback(callback);

 }

void setup_wifi() {
  delay(10);
  // We start by connecting to a WiFi network
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);

  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
}

void callback(char* topic, byte* message, unsigned int length) {
  Serial.print("Message arrived on topic: ");
  Serial.print(topic);
  Serial.print(". Message: ");
  String messageTemp;
  
  Serial.println();
}

void reconnect() {
  // Loop until we're reconnected
  while (!client.connected()) {
    Serial.print("Attempting MQTT connection...");
    // Attempt to connect
    if (client.connect("ESP32S2")) {
      Serial.println("connected");
      // Subscribe
      client.subscribe(CO2_TOPIC);
    } else {
      Serial.print("failed, rc=");
      Serial.print(client.state());
      Serial.println(" try again in 5 seconds");
      // Wait 5 seconds before retrying
      delay(5000);
    }
  }
}
void loop() {
  if (!client.connected()) {
    reconnect();
  }
  client.loop();

  delay(5000);
}

```

Log from ESP32-S2-WROVER (DevKit-Lipo board from Olimex):

### Debug Messages ESP32-S2:
```
11:45:56.970 -> ESP-ROM:esp32s2-rc4-20191025
11:45:56.970 -> Build:Oct 25 2019
11:45:56.970 -> rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
11:45:56.970 -> SPIWP:0xee
11:45:56.970 -> mode:DIO, clock div:1
11:45:56.970 -> load:0x3ffe6100,len:0x8
11:45:56.970 -> load:0x3ffe6108,len:0x608
11:45:56.970 -> load:0x4004c000,len:0xa38
11:45:56.970 -> load:0x40050000,len:0x2848
11:45:56.970 -> entry 0x4004c190
11:45:57.726 -> 
11:45:57.726 -> Connecting to MyWifi1234
11:45:57.726 -> I (1282) wifi:wifi driver task: 3ffd3588, prio:23, stack:3584, core=0
11:45:57.726 -> I (1286) wifi:wifi firmware version: a94a000
11:45:57.726 -> I (1286) wifi:wifi certification version: v7.0
11:45:57.726 -> I (1286) wifi:config NVS flash: enabled
11:45:57.773 -> I (1289) wifi:config nano formating: disabled
11:45:57.773 -> I (1293) wifi:Init data frame dynamic rx buffer num: 32
11:45:57.773 -> I (1298) wifi:Init management frame dynamic rx buffer num: 32
11:45:57.773 -> I (1303) wifi:Init management short buffer num: 32
11:45:57.773 -> I (1308) wifi:Init static tx buffer num: 8
11:45:57.773 -> I (1312) wifi:Init tx cache buffer num: 16
11:45:57.773 -> I (1316) wifi:Init static rx buffer size: 2212
11:45:57.773 -> I (1320) wifi:Init static rx buffer num: 8
11:45:57.773 -> I (1323) wifi:Init dynamic rx buffer num: 32
11:45:57.873 -> I (1442) wifi:mode : sta (7c:df:a1:01:a1:1c)
11:45:57.927 -> I (1443) wifi:enable tsf
11:45:57.927 -> I (1443) wifi:Set ps type: 0
11:45:57.927 -> 
11:45:58.429 -> ....I (3543) wifi:new:<9,0>, old:<1,0>, ap:<255,255>, sta:<9,0>, prof:1
11:46:00.337 -> I (3890) wifi:state: init -> auth (b0)
11:46:00.337 -> I (3899) wifi:state: auth -> assoc (0)
11:46:00.337 -> I (3907) wifi:state: assoc -> run (10)
11:46:00.384 -> I (3937) wifi:connected with MyWifi1234, aid = 2, channel 9, BW20, bssid = 2c:91:ab:4b:77:2d
11:46:00.384 -> I (3937) wifi:security: WPA2-PSK, phy: bgn, rssi: -35
11:46:00.384 -> I (3940) wifi:pm start, type: 0
11:46:00.384 -> 
11:46:00.437 -> I (3959) wifi:AP's beacon interval = 102400 us, DTIM period = 1
11:46:00.437 -> .
11:46:00.437 -> WiFi connected
11:46:00.437 -> IP address: 
11:46:00.437 -> 192.168.178.51
11:46:00.437 -> Attempting MQTT connection...I (5444) wifi:state: run -> init (22c0)
11:46:01.890 -> I (5444) wifi:pm stop, total sleep time: 0 us / 1502125 us
11:46:01.890 -> 
11:46:01.890 -> W (5444) wifi:<ba-del>idx
11:46:01.890 -> I (5445) wifi:new:<9,0>, old:<9,0>, ap:<255,255>, sta:<9,0>, prof:1
11:46:01.890 -> failed, rc=-2 try again in 5 seconds
11:46:06.910 -> Attempting MQTT connection...failed, rc=-2 try again in 5 seconds
11:46:11.929 -> Attempting MQTT connection...failed, rc=-2 try again in 5 seconds
11:46:16.901 -> Attempting MQTT connection...failed, rc=-2 try again in 5 seconds_
```

### Problem:
As you can see in this line (from above)
**_11:46:01.890 -> I (5444) wifi:pm stop, total sleep time: 0 us / 1502125 us_**
Wifi is **stopped and disconnected** - Why? This does not happen on ESP32-boards - see below:

The same sketch works on ESP32 (Node32s) board – with “standard” ESP32 package:
### Debug Messages ESP32:
```
13:03:11.982 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
13:03:11.982 -> configsip: 0, SPIWP:0xee
13:03:11.982 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
13:03:11.982 -> mode:DIO, clock div:1
13:03:11.982 -> load:0x3fff0018,len:4
13:03:11.982 -> load:0x3fff001c,len:1044
13:03:11.982 -> load:0x40078000,len:8896
13:03:11.982 -> load:0x40080400,len:5816
13:03:11.982 -> entry 0x400806ac
13:03:12.779 -> 
13:03:12.779 -> Connecting to MyWifi1234
13:03:12.779 -> I (545) wifi: wifi driver task: 3ffb57a4, prio:23, stack:3584, core=0
13:03:12.779 -> I (1159) wifi: wifi firmware version: 7997e4b
13:03:12.779 -> I (1160) wifi: config NVS flash: enabled
13:03:12.779 -> I (1160) wifi: config nano formating: disabled
13:03:12.779 -> I (1161) wifi: Init dynamic tx buffer num: 32
13:03:12.779 -> I (1165) wifi: Init data frame dynamic rx buffer num: 32
13:03:12.826 -> I (1170) wifi: Init management frame dynamic rx buffer num: 32
13:03:12.826 -> I (1175) wifi: Init management short buffer num: 32
13:03:12.826 -> I (1180) wifi: Init static rx buffer size: 1600
13:03:12.826 -> I (1184) wifi: Init static rx buffer num: 16
13:03:12.826 -> I (1188) wifi: Init dynamic rx buffer num: 32
13:03:12.873 -> I (1255) wifi: mode : sta (30:ae:a4:45:6c:e0)
13:03:13.014 -> I (1378) wifi: n:9 0, o:1 0, ap:255 255, sta:9 0, prof:1
13:03:13.014 -> I (1379) wifi: state: init -> auth (b0)
13:03:13.014 -> I (1383) wifi: state: auth -> assoc (0)
13:03:13.014 -> I (1391) wifi: state: assoc -> run (10)
13:03:13.061 -> I (1421) wifi: connected with MyWifi1234, channel 9, BW20
13:03:13.061 -> I (1427) wifi: pm start, type: 1
13:03:13.061 -> 
13:03:13.389 -> .
13:03:13.389 -> WiFi connected
13:03:13.389 -> IP address: 
13:03:13.389 -> 192.168.178.102
13:03:13.389 -> Attempting MQTT connection...connected
13:03:23.421 -> Message arrived on topic: infodisplay/co2. Message: 853
13:03:38.424 -> Message arrived on topic: infodisplay/co2. Message: 916
13:04:08.427 -> Message arrived on topic: infodisplay/co2. Message: 902
```
I appreciate any help.
Thank you!

me-no-dev, 2020-11-15:
- Remove the ESP32 package from the board manager and remove the json URL
- Clone `idf-release/v4.2` branch of this repository, otherwise following the instructions for the development version [here](https://github.com/espressif/arduino-esp32#installation-instructions). This will update everything to the proper version and will pull some updates that are not in the esp32s2 branch. That one is just for history.
- Try again and make sure that you have set the debug level in the board menu to `Debug` or `Verbose`

BavariaESP32, 2020-11-15:
Hi me-no-dev,
Thanks for the quick reply. I followed your advise and deleted the ESP32 package as well as the JSON-URL from the Boardmanager setup. Then downloaded the idf-release/v4.2-branch and executed the get.exe 

Unfortunately this also doesn't seems to solve the problem.

One time I got "MQTT connected" but around the time frame where the "topic" should come in the Wifi was disconnected again (see Log1) 
and I also got the same behaviour as mentioned in the first post -  please see Log 2 - now with VERBOSE debug info:

### Debug output Log1
```
18:11:14.924 -> ESP-ROM:esp32s2-rc4-20191025
18:11:14.924 -> Build:Oct 25 2019
18:11:14.924 -> rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
18:11:14.924 -> SPIWP:0xee
18:11:14.924 -> mode:DIO, clock div:1
18:11:14.924 -> load:0x3ffe6100,len:0x8
18:11:14.924 -> load:0x3ffe6108,len:0x620
18:11:14.924 -> load:0x4004c000,len:0xa38
18:11:14.924 -> load:0x40050000,len:0x2848
18:11:14.924 -> entry 0x4004c190
18:11:15.673 -> 
18:11:15.673 -> Connecting to MyWifi1234
18:11:15.673 -> I (1284) wifi:wifi driver task: 3ffd3c40, prio:23, stack:3584, core=0
18:11:15.673 -> I (1288) wifi:wifi firmware version: d5c1f19
18:11:15.673 -> I (1288) wifi:wifi certification version: v7.0
18:11:15.673 -> I (1288) wifi:config NVS flash: enabled
18:11:15.673 -> I (1291) wifi:config nano formating: disabled
18:11:15.673 -> I (1295) wifi:Init data frame dynamic rx buffer num: 32
18:11:15.673 -> I (1300) wifi:Init management frame dynamic rx buffer num: 32
18:11:15.727 -> I (1305) wifi:Init management short buffer num: 32
18:11:15.727 -> I (1310) wifi:Init static tx buffer num: 8
18:11:15.727 -> I (1314) wifi:Init tx cache buffer num: 16
18:11:15.727 -> I (1318) wifi:Init static rx buffer size: 2212
18:11:15.727 -> I (1322) wifi:Init static rx buffer num: 8
18:11:15.727 -> I (1325) wifi:Init dynamic rx buffer num: 32
18:11:15.827 -> I (1452) wifi:mode : sta (7c:df:a1:01:a1:1c)
18:11:15.827 -> I (1452) wifi:enable tsf
18:11:15.827 -> [  1438][V][WiFiGeneric.cpp:273] _arduino_event_cb(): STA Started
18:11:15.827 -> [  1438][D][WiFiGeneric.cpp:795] _eventCallback(): Arduino Event: 2 - STA_START
18:11:15.874 -> I (1460) wifi:Set ps type: 0
18:11:15.874 -> 
18:11:15.874 -> [  1453][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
18:11:16.376 -> ....I (3527) wifi:new:<9,0>, old:<1,0>, ap:<255,255>, sta:<9,0>, prof:1
18:11:18.237 -> I (3875) wifi:state: init -> auth (b0)
18:11:18.284 -> I (3882) wifi:state: auth -> assoc (0)
18:11:18.284 -> I (3890) wifi:state: assoc -> run (10)
18:11:18.284 -> I (3918) wifi:connected with MyWifi1234, aid = 2, channel 9, BW20, bssid = 2c:91:ab:4b:77:2d
18:11:18.284 -> I (3919) wifi:security: WPA2-PSK, phy: bgn, rssi: -34
18:11:18.338 -> I (3922) wifi:pm start, type: 0
18:11:18.338 -> 
18:11:18.338 -> [  3909][V][WiFiGeneric.cpp:285] _arduino_event_cb(): STA Connected: SSID: MyWifi1234, BSSID: 2c:91:ab:4b:77:2d, Channel: 9, Auth: UNKNOWN
18:11:18.338 -> [  3921][D][WiFiGeneric.cpp:795] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
18:11:18.338 -> [  3948][V][WiFiGeneric.cpp:295] _arduino_event_cb(): STA Got New IP:192.168.178.51
18:11:18.338 -> [  3949][D][WiFiGeneric.cpp:795] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
18:11:18.338 -> [  3952][D][WiFiGeneric.cpp:844] _eventCallbI (3971) wifi:AP's beacon interval = 102400 us, DTIM period = 1
18:11:18.385 -> ack(): STA IP: 192.168.178.51, MASK: 255.255.255.0, GW: 192.168.178.1
18:11:18.385 -> .
18:11:18.385 -> WiFi connected
18:11:18.385 -> IP address: 
18:11:18.385 -> 192.168.178.51
18:11:18.385 -> Attempting MQTT connection...connected
18:11:33.647 -> I (19251) wifi:state: run -> init (22c0)
18:11:33.647 -> I (19252) wifi:pm stop, total sleep time: 0 us / 15328214 us
18:11:33.647 -> 
18:11:33.647 -> W (19252) wifi:<ba-del>idx
18:11:33.647 -> W (19252) wifi:<ba-del>idx
18:11:33.647 -> I (19255) wifi:new:<9,0>, old:<9,0>, ap:<255,255>, sta:<9,0>, prof:1
18:11:33.647 -> [ 19246][V][WiFiGeneric.cpp:290] _arduino_event_cb(): STA Disconnected: SSID: MyWifi1234, BSSID: 2c:91:ab:4b:77:2d, Reason: 34
18:11:33.647 -> [ 19257][D][WiFiGeneric.cpp:795] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
18:11:33.701 -> [ 19264][W][WiFiGeneric.cpp:816] _eventCallback(): Reason: 34 - STA_STOP
[... 53 lines cut ...]
18:20:17.356 -> [  3992][D][WiFiGeneric.cpp:795] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
18:20:17.356 -> [  3994][D][WiFiGeneric.cpp:844] _eventCallback(): STA IP: 192.168.178.51, MASK: 255.255.255.0, GW: 192.168.178.1
18:20:17.403 -> I (4059) wifi:AP's beacon interval = 102400 us, DTIM period = 1
18:20:17.804 -> .
18:20:17.804 -> WiFi connected
18:20:17.804 -> IP address: 
18:20:17.857 -> 192.168.178.51
18:20:17.857 -> Attempting MQTT connection...I (6487) wifi:state: run -> init (22c0)
18:20:19.812 -> I (6487) wifi:pm stop, total sleep time: 0 us / 2519191 us
18:20:19.812 -> 
18:20:19.812 -> W (6488) wifi:<ba-del>idx
18:20:19.865 -> I (6488) wifi:new:<9,0>, old:<9,0>, ap:<255,255>, sta:<9,0>, prof:1
18:20:19.865 -> [  6484][V][WiFiGeneric.cpp:290] _arduino_event_cb(): STA Disconnected: SSID: MyWifi1234, BSSID: 2c:91:ab:4b:77:2d, Reason: 34
18:20:19.865 -> [  6495][D][WiFiGeneric.cpp:795] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
18:20:19.865 -> [  6502][W][WiFiGeneric.cpp:816] _eventCallback(): Reason: 34 - STA_STOP
[... cut ...]
18:20:24.886 -> Attempting MQTT connection...[ 11522][E][WiFiClient.cpp:240] connect(): connect on fd 54, errno: 118, "Host is unreachable"
18:20:24.886 -> failed, rc=-2 try again in 5 seconds
18:20:29.853 -> Attempting MQTT connection...[ 16525][E][WiFiClient.cpp:240] connect(): connect on fd 54, errno: 118, "Host is unreachable"
18:20:29.906 -> failed, rc=-2 try again in 5 seconds
18:20:34.872 -> Attempting MQTT connection...[ 21528][E][WiFiClient.cpp:240] connect(): connect on fd 54, errno: 118, "Host is unreachable"
18:20:34.872 -> failed, rc=-2 try again in 5 seconds

```

me-no-dev, 2020-11-15:
Thanks for going through it :) I went through the PubSub source as well and did not find code that would interact with WiFi. I'll poke our internal engineers. They might have a clue.

BavariaESP32, 2020-11-16:
Hello me-no-dev,
Thanks for having a look into that. Meanwhile I have tried to exclude other side-effects. So I setup up from scratch (in a new Linux VM under Win10) - the Arduino IDE with idf-release/v4.2 branch. But I got the same result as before.

Maybe an other interesting observation is that when I use the idf-release/v4.2 branch and try to compile and run the sketch one my ESP32 board (Node32s) - I got a quite similar erroneous behavior than on the ESP32-S2 board. (see below – log)

I guess the ESP32-targets in the idf-release/v4.2 are expected to work fine, right? (not like in esp32s2 branch).
If so – then it seems there might be a general issue with Wifi with ESP32/ESP32-S2 boards in current idf-release/v4.2 branch?
Maybe I should also mention that my router/access point is a AVM Fritzbox ... (I just want to mention that because this type of router causes some other issues in the past  with ESP devices).

### Log from ESP32 dev board (Node32s) with idf-release/v4.2
```
18:43:59.635 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
18:43:59.635 -> configsip: 0, SPIWP:0xee
18:43:59.635 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
18:43:59.635 -> mode:DIO, clock div:1
18:43:59.635 -> load:0x3fff0030,len:4
18:43:59.635 -> load:0x3fff0034,len:1576
18:43:59.635 -> load:0x40078000,len:12700
18:43:59.635 -> load:0x40080400,len:3040
18:43:59.635 -> entry 0x400805e4
18:44:00.479 -> 
18:44:00.479 -> Connecting to MyWifi1234
18:44:00.479 -> I (549) wifi:wifi driver task: 3ffd6f98, prio:23, stack:3584, core=0
18:44:00.526 -> I (1351) wifi:wifi firmware version: d5c1f19
18:44:00.526 -> I (1352) wifi:wifi certification version: v7.0
18:44:00.526 -> I (1352) wifi:config NVS flash: enabled
18:44:00.526 -> I (1352) wifi:config nano formating: disabled
18:44:00.526 -> I (1357) wifi:Init data frame dynamic rx buffer num: 32
18:44:00.526 -> I (1361) wifi:Init management frame dynamic rx buffer num: 32
18:44:00.526 -> I (1367) wifi:Init management short buffer num: 32
18:44:00.526 -> I (1372) wifi:Init static tx buffer num: 8
18:44:00.526 -> I (1376) wifi:Init tx cache buffer num: 16
18:44:00.573 -> I (1379) wifi:Init static rx buffer size: 1600
18:44:00.573 -> I (1383) wifi:Init static rx buffer num: 8
18:44:00.573 -> I (1387) wifi:Init dynamic rx buffer num: 32
18:44:00.620 -> I (1477) wifi:mode : sta (30:ae:a4:45:6c:e0)
18:44:00.620 -> I (1477) wifi:enable tsf
18:44:00.620 -> I (1478) wifi:Set ps type: 1
18:44:00.667 -> 
18:44:01.135 -> ....I (3543) wifi:new:<9,0>, old:<1,0>, ap:<255,255>, sta:<9,0>, prof:1
18:44:03.152 -> I (3972) wifi:state: init -> auth (b0)
18:44:03.152 -> I (3980) wifi:state: auth -> assoc (0)
18:44:03.152 -> I (3988) wifi:state: assoc -> run (10)
18:44:03.152 -> .I (4018) wifi:connected with MyWifi1234, aid = 2, channel 9, BW20, bssid = 2c:91:ab:4b:77:2d
18:44:03.199 -> I (4018) wifi:security: WPA2-PSK, phy: bgn, rssi: -45
18:44:03.199 -> I (4024) wifi:pm start, type: 1
18:44:03.199 -> 
18:44:03.246 -> I (4063) wifi:AP's beacon interval = 102400 us, DTIM period = 1
18:44:03.667 -> .
18:44:03.667 -> WiFi connected
18:44:03.667 -> IP address: 
18:44:03.667 -> 192.168.178.102
18:44:03.667 -> Attempting MQTT connection...I (5522) wifi:state: run -> init (22c0)
18:44:04.699 -> I (5523) wifi:pm stop, total sleep time: 706302 us / 1498511 us
18:44:04.699 -> 
18:44:04.699 -> W (5523) wifi:<ba-del>idx
18:44:04.699 -> I (5523) wifi:new:<9,0>, old:<9,0>, ap:<255,255>, sta:<9,0>, prof:1
18:44:04.699 -> failed, rc=-2 try again in 5 seconds
18:44:09.715 -> Attempting MQTT connection...failed, rc=-2 try again in 5 seconds
18:44:14.685 -> Attempting MQTT connection...failed, rc=-2 try again in 5 seconds

```

lbernstone, 2020-11-16:
Can you throw `Serial.printf("Free: %d\tMaxAlloc: %d\t PSFree: %d\n",ESP.getFreeHeap(),ESP.getMaxAllocHeap(),ESP.getFreePsram())` in your code in a few places to monitor memory usage?  I have seen some weirdness with esp32s2 wifi when memory gets critical.

BavariaESP32, 2020-11-18:
Hi lbernstone,
Good idea to log the memory consumption while running the program.
Although I would not expect any critical memory level at this very simple sample program. 
But please have a look at the log below:

### Log from ESP32-S2 (IDFv4.2 branch) with RAM consumption logging:
```
18:14:30.755 -> ESP-ROM:esp32s2-rc4-20191025
18:14:30.755 -> Build:Oct 25 2019
18:14:30.755 -> rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
18:14:30.755 -> SPIWP:0xee
18:14:30.755 -> mode:DIO, clock div:1
18:14:30.755 -> load:0x3ffe6100,len:0x8
18:14:30.755 -> load:0x3ffe6108,len:0x620
18:14:30.809 -> load:0x4004c000,len:0xa38
18:14:30.809 -> load:0x40050000,len:0x2848
18:14:30.809 -> entry 0x4004c190
18:14:31.512 -> 
18:14:31.512 -> Free: 195715	MaxAlloc: 131060	 PSFree: 0
18:14:31.512 -> 
18:14:31.512 -> Connecting to Gartenhaus1
18:14:31.512 -> I (1284) wifi:wifi driver task: 3ffd3c50, prio:23, stack:3584, core=0
18:14:31.558 -> I (1288) wifi:wifi firmware version: d5c1f19
18:14:31.558 -> I (1288) wifi:wifi certification version: v7.0
18:14:31.558 -> I (1288) wifi:config NVS flash: enabled
18:14:31.558 -> I (1291) wifi:config nano formating: disabled
18:14:31.558 -> I (1295) wifi:Init data frame dynamic rx buffer num: 32
18:14:31.558 -> I (1300) wifi:Init management frame dynamic rx buffer num: 32
18:14:31.558 -> I (1305) wifi:Init management short buffer num: 32
18:14:31.558 -> I (1310) wifi:Init static tx buffer num: 8
18:14:31.558 -> I (1314) wifi:Init tx cache buffer num: 16
18:14:31.558 -> I (1318) wifi:Init static rx buffer size: 2212
18:14:31.612 -> I (1322) wifi:Init static rx buffer num: 8
18:14:31.612 -> I (1325) wifi:Init dynamic rx buffer num: 32
18:14:31.712 -> I (1447) wifi:mode : sta (7c:df:a1:01:a1:1c)
18:14:31.712 -> I (1447) wifi:enable tsf
18:14:31.712 -> [  1433][V][WiFiGeneric.cpp:273] _arduino_event_cb(): STA Started
18:14:31.712 -> [  1434][D][WiFiGeneric.cpp:795] _eventCallback(): Arduino Event: 2 - STA_START
18:14:31.712 -> I (1455) wifi:Set ps type: 0
18:14:31.712 -> 
18:14:31.759 -> [  1492][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
18:14:31.759 -> Free: 110339	MaxAlloc: 65524	 PSFree: 0
18:14:32.261 -> ....I (3557) wifi:new:<9,0>, old:<1,0>, ap:<255,255>, sta:<9,0>, prof:1
18:14:34.222 -> I (3947) wifi:state: init -> auth (b0)
18:14:34.222 -> I (3957) wifi:state: auth -> assoc (0)
18:14:34.222 -> I (3970) wifi:state: assoc -> run (10)
18:14:34.269 -> I (4002) wifi:connected with Gartenhaus1, aid = 3, channel 9, BW20, bssid = 2c:91:ab:4b:77:2d
18:14:34.269 -> I (4003) wifi:security: WPA2-PSK, phy: bgn, rssi: -38
18:14:34.269 -> I (4006) wifi:pm start, type: 0
18:14:34.269 -> 
18:14:34.269 -> [  3993][V][WiFiGeneric.cpp:285] _arduino_event_cb(): STA Connected: SSID: Gartenhaus1, BSSID: 2c:91:ab:4b:77:2d, Channel: 9, Auth: UNKNOWN
18:14:34.269 -> [  4005][D][WiFiGeneric.cpp:795] _eventI (4023) wifi:AP's beacon interval = 102400 us, DTIM period = 1
18:14:34.269 -> Callback(): Arduino Event: 4 - STA_CONNECTED
18:14:34.323 -> .[  4034][V][WiFiGeneric.cpp:295] _arduino_event_cb(): STA Got New IP:192.168.178.51
18:14:34.323 -> [  4034][D][WiFiGeneric.cpp:795] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
18:14:34.323 -> [  4037][D][WiFiGeneric.cpp:844] _eventCallback(): STA IP: 192.168.178.51, MASK: 255.255.255.0, GW: 192.168.178.1
18:14:34.771 -> .
18:14:34.771 -> WiFi connected
18:14:34.771 -> IP address: 
18:14:34.771 -> 192.168.178.51
18:14:34.771 -> 
18:14:34.771 -> Free: 109655	MaxAlloc: 65524	 PSFree: 0
18:14:34.771 -> Attempting MQTT connection...
18:14:34.825 -> Free: 109655	MaxAlloc: 65524	 PSFree: 0
18:14:36.679 -> I (6446) wifi:state: run -> init (22c0)
18:14:36.679 -> I (6446) wifi:pm stop, total sleep time: 0 us / 2438699 us
18:14:36.733 -> 
18:14:36.733 -> W (6446) wifi:<ba-del>idx
18:14:36.733 -> I (6447) wifi:new:<9,0>, old:<9,0>, ap:<255,255>, sta:<9,0>, prof:1
18:14:36.733 -> [  6438][V][WiFiGeneric.cpp:290] _arduino_event_cb(): STA Disconnected: SSID: Gartenhaus1, BSSID: 2c:91:ab:4b:77:2d, Reason: 34
18:14:36.733 -> [  6449][D][WiFiGeneric.cpp:795] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
18:14:36.733 -> [  6456][W][WiFiGeneric.cpp:816] _eventCallback(): Reason: 34 - STA_STOP
18:14:36.733 -> [  6464][E][WiFiClient.cpp:266] connect(): socket error on fd 54, errno: 113, "Software caused connection abort"
18:14:36.733 -> failed, rc=-2 try again in 5 seconds
18:14:41.752 -> Attempting MQTT connection...
18:14:41.752 -> Free: 110051	MaxAlloc: 65524	 PSFree: 0
18:14:41.752 -> [ 11476][E][WiFiClient.cpp:240] connect(): connect on fd 54, errno: 118, "Host is unreachable"
18:14:41.752 -> failed, rc=-2 try again in 5 seconds
18:14:46.773 -> Attempting MQTT connection...
18:14:46.773 -> Free: 110051	MaxAlloc: 65524	 PSFree: 0
18:14:46.773 -> [ 16482][E][WiFiClient.cpp:240] connect(): connect on fd 54, errno: 118, "Host is unreachable"

```

BavariaESP32, 2020-11-23:
Hi all.. any news here?

**Recap**: Tests are made with idf-release/v4.2
Boards used: ESP32-S2-WROVER and ESP32 (WROOM – Node32s)
The issue exits on both boards.
The issue is not only related to Wifi in connection with MQTT (PubSubClient)

I have made some more tests with other scenarios getting some data via Wifi and „working“ with them in the program code later on… all of them failed (on ESP32-S2 and ESP32) when using the idf-release/v4.2 branch.
When using instead the „master“ branch (only with ESP32 possible) the program works fine.

Here is one additional – very simple – example – it’s taken from the „samples“ provided by the Arduino IDE (see File->Examples→ESP32→ time→Simple time).
I have only added a few lines for tracing and removed the disabling of the Wifi after succesful connection to a NTP-server.

Also here – with the idf-release/v4.2 the Wifi-connection seems to be establish but no “user data” might be exchanged over it.

Please fine here the code: 
```c
	#include <WiFi.h>
	#include "time.h"
	#include "esp32-hal-log.h"

	const char* ssid       = "YOUR_SSID";
	const char* password   = "YOUR_PASSWORD";

	const char* ntpServer = "pool.ntp.org";
	const long  gmtOffset_sec = 3600;
	const int   daylightOffset_sec = 3600;

	void printLocalTime()
	{
	  struct tm timeinfo;
	  if(!getLocalTime(&timeinfo)){
		Serial.println("Failed to obtain time");
		return;
	  }
	  Serial.println(&timeinfo, "%A, %B %d %Y %H:%M:%S");
	}

	void setup()
	{
	  Serial.begin(115200);
	   esp_log_level_set("wifi", ESP_LOG_VERBOSE);      // enable WARN logs from WiFi stack
	   esp_log_level_set("*", ESP_LOG_ERROR);        // set all components to ERROR level

	  
	  //connect to WiFi
	  Serial.printf("Connecting to %s ", ssid);
	  WiFi.disconnect(true);
	  WiFi.mode(WIFI_OFF);
	  delay(500);
	  WiFi.begin(ssid, password);
	  while (WiFi.status() != WL_CONNECTED) {
		  delay(500);
		  Serial.print(".");
	  }
	  Serial.println("");
	  Serial.println("WiFi connected");
	  Serial.println("IP address: ");
	  Serial.print(WiFi.localIP());
	  Serial.println("");
	  
	  //init and get the time
	  Serial.println(" Try to connect to NTP-server and get time  ");
	  configTime(gmtOffset_sec, daylightOffset_sec, ntpServer);

	  printLocalTime();

	  //disconnect WiFi - not active for this special test 
	  //WiFi.disconnect(true);
	  //WiFi.mode(WIFI_OFF);
	}

	void loop()
	{
	  delay(1000);
	  printLocalTime();
	}
```

### The log when idf-relase/v4.2 is used
```
17:16:12.623 -> ESP-ROM:esp32s2-rc4-20191025
17:16:12.623 -> Build:Oct 25 2019
17:16:12.623 -> rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
17:16:12.623 -> SPIWP:0xee
17:16:12.623 -> mode:DIO, clock div:1
17:16:12.623 -> load:0x3ffe6100,len:0x8
17:16:12.623 -> load:0x3ffe6108,len:0x620
17:16:12.623 -> load:0x4004c000,len:0xa38
17:16:12.623 -> load:0x40050000,len:0x2848
17:16:12.670 -> entry 0x4004c190
17:16:12.870 -> Connecting to MyWifi1234 [  1377][V][WiFiGeneric.cpp:273] _arduino_event_cb(): STA Started
17:16:13.525 -> [  1378][D][WiFiGeneric.cpp:795] _eventCallback(): Arduino Event: 2 - STA_START
17:16:13.572 -> [  1428][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
17:16:14.074 -> ....[  3872][V][WiFiGeneric.cpp:285] _arduino_event_cb(): STA Connected: SSID: MyWifi1234, BSSID: 2c:91:ab:4b:77:2d, Channel: 9, Auth: UNKNOWN
17:16:16.033 -> [  3873][D][WiFiGeneric.cpp:795] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
17:16:16.033 -> [  3896][V][WiFiGeneric.cpp:295] _arduino_event_cb(): STA Got New IP:192.168.178.51
17:16:16.033 -> [  3897][D][WiFiGeneric.cpp:795] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
17:16:16.033 -> [  3900][D][WiFiGeneric.cpp:844] _eventCallback(): STA IP: 192.168.178.51, MASK: 255.255.255.0, GW: 192.168.178.1
17:16:16.080 -> .
17:16:16.080 -> WiFi connected
17:16:16.080 -> IP address: 
17:16:16.080 -> 192.168.178.51
17:16:16.080 ->  Try to connect to NTP-server and get time  
17:16:20.545 -> [  8429][V][WiFiGeneric.cpp:290] _arduino_event_cb(): STA Disconnected: SSID: MyWifi1234, BSSID: 2c:91:ab:4b:77:2d, Reason: 34
17:16:20.592 -> [  8429][D][WiFiGeneric.cpp:795] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
17:16:20.592 -> [  8437][W][WiFiGeneric.cpp:816] _eventCallback(): Reason: 34 - STA_STOP
17:16:21.046 -> Failed to obtain time
17:16:27.066 -> Failed to obtain time
```

### The log when "master" branch is use (with ESP32)  - which works as expected
```
17:31:46.453 -> ets Jun  8 2016 00:22:57
17:31:46.453 -> 
17:31:46.453 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
17:31:46.453 -> configsip: 0, SPIWP:0xee
17:31:46.453 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
17:31:46.453 -> mode:DIO, clock div:1
17:31:46.453 -> load:0x3fff0018,len:4
17:31:46.453 -> load:0x3fff001c,len:928
17:31:46.453 -> ho 0 tail 12 room 4
17:31:46.453 -> load:0x40078000,len:10080
17:31:46.453 -> load:0x40080400,len:5856
17:31:46.453 -> entry 0x400806a8
17:31:46.708 -> Connecting to MyWifi1234 [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 0 - WIFI_READY
17:31:47.357 -> [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 2 - STA_START
17:31:47.557 -> [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 4 - STA_CONNECTED
17:31:47.557 -> [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 7 - STA_GOT_IP
17:31:47.557 -> [D][WiFiGeneric.cpp:418] _eventCallback(): STA IP: 192.168.178.102, MASK: 255.255.255.0, GW: 192.168.178.1
17:31:47.858 -> .
17:31:47.858 -> WiFi connected
17:31:47.858 -> IP address: 
17:31:47.858 -> 192.168.178.102
17:31:47.858 ->  Try to connect to NTP-server and get time  
17:31:47.911 -> Monday, November 23 2020 17:31:47
17:31:48.914 -> Monday, November 23 2020 17:31:48
```

lbernstone, 2020-11-23:
I am unable to reproduce, which leads me to believe this is something in your environment.
1) Try to [erase the flash](https://github.com/espressif/esptool#erase-flash-erase_flash--erase-region) on your esp32s2 and confirm that you have the error from a clean start.
2) What is your AP?  This is the most likely source of an issue that affects both esp32 and esp32s2.

BavariaESP32, 2020-11-23:
Hi Ibernstone,
Thanks for the hints. I have good :smiley: and bad :worried: news.

to 1) I tried to erase the flash and run the rest as before - no change - doesn't work

to 3) I added the WiFi powersave off command after connection established - no change - doesn't work

to 2) My AP is an AVM FritzBox (7590) - especially in Germany the most poplar DSL-modem/router - 
        Then I tried to open an AP from my mobile phone - and there the NTP-server example works fine even with the ESP32-S2.
        So it seems to be a problem when using the FritzBox and an ESP32(-S2) when using IDF/v4.2
        When you have a look into forums, it's not the first time ESP32 (and ESP8226) having problems with Fritzboxes as APs.
        Nevertheless when using the "master" branch on ESP32 this kind of problems are not exiting. 

An idea how to solve this problem? 
Thanks

closed 2021-02-07 by stale[bot]