espressif/arduino-esp32 #4546: SerialToSerialBT Example crash with 1.0.5 RC1 & RC2

Cjcr, 2020-11-17:
Board: ESP32 Dev Module
ESP32: 1.0.5-rc2 (also rc1)

Using Example file "SerialToSerialBT.ino", it **crash everytime**, the only way to fix it is removing the "SerialBT.begin("ESP32");" line, but of course, all don't work then.

```
4:38:03.944 -> Rebooting...
14:38:03.944 -> ets Jun  8 2016 00:22:57
14:38:03.944 -> 
14:38:03.944 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
14:38:03.944 -> configsip: 0, SPIWP:0xee
14:38:03.944 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
14:38:03.991 -> mode:DIO, clock div:1
14:38:03.991 -> load:0x3fff0018,len:4
14:38:03.991 -> load:0x3fff001c,len:1100
14:38:03.991 -> load:0x40078000,len:10900
14:38:03.991 -> load:0x40080400,len:6388
14:38:03.991 -> entry 0x400806b4
14:38:04.977 -> Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
14:38:04.977 -> Core 0 register dump:
14:38:04.977 -> PC      : 0x400def30  PS      : 0x00060730  A0      : 0x800df5d1  A1      : 0x3ffd01b0  
14:38:04.977 -> A2      : 0x3ffdeaec  A3      : 0x3ffd0228  A4      : 0x00000001  A5      : 0x00000001  
14:38:05.023 -> A6      : 0x00060920  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x3ffd01b0  
14:38:05.023 -> A10     : 0x00000003  A11     : 0x00060923  A12     : 0x00060920  A13     : 0x3ffd0230  
14:38:05.023 -> A14     : 0x000f1400  A15     : 0x3ffdeaec  SAR     : 0x0000001e  EXCCAUSE: 0x0000001c  
14:38:05.023 -> EXCVADDR: 0x0000002c  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  
14:38:05.023 -> 
14:38:05.023 -> ELF file SHA256: 0000000000000000
14:38:05.023 -> 
14:38:05.023 -> Backtrace: 0x400def30:0x3ffd01b0 0x400df5ce:0x3ffd0200 0x400d5969:0x3ffd0220 0x4008f49e:0x3ffd0250

```

**Decoded:**

```
Decoding stack results
0x400def30: queue_send_wrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bt.c line 701
0x400df5ce: esp_bt_sleep_disable at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bt.c line 1475
0x400d5969: get_command(String) at Z:\SYNC\arduino_esp32_rc\portable\packages\esp32\hardware\esp32\1.0.5-rc2\cores\esp32/WString.h line 325
0x4008f49e: xEventGroupSetBits at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/event_groups.c line 560
```

It works well with 1.0.4 stable.

praisedpern, 2020-11-21:
Hi, I'm also having this issue with 1.0.5-rc2 and Arduino IDE 1.8.13. The board I'm using is a LilyGO TTGO T-Display (TTGO Lora32-OLED V1 selected on Tools > Board menu).

```
16:12:01.712 -> Rebooting...
16:12:01.759 -> ets Jun  8 2016 00:22:57
16:12:01.759 -> 
16:12:01.759 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
16:12:01.759 -> configsip: 0, SPIWP:0xee
16:12:01.759 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
16:12:01.759 -> mode:DIO, clock div:1
16:12:01.759 -> load:0x3fff0018,len:4
16:12:01.759 -> load:0x3fff001c,len:928
16:12:01.759 -> ho 0 tail 12 room 4
16:12:01.759 -> load:0x40078000,len:10080
16:12:01.759 -> load:0x40080400,len:5856
16:12:01.759 -> entry 0x400806a8
16:12:02.602 -> The device started, now you can pair it with bluetooth!
16:12:02.602 -> Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
16:12:02.602 -> Core 0 register dump:
16:12:02.602 -> PC      : 0x400def30  PS      : 0x00060730  A0      : 0x800df5d1  A1      : 0x3ffd00f0  
16:12:02.602 -> A2      : 0x3ffdea40  A3      : 0x3ffd0168  A4      : 0x00000001  A5      : 0x00000001  
16:12:02.602 -> A6      : 0x00060920  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x3ffd00f0  
16:12:02.602 -> A10     : 0x00000003  A11     : 0x00060923  A12     : 0x00060920  A13     : 0x3ffd0170  
16:12:02.649 -> A14     : 0x000f1400  A15     : 0x3ffdea40  SAR     : 0x0000001a  EXCCAUSE: 0x0000001c  
16:12:02.649 -> EXCVADDR: 0x0000002c  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  
16:12:02.649 -> 
16:12:02.649 -> ELF file SHA256: 0000000000000000
16:12:02.649 -> 
16:12:02.649 -> Backtrace: 0x400def30:0x3ffd00f0 0x400df5ce:0x3ffd0140 0x400d5969:0x3ffd0160 0x4008f49e:0x3ffd0190
16:12:02.649 ->
```

Cjcr, 2020-11-22:
> Hi, I'm also having this issue with 1.0.5-rc2 and Arduino IDE 1.8.13. The board I'm using is a LilyGO TTGO T-Display (TTGO Lora32-OLED V1 selected on Tools > Board menu).
>
> ```
> 16:12:01.712 -> [...]

You are using Bluetooth too when it fails?

praisedpern, 2020-11-22:
@Cjcr yes. I'm attempting to use the same SerialToSerialBT example sketch. The ESP32 crashes and reboots in one or two seconds, too soon for me to try and connect from another device.

me-no-dev, 2020-11-22:
I found the problem. Its here: https://github.com/espressif/esp-idf/blob/release/v3.3/components/bt/bluedroid/btc/profile/std/spp/btc_spp.c#L366

When dynamic memory is used, this `if` get's called before the structure is actually allocated. It's a bug, but what can we do...
- If we disable dynamic memory for BT, any sketch that uses BT will automatically have less available memory
- If I fix it in IDF, not sure how soon it will land on github

What should I do? BTW if you switch to github and use the `idf-release/v4.2` branch, BT is working fine.

Cjcr, 2020-11-26:
@me-no-dev Using idf-release/v4.2 give me this error:

```
Z:\SYNC\arduino-1.8.13\portable\sketchbook\hardware\espressif\esp32/tools/sdk/esp32/include/esp32/include/rom/rtc.h:1:2: warning: #warning rom/rtc.h is deprecated, please use esp32/rom/rtc.h instead [-Wcpp]
 #warning rom/rtc.h is deprecated, please use esp32/rom/rtc.h instead
  ^~~~~~~
In file included from Z:\SYNC\arduino-1.8.13\portable\sketchbook\libraries\OneWire-master\OneWire.cpp:144:
Z:\SYNC\arduino-1.8.13\portable\sketchbook\libraries\OneWire-master\util/OneWire_direct_gpio.h: In function 'void directModeInput(uint32_t)':
Z:\SYNC\arduino-1.8.13\portable\sketchbook\libraries\OneWire-master\util/OneWire_direct_gpio.h:161:26: error: 'rtc_gpio_desc' was not declared in this scope
         uint32_t rtc_reg(rtc_gpio_desc[pin].reg);
                          ^~~~~~~~~~~~~
Z:\SYNC\arduino-1.8.13\portable\sketchbook\libraries\OneWire-master\util/OneWire_direct_gpio.h:161:26: note: suggested alternative: 'rtc_io_desc'
         uint32_t rtc_reg(rtc_gpio_desc[pin].reg);
                          ^~~~~~~~~~~~~
                          rtc_io_desc
Z:\SYNC\arduino-1.8.13\portable\sketchbook\libraries\OneWire-master\util/OneWire_direct_gpio.h: In function 'void directModeOutput(uint32_t)':
Z:\SYNC\arduino-1.8.13\portable\sketchbook\libraries\OneWire-master\util/OneWire_direct_gpio.h:189:26: error: 'rtc_gpio_desc' was not declared in this scope
         uint32_t rtc_reg(rtc_gpio_desc[pin].reg);
                          ^~~~~~~~~~~~~
Z:\SYNC\arduino-1.8.13\portable\sketchbook\libraries\OneWire-master\util/OneWire_direct_gpio.h:189:26: note: suggested alternative: 'rtc_io_desc'
         uint32_t rtc_reg(rtc_gpio_desc[pin].reg);
                          ^~~~~~~~~~~~~
                          rtc_io_desc
exit status 1
Error compilando para la tarjeta ESP32 Dev Module.

```

If I use idf-release/v4.0 it works without problem (seems).

lbernstone, 2020-11-26:
OneWire is broken, not the core.  Do as it recommends, and rename rtc_gpio_desc to rtc_io_desc, and OneWire will work.

Cjcr, 2020-11-26:
> OneWire is broken, not the core. Do as it recommends, and rename rtc_gpio_desc to rtc_io_desc, and OneWire will work in v4.2.

I did that, but this happens:

```
Decoding stack results
0x400d89cd: OneWire::reset() at Z:\SYNC\arduino-1.8.13\portable\sketchbook\libraries\OneWire-master\util/OneWire_direct_gpio.h line 165
0x400d884e: DallasTemperature::requestTemperatures() at Z:\SYNC\arduino-1.8.13\portable\sketchbook\libraries\DallasTemperature\DallasTemperature.cpp line 404
0x400d780f: loop() at xxxx.ino line 800
0x400dbfcd: loopTask(void*) at Z:\SYNC\arduino-1.8.13\portable\sketchbook\hardware\espressif\esp32\cores\esp32\main.cpp line 35

```

```
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:4
load:0x3fff0034,len:1576
load:0x40078000,len:12700
load:0x40080400,len:3040
entry 0x400805e4
0:00:00:01:625: ############ DH v0.93 ############
0:00:00:01:626: SDK_VERSION: v4.3-dev-1720-g494a124d9-dirty
0:00:00:01:626: RESET(CPU_0): Software reset CPU
0:00:00:01:630: RESET(CPU_1): Software reset CPU
0:00:00:03:636: turnOn_phone()
0:00:00:08:838: HC05 = 
```

lbernstone, 2020-11-26:
This is not the support forum for that library.  It has not been updated to support the beta code.

me-no-dev, 2020-12-01:
can any of you try the current master and confirm that the issue is gone?

praisedpern, 2020-12-01:
Using 1.0.5 RC3 my ESP32 is crashing and rebooting when running this sketch, but looks like a different error this time:

```
15:36:49.461 -> Rebooting...
15:36:49.461 -> ets Jun  8 2016 00:22:57
15:36:49.461 -> 
15:36:49.461 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
15:36:49.461 -> configsip: 0, SPIWP:0xee
15:36:49.461 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
15:36:49.461 -> mode:DIO, clock div:1
15:36:49.461 -> load:0x3fff0018,len:4
15:36:49.461 -> load:0x3fff001c,len:928
15:36:49.461 -> ho 0 tail 12 room 4
15:36:49.461 -> load:0x40078000,len:10080
15:36:49.461 -> load:0x40080400,len:5856
15:36:49.461 -> entry 0x400806a8
15:36:50.352 -> Guru Meditation Error: Core  0 panic'ed (LoadStoreError). Exception was unhandled.
15:36:50.352 -> Core 0 register dump:
15:36:50.352 -> PC      : 0x40114b89  PS      : 0x00060b30  A0      : 0x800defa4  A1      : 0x3ffd0090  
15:36:50.352 -> A2      : 0x400deff8  A3      : 0x3ffbdc68  A4      : 0x3f407068  A5      : 0x00001400  
15:36:50.352 -> A6      : 0x00001800  A7      : 0x00000003  A8      : 0x80114b89  A9      : 0x3ffd0040  
15:36:50.352 -> A10     : 0x3ffe0464  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x00001800  
15:36:50.352 -> A14     : 0x00000000  A15     : 0x3ffcdfe0  SAR     : 0x00000010  EXCCAUSE: 0x00000003  
15:36:50.398 -> EXCVADDR: 0x3f40706c  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  
15:36:50.398 -> 
15:36:50.398 -> ELF file SHA256: 0000000000000000
15:36:50.398 -> 
15:36:50.398 -> Backtrace: 0x40114b89:0x3ffd0090 0x400defa1:0x3ffd00b0 0x400df5f2:0x3ffd0100 0x400d5969:0x3ffd0120 0x4008f49e:0x3ffd0150
15:36:50.398 -> 

```

me-no-dev, 2020-12-01:
can you decode that backtrace please?

me-no-dev, 2020-12-01:
got it!

praisedpern, 2020-12-01:
Thanks @me-no-dev 

Here in case anyone else needs it:

```
Decoding stack results
0x40114b89: BTA_JvEnable at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/bta/jv/bta_jv_api.c line 75
0x400defa1: btc_spp_init at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/btc/profile/std/spp/btc_spp.c line 390
0x400df5f2: btc_spp_call_handler at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/btc/profile/std/spp/btc_spp.c line 635
0x400d5969: btc_task at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/common/btc/core/btc_task.c line 160
0x4008f49e: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143

```

me-no-dev, 2020-12-01:
the issue was found :) will push update as soon as it makes it to IDF v3.3 branch

closed 2021-02-16 by stale[bot]