espressif/arduino-esp32 #4549: SDMMC host driver does't work in Slot 0 mode

vilyansky, 2020-11-18:
Hello, I am trying to mount sd card in slot0 mode, following this https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/sdmmc_host.html#_CPPv420sdmmc_host_init_slotiPK19sdmmc_slot_config_t

And It seems that this library does't support setup this mode in 
```c
SDMMCFS::begin(const char * mountpoint, bool mode1bit)
```
Next I hardcode in library https://github.com/espressif/arduino-esp32/blob/master/libraries/SD_MMC/src/SD_MMC.cpp
```c
        .slot = SDMMC_HOST_SLOT_0,
```

### Hardware:
Board:							ESP-WROOM-32 module
Core Installation version:			1.0.4
IDE name:						Arduino IDE
Flash Frequency:					40Mhz
PSRAM enabled:                                    ?no? ?yes?
Upload Speed:					115200
Computer OS:                                        Ubuntu

### Description:
And this provide a cyclic reboot with the debug message below

### Sketch:
```cpp

#include <Arduino.h>

void setup() {
    Serial.begin(115200);
    if(!SD_MMC.begin()){
        Serial.println("Card Mount Failed");
        return;
    }
}

void loop() {
}
```

### Debug Messages:
```
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400f2be0: 00000000 ff6f4400 ffffffff
Core 1 register dump:
PC      : 0x400f2be7  PS      : 0x00060530  A0      : 0x800f35c6  A1      : 0x3ffb1e40  
A2      : 0x00013ffc  A3      : 0x00000000  A4      : 0x400f2a30  A5      : 0x3ffb84ec  
A6      : 0x3ffc12f0  A7      : 0x00000000  A8      : 0x00049060  A9      : 0x00000b00  
A10     : 0x3ff49060  A11     : 0x00000100  A12     : 0x3ffb86c4  A13     : 0x3ffb86ac  
A14     : 0x00000004  A15     : 0x3ffb84ec  SAR     : 0x0000001c  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace: 0x400f2be7:0x3ffb1e40 0x400f35c3:0x3ffb1e60 0x400de92c:0x3ffb1eb0 0x400d1f0d:0x3ffb1f00 0x400d1456:0x3ffb1f80 0x400d2a9f:0x3ffb1fb0 0x4008badd:0x3ffb1fd0
```

igrr, 2020-11-18:
@vilyansky do you happen to have SPI flash chip connected to the ESP32 using the default set of pins (IOs 6-11)? Overlapping SPI flash interface and SD card interface will not work, since SPI flash lines are continuously used for code execution (XIP).

It is possible to use SD slot 0 if SPI flash is connected to a different set of pins of the ESP32, though. In that case, the set of SPI flash pins can be programmed by burning eFuses. For example, to use pins CLK=22, Q=19, D=21, CS0=23 you can run:
```
espefuse.py burn_efuse SPI_PAD_CONFIG_CLK 22
espefuse.py burn_efuse SPI_PAD_CONFIG_Q 19
espefuse.py burn_efuse SPI_PAD_CONFIG_D 21
espefuse.py burn_efuse SPI_PAD_CONFIG_CS0 23
```
(careful, though, this is a one-time operation.)

vilyansky, 2020-11-18:
Thanks, igrr.
Yes, I have connected using default sets of pins 6-11. It is custom designed board.
I wil think about this problem.

vilyansky, 2020-11-18:
Hmm... I have no external SPI flash connected to these pins. Or your about internal connection in WROOM module inside?

igrr, 2020-11-18:
If you are using a WROOM-32 module, it does have an SPI flash chip connected to these pins: https://www.espressif.com/sites/default/files/documentation/esp32-wroom-32_datasheet_en.pdf (see section 6, "Schematics").

vilyansky, 2020-11-18:
And why I can't use slot0 configuration with this library?
Why I have cyclic reboot instead executing this code:
```c
esp_err_t ret = esp_vfs_fat_sdmmc_mount(mountpoint, &host, &slot_config, &mount_config, &_card);
    if (ret != ESP_OK) {
        if (ret == ESP_FAIL) {
            log_e("Failed to mount filesystem. If you want the card to be formatted, set format_if_mount_failed = true.");
        } else if (ret == ESP_ERR_INVALID_STATE) {
            _impl->mountpoint(mountpoint);
            log_w("SD Already mounted");
            return true;
        } else {
            log_e("Failed to initialize the card (%d). Make sure SD card lines have pull-up resistors in place.", ret);
        }
        _card = NULL;
        return false;
    }
```

igrr, 2020-11-18:
The pins used for slot 0 of SDMMC peripheral overlap the pins used for SPI Flash chip.
What happens is when SDMMC peripheral gets initialized, the driver routes the SDMMC signals to/from these pins. After this, SPI peripheral can no longer communicate with SPI flash chip correctly. SPI connection to the flash chip is used by hardware to fetch the code for execution. So as soon as the pins get reconfigured, the hardware can not fetch correct instructions from Flash, which results in an IllegalInstruction exception.

closed 2021-02-02 by stale[bot]