espressif/arduino-esp32 #4551: NEED help in setting up partition Scheme/Table for ESP32-WROOM-32D (8MB)

Electro-boy, 2020-11-18:
Am trying to find partition table for enabling partition scheme with boards.txt.
Am using a 8MB flash version of ESP32-WROOM-32D from digikey.
https://www.digikey.in/product-detail/en/espressif-systems/ESP32-WROOM-32D-8MB/1904-1024-6-ND/9381749

I am trying to increase the APP size to 4MB and SPIFFS to 3MB without OTA.

how to create the partition table for the 8MB FLASH Module.

Please Someone guide me.

I have arrived at these lines for scheme. will it work?

```
# Name,Type,SubType,Offset,Size,Flags
nvs,      data,  nvs,     0x9000,    0x5000	,
phy_init, data,  phy,     0xe000,    0x2000	,
factory,  app,   factory, 0x10000,   0x3500000	,
spiffs,   data,  spiffs,  0x3510000,  0xF0000	,

```

lbernstone, 2020-11-18:
arduino-esp32 includes OTA whether you need it or not.  So, you need to have otadata and ota_0 rather than a factory partition.  Note that the app will calculate offsets if not included. 8MB = 0x800000.  If you save this file as `partitions.csv` in your sketch folder, Arduino IDE will use it instead of the selected partition scheme (but the maximum sketch size still comes from that...)
```
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000, 0x5000,
otadata,  data, ota,           , 0x2000,
app0,     app,  ota_0,         , 0x700000,
spiffs,   data, spiffs,        , 0xF0000,
```

lorol, 2020-11-19:
@lbernstone 
If this is the case (mandatory including OTA), then could you please revise the examples proposed here for correctness?
https://github.com/espressif/arduino-esp32/tree/master/tools/partitions
There are 8M and 16MB templates

lbernstone, 2020-11-19:
All the included partition schemes include otadata. https://docs.espressif.com/projects/esp-idf/en/v3.3.4/api-reference/system/ota.html#ota-data-partition.  If you find the cosmetic discrepancies a problem, you are welcome to fix them.

lorol, 2020-11-19:
Got it, I missed ota-**data** word. All good.

lorol, 2020-11-20:
@lbernstone 
>  If you save this file as partitions.csv in your sketch folder, Arduino IDE will use it instead of the selected partition scheme
I didn't  know this ans it is interesting ... I am testing it also on the Java arduino-esp32fs-plugin and it works well.
Just the Offset and Size values must be present (for now) so the tool can flash properly the data partition, unless offsets calculation is done as well ...

lorol, 2020-11-24:
Hi @lbernstone again.
Slightly off-topic but on identical subject.
I have really hard time to let Arduino IDE under win10 zip installation to "take" the partition.csv from sketch folder.
Really annoying and spent a lot of time.
I see with debug that it copies from sketch to build place according to platform.txt line:
``` 
recipe.hooks.prebuild.1.pattern.windows=cmd /c if exist "{build.source.path}\partitions.csv" copy /y "{build.source.path}\partitions.csv" "{build.path}\partitions.csv"
```
And file is there,
Even later gen_esp32part.exe makes the bin from it.
But at the end, it always gives the default Sketch uses 239044 bytes (18%) of program storage space. Maximum is 1310720 bytes.
The partition.csv table is:
```
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  ota_0,   0x10000, 0x100000,
app1,     app,  ota_1,           ,0x100000,
spiffs,   data, spiffs,          ,0x1F0000,
```
app should not be 1310720 size, right? Any help on this?

lbernstone, 2020-11-24:
As mentioned above, Arduino decides the maximum sketch size from the chosen partition scheme.  It is not looking at the partition table.  Pick a scheme with a very large app size to get around this.

lorol, 2020-11-24:
Actually, the Arduino IDE doesn't seem to really care about partition.csv at sketch folder!
Same as described here: https://github.com/espressif/arduino-esp32/issues/703
You need to modify boards.txt with 3 new entries for your board and add the custom csv together with others, example, I copied my custom partition.csv as my.csv where all others are for Arduino IDE pick-up, and modified the boards.txt:
```
ttgo-t1.menu.PartitionScheme.my=My with OTA (1MB APP/2MB SPIFFS)
ttgo-t1.menu.PartitionScheme.my.build.partitions=my
ttgo-t1.menu.PartitionScheme.my.upload.maximum_size=1048576
```
to get custom partition taken from IDE menu

lorol, 2020-11-24:
I got that ... But why it copies it to build and makes bin anyway?

lorol, 2020-11-24:
Wow! It is very confusing,
Sorry ... after tests it turns out that IDE considers the partition.csv at sketch folder but is really confusing.
IDE is taking the upload.maximum_size from selected menu scheme and what you see for "Maximum is 1310720 bytes." is not exactly the truth ...

jshep321, 2021-01-18:
Hi,
I made a quick reference excel sheet for the ESP32-WROOM (4MB flash) partition tables.  It's quick and dirty, but gives an overview of what is available in the commonly used settings.
[image]

The worksheet:

[calcs for ESP32 partitions.xlsx](https://github.com/espressif/arduino-esp32/files/5831857/calcs.for.ESP32.partitions.xlsx)

capedra, 2021-07-05:
Wait. What if I want to put 4 mb for the app and 4 mb for the ota?

capedra, 2021-07-05:
What about 8 mb for the app and 8 mb for the ota? Does the ESP-32 still have the 2 mb limit for the sketch?

lbernstone, 2021-07-05:
ESP32 has no restriction on the app size.  Arduino will complain if you exceed the partition size as listed by the selected scheme.  This can be changed in boards.txt

aminbaig, 2021-07-26:
Hi all,

I tweaked my esp32 cam module and removed the 4mb flash chip, instead replaced it with a 16mb chip. It is detected and working with the standard 16mb partition scheme, I have been trying to change the partition to have 6mb capacity for ota_0 and OTA_1. that should leave me with 4mb for other needed partitions. 

Can anyone share a partition scheme to achieve that or point me to the right direction?

VojtechBartoska, 2022-03-30:
Hello, we will take a look on this issue and evaluate it soon. Thanks for your contribution.

igrr, 2022-04-05:
It seems that the partition tables can be customized at sketch level. The task here is to document this capability and maybe add an example.

VojtechBartoska, 2022-04-22:
Related issues which can be adressed by Documentation:
- #6495
- #6114
- #5952

zekageri, 2022-04-27:
```
# Name,     Type,   SubType,    Offset, Size,       Flags
nvs,        data,   nvs,        0x9000, 0x5000,
otadata,    data,   ota,        ,   0x2000,
app0,       app,    ota_0,          ,   0x2DC6C0,
app1,       app,    ota_1,          ,   0x2DC6C0,
spiffs,     data,   spiffs,         ,   0x982680,
```

**On sketch upload:**
`WARNING: Partition spiffs not aligned to 0x1000.This is deprecated and will be considered an error in the future release.`
`On flash upload:`
Auto-detected Flash size: 16MB
`WARNING: Flash address 0x005c8d80 is not aligned to a 0x1000 byte flash sector. 0xd80 bytes before this address will be erased.`

What is that?

me-no-dev, 2022-04-27:
means that the last digits of partition addresses and sizes must end on 0x000 instead of 0x6C0

zekageri, 2022-04-27:
I see. Thanks. So it can not calculate the offsets by itself? Or the problem is that i added rounded numbers. 👍 ( i cant find the exact flash size it's says just 16mb :| )

hamza765, 2022-04-27:
@zekageri use binary bytes. So 16M = 16777216 Bytes. In hex, that's 0x1000000

This is included in the repo: https://github.com/espressif/arduino-esp32/blob/master/tools/partitions/large_spiffs_16MB.csv

zekageri, 2022-04-27:
> @zekageri use binary bytes. So 16M = 16777216 Bytes. In hex, that's 0x1000000
> 
> This is included in the repo: https://github.com/espressif/arduino-esp32/blob/master/tools/partitions/large_spiffs_16MB.csv
> 
> 

Yeah, thank you that was obvious.  Iam dumb a.f. thanks

PR #6758 (2022-05-30): [DOCS] Added partition table tutorial
## Description of Change

This PR adds the partition table details and the steps for creating a custom one.

## Tests scenarios

N/A

## Related links

Closes #4551
Closes #6495
closed 2022-05-30 by me-no-dev