espressif/arduino-esp32 #4554: [1.0.5 RC1/RC2] softAP unstable when performing a WiFi.scanNetworks() - Windows 10 clients get disconnected.

gonzabrusco, 2020-11-19:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.4 and 1.0.5 RC1/RC2
IDE name:						Platform.io
Flash Frequency:					40Mhz
PSRAM enabled:                                     no
Upload Speed:					115200
Computer OS:                                        Windows 10 and Android 9

### Description:
softAP gets unstable when performing a wifi scan in latest 1.0.5 RC1 and RC2. In version 1.0.4 this can be done without problems. On 1.0.5 RC1/RC2 the AP goes down while scanning. This happens when I'm connected to the AP with a PC with Windows 10 (I tried  two different laptops with Windows 10 and both have the same problem). Curiously, on two Android phones this worked on both versions.

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
The following is the WiFi AP example with minor modifications for showing scan results. This code compiled against 1.0.4 works perfectly but compiled against 1.0.5 RC1/RC2 does not work.
```
#include <WiFi.h>
#include <WiFiClient.h>
#include <WiFiAP.h>

#define LED_BUILTIN 2   // Set the GPIO pin where you connected your test LED or comment this line out if your dev board has a built-in LED

// Set these to your desired credentials.
const char *ssid = "helloworld";
const char *password = "";

WiFiServer server(80);

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);

  Serial.begin(115200);
  Serial.println();
  Serial.println("Configuring access point...");

  // You can remove the password parameter if you want the AP to be open.
  WiFi.softAP(ssid, password);
  IPAddress myIP = WiFi.softAPIP();
  Serial.print("AP IP address: ");
  Serial.println(myIP);
  server.begin();

  Serial.println("Server started");
}

void loop() {
  WiFiClient client = server.available();   // listen for incoming clients

  if (client) {                             // if you get a client,
    Serial.println("New Client.");           // print a message out the serial port
    String currentLine = "";                // make a String to hold incoming data from the client
    while (client.connected()) {            // loop while the client's connected
      if (client.available()) {             // if there's bytes to read from the client,
        char c = client.read();             // read a byte, then
        Serial.write(c);                    // print it out the serial monitor
        if (c == '\n') {                    // if the byte is a newline character

          // if the current line is blank, you got two newline characters in a row.
          // that's the end of the client HTTP request, so send a response:
          if (currentLine.length() == 0) {
            // HTTP headers always start with a response code (e.g. HTTP/1.1 200 OK)
            // and a content-type so the client knows what's coming, then a blank line:
            client.println("HTTP/1.1 200 OK");
            client.println("Content-type:text/html");
            client.println();

            // the content of the HTTP response follows the header:
            client.print("Starting scan now<br>");

            uint8_t numSsid = WiFi.scanNetworks(); // Scanning Networks
            client.printf("Number of available WiFi networks discovered: %d <br>", numSsid);
            for (int i = 0; i < numSsid; i++)
            {
              client.printf("%d: %s, Ch:%d (%ddBm) <br>", i + 1, WiFi.SSID(i).c_str(), WiFi.channel(i), WiFi.RSSI(i));
            }

            // The HTTP response ends with another blank line:
            client.println();
            // break out of the while loop:
            break;
          } else {    // if you got a newline, then clear currentLine:
            currentLine = "";
          }
        } else if (c != '\r') {  // if you got anything else but a carriage return character,
          currentLine += c;      // add it to the end of the currentLine
        }

        // Check to see if the client request was "GET /H" or "GET /L":
        if (currentLine.endsWith("GET /H")) {
          digitalWrite(LED_BUILTIN, HIGH);               // GET /H turns the LED on
        }
        if (currentLine.endsWith("GET /L")) {
          digitalWrite(LED_BUILTIN, LOW);                // GET /L turns the LED off
        }
      }
    }
    // close the connection:
    client.stop();
    Serial.println("Client Disconnected.");
  }
}
```

### Debug Messages:
No error message. The PC gets disconnected and does not receive the rest of the webpage (the scan results).

me-no-dev, 2020-11-19:
I have the feeling that this is caused by some new changes to the WiFi subsystem and scan parameters. Looks like briefly the radio was switched to other channels (so it can scan them) and computer detected that as disconnect. I will ask internally if someone has a clue how to get around this.

gonzabrusco, 2020-11-28:
@me-no-dev any news about this?

me-no-dev, 2020-11-28:
unfortunately nothing back yet. There are some updates that I will push probably tomorrow. Check back in RC3 :) I'll poke my colleagues again monday

gonzabrusco, 2020-12-02:
I'm sorry to inform that the problem persists in RC3. When the scanning starts my Windows 10 PC gets disconnected from the AP.

YouDONG-ESP, 2020-12-02:
Hi @gonzabrusco, thanks for reporting. Could you tell me, on the PC side, is there anything else you did besides connecting to wifi?

gonzabrusco, 2020-12-02:
No. Just connect and browse to 182.168.4.1.

gonzabrusco, 2020-12-09:
RC4. This keeps happening. Sorry. When a Windows 10 connect to the AP and the ESP32 starts the scan, the AP gets unstable and the Windows PC gets disconnected.

YouDONG-ESP, 2020-12-09:
@gonzabrusco We had found the root cause: while scanning, our devices will be in different channels, causing AP to not send beacon when not in the home channel. We are discussing solutions. A temporary workaround is to 
Reduce the time spent on each channel `WiFi.scanNetworks(false,false,false,100U);`。

gonzabrusco, 2021-01-15:
Sorry but this still keeps happening. Even with 100u scan time between channels. With latest master as of today (RC6).

RafigRzayev, 2021-01-19:
@YouDONG-ESP Could you please give approximation regarding when it will be fixed?  I also have the same problem.  Tried with 50, still got disconnection.

YouDONG-ESP, 2021-01-19:
@RafigRzayev This problem is pretty complicated to fix. We
 don't have a perfect solution, the best fix we found is let esp32 goback to home channel after finish scanning each channel, but it will also make scan finish slower. We are doing test to see if it will affect the use.

RafigRzayev, 2021-01-19:
@YouDONG-ESP Thank you for informing. 
@gonzabrusco Can you try WiFi.scanNetworks(false, false, true, 100) instead of WiFi.scanNetworks(false, false, false, 100). After switching to passive scanning my connection seems ok at the moment.

gonzabrusco, 2021-06-18:
I can confirm this is still happening with latest framework version 1.0.6

gonzabrusco, 2022-02-08:
I can confirm this issue still exists at 2.0.2.

PilnyTomas, 2022-02-28:
Hi, I just tried the sketch on ESP32-S2 (WROVER module)
using Arduino IDE 1.8.15
I tried it with the following systems and all of them worked:
Linux Fedora 34 64bit on ThinkPad E15
Android 9 on LG G6
iPhone 13
Windows 10 on HP EliteBook 855
@gonzabrusco do you still have those issues?

gonzabrusco, 2022-03-07:
Sorry for the late response. This still happen to me but ONLY with laptops. With cellphones and with a desktop PC with a WiFi pci-e adapter, works fine. But with two different Laptops, this issue exists. Try it a couple of times, repeatedly, from a laptop, and you **_should_** see it.

PilnyTomas, 2022-03-07:
Hi, I just tried it with WROOM-32D + laptop with Linux and it works ok for me. Tried to disconnect/connect and wait for a while. Cycled a few times. The connection was not lost.
How far is your ESP from the computer? Same table/room, or somewhere further? Maybe the signal is weak.
Do you have the ESP connected directly to the computer/power source (without USB hubs)? Maybe it is power related problem.
To be honest I don't know what could be causing it.

gonzabrusco, 2022-03-07:
Did you tried with Windows 10? 

This issue was reproduced by me an another collegue in remote locations with different hardware (different laptops, different devkits but same ESP32 version). Always connected to the USB of the laptop, very close to it.

PilnyTomas, 2022-03-21:
Hi, I tried it with Win 10 and Win 7 using ESP32 WROVER-B and WROOM-32D and everything was ok.
I'm sorry but I am unable to replicate your issue.

VojtechBartoska, 2022-03-23:
Closing this one as we cannot Replicate it, if needed you can reopen it. Thanks!

closed 2022-03-23 by VojtechBartoska

VojtechBartoska, 2022-03-24:
@gonzabrusco Good idea, it can be related to specific HW issue.

everslick, 2022-04-27:
Just FYI, to whomever it is concerned: The issue seems to be the same, when the configured SSID is unavailable or out of range. Connecting to the ESP32 via softAP is practically impossible in that case, which IMHO is a common failure MODE, that _SHOULD_ be handled by the WiFi stack, but isn't. The only workaround I know of is to call `WiFi.enableSTA(false);` if you detect that scenario and re-enable STA when the softAP gets disabled again.

Hichiii, 2022-07-05:
I have the same problem with a c3 mini 1.

As soon as WiFi.scanNetworks() is executed, the Windows laptop flies out of the AP.
It works with Android phones.

I have the WiFi.scanNetworks(false,false,false,100U); and
WiFi.scanNetworks(false,false,true,100U);

tries.

But it doesn't help. is there a solution now?

I want to run Tasmote on the ESP32 c3. there WiFi.scanNetworks(); used

lg
Hichi

roysG, 2023-01-07:
In our side, 
We are using in ESP32, and run scanWifi and SoftAP to allow other device detect the device.

Sometime the softAP not start without any possible to detect it!

Is there any solution for this problem? 

It looks that it hapenning when scanNetwork start before the softAP...

kapyaar, 2023-01-14:
I faced the same issue on Windows 11 with a standard ESP32 Dev Module. Original scanNetwork was taking around 6-8 seconds, and windows will drop the connection and jump back to regular wifi. 

For those who still face the issue, 
WiFi.scanNetworks(false,false,true,100U); helped in my case. Though it was still borderline with 100U. I reduced this to 50, and started showing better results. After playing around for a bit, I settled in at 75U. YMMV.

trullock, 2023-08-31:
I have this exact problem with my Android client disconnecting from the SoftAP when doing a WiFi scan

italocjs, 2024-02-26:
WiFi.scanNetworks(false,false,true,75U); did help.

I think the core issue may be related to windows 10/11 testing the connection, with android phones the connections is very stable, the issue only seems to happen on windows.  I tried to add an dns responder but did not see any improvements (or i didnt made it right, i'm new at wifi / web stuff)