espressif/arduino-esp32 #4555: BLE with Sleep mode

MihirKhara, 2020-11-20:
I wanted to know if it is possible to use BLE as a wake-up source for sleep modes (preferably deep sleep) in Arduino Core.

The ESP-IDF GitHub repo shows that they were going to add this feature in version ESP-IDF v3.1, although there is no official documentation for this on their website that indicates it was added.

I'm using -
Arduino IDE - 1.8.12
Latest ESP32 Arduino Core (v1.0.4)
Board - DOIT ESP32 Devkit v1

I will be hosting a BLE server on ESP32. I would like to add functionality that wakes up the device whenever there is an incoming connection request from a mobile app via BLE.
If a BLE connection is requested, the device should exit deep sleep mode and fetch sensor data, transmit it, and go back to sleep again.

Is there any way this can be implemented in Arduino IDE?
If not in Arduino Core, can it be implemented using ESP-IDF?

chegewara, 2020-11-20:
Lets think about it. Turn off radio and use radio, is it possible? I think its not.

MihirKhara, 2020-11-20:
That's the question I have asked. I didn't say anything like turning it off and then using it (which btw is common sense).
My question was, is it possible with Arduino Core, because it seems possible with ESP-IDF according to a few Github issue chats.

chegewara, 2020-11-20:
The whole idea of sleep mode is to turn off radio which is most power hungry peripheral, but maybe there is some other sleep mode im not aware of.

MihirKhara, 2020-11-20:
Okay, it seems it's not possible with the current version of Arduino core, will need to find some other workaround.
Thanks for the immediate reply.

closed 2020-11-20 by MihirKhara

chegewara, 2020-11-20:
There is option to use light sleep mode but requires external XTAL IIRC and is not turned on in arduino, only esp-idf.

chegewara, 2020-11-20:
https://github.com/espressif/esp-idf/issues/947#issuecomment-500312453

MihirKhara, 2020-11-20:
Yeah I did find this, seems like the option to go with right now.

chegewara, 2020-11-20:
One more thing. This library works pefectly fine with esp-idf, and if you need to use arduino libraries then arduino as component is the way to go.

MihirKhara, 2020-11-20:
Yes I am aware, just need to import Arduino libraries in Eclipse IDE with ESP-IDF right?

chegewara, 2020-11-20:
https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md

MihirKhara, 2020-11-20:
Thanks for posting.