espressif/arduino-esp32 #4561: Adafruit_MLX90614 and MAX30102 Not working

fervidautomation, 2020-11-22:
Make your question, not a Statement, inclusive. Include all pertinent information:

What you are trying to do  
I tried to use ESP32 wroom dev board and use Adafruit_MLX90614 and MAX30102 sensor on I2C communication. As you might know both works on I2C communication. but It did not work. Any piece of advice will be highly appreciated.

Describe your system (Hardware, computer, O/S, core version, environment)  
ESP32 Wroom dev MLX90614 and MAX30102
Describe what is failing  
MAX30102 provide the data but mlx does not. If I am using a separate-2 2 programs then It is working fine.Which is kind of indication that the wiring is correct.
  
/*
  #  simple SpO2 plotter for MH-ET LIVE MAX30102 breakout board and ESP32 devkit-C
  Using Sparkfun MAX3010X library
  https://github.com/sparkfun/SparkFun_MAX3010x_Sensor_Library

  ESP32_MAX30102_simple-SpO2_plotter.ino
  by coniferconifer Copyright 2020
  LICENSED under Apache License 2.0

  Version 1.0

  Shows SpO2 and the user's heart beat on Arduino's serial plotter.
  No display hardware is required.
  This program should not be used for medical purposes.
  I wrote this to learn how SpO2 can be measured and pay tributes for the inventors.

  Pulse oximetry was developed in 1972, by Takuo Aoyagi and Michio Kishi,
  bioengineers, at Nihon Kohden in Japan.
  https://ethw.org/Takuo_Aoyagi

  Since MH-ET LIVE MAX30102 breakout board seems outputting IR and RED swapped.
  red = particleSensor.getFIFOIR();
  ir = particleSensor.getFIFORed();
  is used in my code. If you have Sparkfun's MAX30105 breakout board , try to 
  use #define MAX30105

  ## Tips:
  SpO2 is calculated as R=((square root means or Red/Red average )/((square root means of IR)/IR average))
  SpO2 = -23.3 * (R - 0.4) + 100;
  // taken from a graph in https://ww1.microchip.com/downloads/jp/AppNotes/00001525B_JP.pdf
  // https://ww1.microchip.com/downloads/en/Appnotes/00001525B.pdf
  ## Instructions:
  0) Install Sparkfun's MAX3010X library
  1) Load code onto ESP32 with MH-ET LIVE MAX30102 board
  2) Put MAX30102 board in plastic bag and insulates from your finger.
     and attach sensor to your finger tip
  3) Run this program by pressing reset botton on ESP32
  4) Wait for 3 seconds and Open Arduino IDE Tools->'Serial Plotter'
     Make sure the drop down is set to 115200 baud
  5) Search the best position and presure for the sensor by watching
     the blips on Arduino's serial plotter
     I recommend to place LED under the backside of nail and wrap you
     finger and the sensor by rubber band softly.
  6) Checkout the SpO2 and blips by seeing serial Plotter
     100%,95%,90%,85% SpO2 lines are always drawn on the plotter

  ## Hardware Connections (Breakoutboard to ESP32 Arduino):
  -VIN = 3.3V
  -GND = GND
  -SDA = 21 (or SDA)
  -SCL = 22 (or SCL)
  -INT = Not connected

  this script also works on Arduino nao
  ## Hardware Connections (Breakoutboard to Arduino nano): experimental
  -VIN = 3.3V 
  -GND = GND
  -SDA = A4 (or SDA)
  -SCL = A5 (or SCL)
  -INT = Not connected  
[... 193 lines cut ...]
        Serial.print(","); Serial.print(90.0); //warning SpO2 line
        Serial.print(","); Serial.print(95.0); //safe SpO2 line
        Serial.print(","); Serial.println(100.0); //max SpO2 line
        //#endif

 
Serial.println();
delay(1000);

      }
    }
    if ((i % Num) == 0) {
      double R = (sqrt(sumredrms) / avered) / (sqrt(sumirrms) / aveir);
      // Serial.println(R);
      SpO2 = -23.3 * (R - 0.4) + 100; //http://ww1.microchip.com/downloads/jp/AppNotes/00001525B_JP.pdf
      ESpO2 = FSpO2 * ESpO2 + (1.0 - FSpO2) * SpO2;
      //  Serial.print(SpO2);Serial.print(",");Serial.println(ESpO2);
      sumredrms = 0.0; sumirrms = 0.0; i = 0;
      break;
    }
    particleSensor.nextSample(); //We're finished with this sample so move to next sample
    //Serial.println(SpO2);
  }
#endif
}

me-no-dev, 2020-11-22:
long post but no logs or good explanation of what "it didn't work" means.

Enable debug, collect logs and screenshots and post a good explanation of the problem.

SuGlider, 2021-09-14:
PR #5664 may solve this issue.
Please test it and let me know.
Thanks.

me-no-dev, 2021-10-01:
closed by https://github.com/espressif/arduino-esp32/pull/5683

closed 2021-10-02 by SuGlider