espressif/arduino-esp32 #4566: Bluetooth classic serial connection PIN provided but pairing is without PIN

Adminius, 2020-11-23:
### Hardware:
Board:							ESP32 WEMOS D1 MINI
Core Installation version:			1.0.4
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                    no
Upload Speed:					115200
Computer OS:                                        Windows 10

### Description:
After flashing: Smartphone sees "ESP32test" asking if I would like to connect and it connects without asking for a PIN (tried different PINs with different length)

Expectation: Smartphone asks for PIN and connects after user inputs a correct PIN code

### Sketch:  
Using example sketch:
https://github.com/espressif/arduino-esp32/blob/master/libraries/BluetoothSerial/examples/SerialToSerialBTM/SerialToSerialBTM.ino
but with activated SerialBT.setPin(pin); (before or after SerialBT.begin is doesn't matter) and changed pin:

```
char *pin = "555555"; 
...
SerialBT.begin("ESP32test", true); 
SerialBT.setPin(pin);
```

### Debug Messages:
```
[I][BluetoothSerial.cpp:510] _init_bt(): device name set
The device started in master mode, make sure remote BT device is on!
[I][BluetoothSerial.cpp:225] esp_spp_cb(): ESP_SPP_INIT_EVT
[I][BluetoothSerial.cpp:702] connect(): master : remoteName
[I][BluetoothSerial.cpp:379] esp_bt_gap_cb(): ESP_BT_GAP_DISC_STATE_CHANGED_EVT
[I][BluetoothSerial.cpp:327] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT
[I][BluetoothSerial.cpp:329] esp_bt_gap_cb(): Scanned device: cc:21:19:94:5b:be
[I][BluetoothSerial.cpp:336] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT : EIR : Galaxy S10e : 11
[I][BluetoothSerial.cpp:327] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT
[I][BluetoothSerial.cpp:329] esp_bt_gap_cb(): Scanned device: cc:21:19:94:5b:be
[I][BluetoothSerial.cpp:336] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT : EIR : Galaxy S10e : 11
[I][BluetoothSerial.cpp:327] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT
[I][BluetoothSerial.cpp:329] esp_bt_gap_cb(): Scanned device: cc:21:19:94:5b:be
[I][BluetoothSerial.cpp:336] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT : EIR : Galaxy S10e : 11
[I][BluetoothSerial.cpp:327] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT
[I][BluetoothSerial.cpp:379] esp_bt_gap_cb(): ESP_BT_GAP_DISC_STATE_CHANGED_EVT
[I][BluetoothSerial.cpp:379] esp_bt_gap_cb(): ESP_BT_GAP_DISC_STATE_CHANGED_EVT
[V][BluetoothSerial.cpp:392] esp_bt_gap_cb(): authentication success: Galaxy S10e
Connected Succesfully!
[I][BluetoothSerial.cpp:742] connect(): master : remoteName
[I][BluetoothSerial.cpp:379] esp_bt_gap_cb(): ESP_BT_GAP_DISC_STATE_CHANGED_EVT
[I][BluetoothSerial.cpp:327] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT
[I][BluetoothSerial.cpp:329] esp_bt_gap_cb(): Scanned device: cc:21:19:94:5b:be

```

Cjcr, 2020-11-25:
Confirmed. The pin doesn't work as it should.

ketan, 2020-12-01:
Can confirm. [This example](https://github.com/espressif/arduino-esp32/blob/38c4c0610846b7193e908b474e2c8db06ae981ba/libraries/BluetoothSerial/examples/SerialToSerialBTM/SerialToSerialBTM.ino) seems to have commented out the `setPin` method. It does not work as expected.

juniorradu, 2021-03-17:
Same problem here. Any solution?

hulktech, 2021-04-01:
Hello, a lot of people also me, we are trying to find an option to set static PIN on Bluetooth classic serial connection when pairing. A way to input custom PIN not the default 123456.An old way was created for 1.0.1 version adding a custom libbt.a file that was disabling ssp as far as i know .We want to be able to have this option on newer versions. I'll appreciate anyone who really want to help

ketan, 2021-04-01:
Same here. It is unclear what the change was in that particular file, that
allowed it to work. Can the original author, please mention the change they
had to make, if they still have it with them?

Thanks

Adminius, 2021-06-19:
*push*

Adminius, 2021-08-24:
Push...

muellerryan, 2021-09-10:
I have been looking for a way around this in order to get a PIN to work but have come up empty-handed for almost a year now.  I am looking forward to a solution to this issue, especially in light of Braktooth vulnerabilities.

chegewara, 2021-09-10:
This is the answer:
> Fixed pin is used for Legacy Pairing, and there is no fixed pin in Simple Secure Pairing
https://github.com/espressif/esp-idf/issues/1541#issuecomment-443575348

That means it requires to build bluetooth library with special config.

muellerryan, 2021-09-10:
> This is the answer:
> 
> > Fixed pin is used for Legacy Pairing, and there is no fixed pin in Simple Secure Pairing
> > [espressif/esp-idf#1541 (comment)](https://github.com/espressif/esp-idf/issues/1541#issuecomment-443575348)
> 
> That means it requires to build bluetooth library with special config.

How do I build a special config?  I was under the impression that wasn't possible with the arduino-esp.

chegewara, 2021-09-10:
Yes, it was not needed earlier, because it was bug in esp-idf v3.1 or 3.2. Since v3.3.x its been fixed.

VojtechBartoska, 2022-03-30:
This seems to me outdated and solved already. Closing, if needed you can reopen it. Thanks for understanding.

dageci, 2022-09-23:
Hey @VojtechBartoska you have closed this issue as it is completed, could you please point to the example where we can see how this works?
I have tried with:

```
  SerialBT.enableSSP();
  SerialBT.setPin("123456"); 
  SerialBT.begin("ESP32device");

``` 
and with this 

```
  SerialBT.setPin("123456"); 
  SerialBT.begin("ESP32device");

``` 
But the ESP32 is paired with arduino device in both cases without asking for PIN.
Could you show us how to do it the right way?

Thanks,
Davor

Alain-Godo, 2023-06-22:
Hi @dageci and @VojtechBartoska, I have the same problem, a PIN is provided but the pairing is made without asking for the PIN. What's wrong?

VojtechBartoska, 2023-06-23:
Hi,
@Alain-Godo & @dageci, take a look [here](https://github.com/espressif/arduino-esp32/pull/7727/files#diff-eae76586b0577454415d67a8a600072cdc6bff96d4f0c32cc01d73debff4264e)

Alain-Godo, 2023-06-23:
Thanks for the reply, the example shows how to use the PIN to connect the master to a slave, but my situation is the opposite, create the slave with a PIN. How can I do that?

Alain-Godo, 2023-06-23:
I tried with this example, the Bluetooth slave is created but without the PIN. When I take my phone and I do the pairing processes the PIN is not required. I don't why....
https://github.com/espressif/arduino-esp32/blob/master/libraries/BluetoothSerial/examples/SerialToSerialBT/SerialToSerialBT.ino

sinmim, 2023-07-01:
I want to use esp32 blt in a commercial device. how it's not possible yet to have a pin while pairing? I directly talk to the esspersive and they don't answer it either. can anybody a working example? I'm using platformIO

Alain-Godo, 2023-07-03:
I'm in the same situation...

luthfiabraham, 2023-07-09:
@VojtechBartoska, any update on this ? I can also confirmed have the same issue with @Alain-Godo that the pin is not working when the Master (phone) connected to Slave (esp32).

PilnyTomas, 2023-07-12:
I tried it on the [esp-idf-v5.1-libs](https://github.com/espressif/arduino-esp32/tree/esp-idf-v5.1-libs) branch

I confirm the issue - I can connect with a phone to a slave which is expecting a PIN, but the phone doesn't even pop up the prompt for PIN and ESP32 reports "authentication success". And when I send something (via BTSerial app) it is transmitted correctly both ways.

Also when using two ESPs (master and slave) with mismatching PINs they connect successfully and are able to communicate.

I can connect with phone to master without pin - however i cannot connect with serial app and communicate with the ESP. Note that phone is also master and connecting 2 masters should not work anyway.

Alain-Godo, 2023-07-12:
Thanks, @PilnyTomas !!! Seems like the PIN is never set actually...

luthfiabraham, 2023-07-12:
Thanks a lot !!! @PilnyTomas for reopening this thread. Below is some output from Serial Monitor with debug level set to "info". But @Alain-Godo ... sadly somehow from the debug output, the pin is being set, looks like somewhere in the code probably ignoring this request.

This code run in setup
```c
SerialBT.begin(DEVICE_NAME);
SerialBT.setPin(BLUETOOTH_PIN);
``` 

Below is the output
```
[  7237][I][BluetoothSerial.cpp:571] esp_bt_gap_cb(): ESP_BT_GAP_CONFIG_EIR_DATA_EVT: stat:0 num:4
[  7237][I][BluetoothSerial.cpp:683] _init_bt(): device name set
[  7240][I][BluetoothSerial.cpp:903] setPin(): pin: 765498
[  7237][I][BluetoothSerial.cpp:263] esp_spp_cb(): ESP_SPP_INIT_EVT
[  7251][I][BluetoothSerial.cpp:270] esp_spp_cb(): ESP_SPP_INIT_EVT: slave: start
[  7259][I][BluetoothSerial.cpp:146] btSetPin(): pin set
[  7264][I][BluetoothSerial.cpp:571] esp_bt_gap_cb(): ESP_BT_GAP_CONFIG_EIR_DATA_EVT: stat:0 num:4
[  7273][I][BluetoothSerial.cpp:397] esp_spp_cb(): ESP_SPP_START_EVT
``` 

And Yes, phone can pair without pin (unfortunately no debug output during pairing process) and bluetooth serial data transfer running smoothly, I hope it can help for further investigation.

PilnyTomas, 2023-07-27:
Hello everyone, it has been a while since dug into this issue, but to be honest, this rabbit hole goes very deep and I was unable to find any clear reason why the legacy pairing using fixed PIN is not working. I will ask around in the company, but I can't promise anything.
Meanwhile please consider using [Simple Secure Pairing](https://github.com/espressif/arduino-esp32/blob/master/libraries/BluetoothSerial/examples/SerialToSerialBT_SSP_pairing/SerialToSerialBT_SSP_pairing.ino)

luthfiabraham, 2023-07-27:
Hi @PilnyTomas , with the sample of [Simple Secure Pairing](https://github.com/espressif/arduino-esp32/blob/master/libraries/BluetoothSerial/examples/SerialToSerialBT_SSP_pairing/SerialToSerialBT_SSP_pairing.ino), If I keep on responding with this when there is a request, it should be good enough to protect anyone to paired it blindly right ?
```c
SerialBT.confirmReply(false);
``` 
Anyway, thank you for your effort, I hope the company might give us some answer or hint to solved it, in the mean time seems that SSP the only work around, I am a bit hesitate to use it because of the callback, last time my experience using callback is disrupting a bit during ongoing bt serial comm.

PilnyTomas, 2023-07-27:
SSP is superior to fixed PIN in security - it is simply newer and better.

Some people already pointed out that disabling sdkconfig `CONFIG_BT_SSP_ENABLED` should enable legacy pairing (fixed PIN) - this is not what the [documentation](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-bt-ssp-enabled) says...

_CONFIG_BT_SSP_ENABLED
    Secure Simple Pairing - This enables the Secure Simple Pairing. If disable this option, Bluedroid will only support Legacy Pairing_

On the other according to [this issue](https://github.com/espressif/esp-idf/issues/2774), it would seem that it's true - to use legacy pairing you need to turn off SSP. This however requires compiling the Arduino core, so doing this globally is not an option as it would create issues for others who are using SSP.

The solution might be using [Arduino as an IDF component](https://espressif-docs.readthedocs-hosted.com/projects/arduino-esp32/en/latest/esp-idf_component.html?highlight=component).

I will take look at this myself and if it works I will make some documentation and examples for this in the PR #8453

luthfiabraham, 2023-07-28:
Understood, it is impossible to always released two different version for Arduino core. Your example will greatly help us when there is a need for legacy pairing, thanks a lot.

PilnyTomas, 2023-08-09:
Hi, I'm finishing the [PR](https://github.com/espressif/arduino-esp32/pull/8453) related to this. I would like to ask all of you test it.
Please read the [README](https://github.com/espressif/arduino-esp32/blob/672395f7df326bc0208c4eaf69179a571563b8a8/libraries/BluetoothSerial/README.md) file as the legacy pairing is not so easy to use.

closed 2023-08-10 by PilnyTomas

Adminius, 2024-02-19:
I've tried now the PIN feature... and it still doen't work:
My Samsung S23 conneted to the ESP32 without any PIN.
I enabled #define USE_PIN in "SeriaToSerialBT" example:

> 15:27:39.315 -> The device with name "ESP32-BT-Slave" is started.
> 15:27:39.315 -> Now you can pair it with Bluetooth!
> 15:27:39.315 -> Using PIN
> 15:30:03.653 -> [     4][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
> 15:30:04.167 -> [   467][I][BluetoothSerial.cpp:571] esp_bt_gap_cb(): ESP_BT_GAP_CONFIG_EIR_DATA_EVT: stat:0 num:4
> 15:30:04.167 -> [   467][I][BluetoothSerial.cpp:683] _init_bt(): device name set
> 15:30:04.167 -> The device with name "ESP32-BT-Slave2" is started.
> 15:30:04.167 -> Now you can pair it with Bluetooth!
> 15:30:04.167 -> [   468][I][BluetoothSerial.cpp:263] esp_spp_cb(): ESP_SPP_INIT_EVT
> 15:30:04.167 -> [   481][I][BluetoothSerial.cpp:903] setPin(): pin: 4321
> 15:30:04.167 -> [   484][I][BluetoothSerial.cpp:270] esp_spp_cb(): ESP_SPP_INIT_EVT: slave: start
> 15:30:04.167 -> [   497][I][BluetoothSerial.cpp:146] btSetPin(): pin set
> 15:30:04.167 -> [   500][I][BluetoothSerial.cpp:571] esp_bt_gap_cb(): ESP_BT_GAP_CONFIG_EIR_DATA_EVT: stat:0 num:4
> 15:30:04.167 -> Using PIN
> 15:30:04.167 -> [   510][I][BluetoothSerial.cpp:397] esp_spp_cb(): ESP_SPP_START_EVT
> 15:30:22.301 -> [ 18650][I][BluetoothSerial.cpp:587] esp_bt_gap_cb(): ESP-BT_GAP_* unknown message: 16
> 15:30:24.293 -> [ 20642][V][BluetoothSerial.cpp:523] esp_bt_gap_cb(): authentication success: S23 von Eugen
> 

What I'm doing wrong? My project doen't have any display to show PIN to compare. And I don't want that somebody "just accepts" any shown pin. And I also doesn't have any physical buttons on the ESP32 to initiate any pairing access.
PIN is only one possibility...

I'm using ESP32 Dev Board with 2.0.14 IDE