espressif/arduino-esp32 #4572: ModuleNotFoundError: No module named 'serial'

hary66, 2020-11-25:
Hi.
I'm facing that error trying to use ESP32 for the first time with Arduino IDE :+1: 
`import serial
ModuleNotFoundError: No module named 'serial'
Plusieurs bibliothèque trouvées pour "WiFi.h"
Utilisé : /home/harold/.arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/WiFi
Non utilisé : /home/harold/arduino-1.8.13/libraries/WiFi
exit status 1
Erreur de compilation pour la carte ESP32 Wrover Module`

I previously programmed esp8266 with no problem. I'm running arduino IDE1.8.13 on Linux MINT20 freshly installed.
ESP32-CAM

To try to solve the problem I ran the following command that didn't solve the problem but created news ones instead as I can't program esp8266 any more. It won't compile !

Here are the the command I ran :+1: 
`ln -s python python3`
and
`sudo apt install python-is-python3`

Previously running the command, `$ python` would launches python2.7.x (from my memory)
It now launches python 3.8.2.

`import serial ` for either python2 or python3 would  through : 
`Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'serial'
`
Would anybody could help ?  Would there be a way at first to get to be able to program esp8266 back  by undoing what i 've done and didn't solve the problem ?

lbernstone, 2020-11-25:
You should just need to run `pip3 install pyserial` (maybe with sudo) to get the missing module installed.

devrim-oguz, 2020-11-25:
You need to install the pyserial module with pip or pip3. This is related with python, you can easily find help from google. I'd recommend you close this issue.

closed 2020-11-25 by hary66

lbernstone, 2020-11-25:
jinx.  You owe me a coke, @devrim-oguz

devrim-oguz, 2020-11-26:
Yeah, seriously. Just before I pressed submit your comment appeared. I pressed anyways 😂 @lbernstone