espressif/arduino-esp32 #4579: Spaces in path causes compilation error

jpasqua, 2020-11-25:
### Hardware:
Board:							ESP32 D1 Mini
Core Installation version:			1.0.5 rc2
IDE name:						Arduino IDE 1.8.13
Flash Frequency:					80Mhz
PSRAM enabled:                                    N/A
Upload Speed:					921600
Computer OS:                                       macOS 10.15.7

### Description:
I'm trying to compile a project in a directory that contains spaces in the path:
`/Users/joe/Dropbox/Dev/Arduino/libraries/TFT_eSPI-master/examples/320 x 240/TFT_Meters`
This is a standard example that comes with the TFT_eSPI library and works fine with other cores like the ESP8266. Removing all spaces from the path causes the error to go away.

### Sketch:  [Any sketch will fail]
```cpp

void setup() {}
void loop() {}
```

### Debug Messages:
```
bash -c "[ ! -f /Users/joe/Dropbox/Dev/Arduino/libraries/TFT_eSPI-master/examples/320 x 240/TFT_Meters/partitions.csv ] || cp -f /Users/joe/Dropbox/Dev/Arduino/libraries/TFT_eSPI-master/examples/320 x 240/TFT_Meters/partitions.csv /var/folders/97/nw96g6nn0k33z212yw4by4yr0000gn/T/arduino_build_953387/partitions.csv"
bash: line 0: [: too many arguments
usage: cp [-R [-H | -L | -P]] [-fi | -n] [-apvXc] source_file target_file
       cp [-R [-H | -L | -P]] [-fi | -n] [-apvXc] source_file ... target_directory
exit status 64
```

Gius-8, 2020-12-22:
Same problem on Mac OS Big Sur (11.1)

coelner, 2020-12-24:
https://github.com/espressif/arduino-esp32/issues/4640#issuecomment-748051578

closed 2021-07-11 by stale[bot]