espressif/arduino-esp32 #4581: pSRAM memory size showing 0

sanketa-54, 2020-11-26:
Make your question, not a Statement, inclusive. Include all pertinent information:

### Hardware:
Board:							ESP32 Wrover standalone kit
Core Installation version:			1.0.4
IDE name:						?Arduino IDE?
Flash Frequency:					?40Mhz?
PSRAM enabled:                                   Not applicable
Upload Speed:					?115200?
Computer OS:                                        ?Windows 10

### Description:
I'm using the ESP32 Wrover standalone chip and trying to detect the psram .
Same I tested on IDF , Its showing me psram memory size = 8388608

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <SPI.h>

#include <Arduino.h>
#include<esp32-hal.h>

void setup() {
  
  Serial.begin(115200);
  //delay(2000);
// pinMode(3,OUTPUT);
// delay(200);
//
//  digitalWrite(3,HIGH);
//  delay(500);

  
  
 // Serial.setDebugOutput(true);
  
}

void loop() {
  Serial.print("Total heap:");
  Serial.println(ESP.getHeapSize());
  Serial.print("Free heap: ");
  Serial.println(ESP.getFreeHeap());
//  Serial.print("TOtal psram size : ");
//  Serial.println(esp_spiram_get_size());
  Serial.print("PSRAM INIT : ");
  Serial.println(psramInit());
   Serial.print("PSRAM Found : ");
  Serial.println(psramFound());
if(psramFound()){
    
//  Serial.print("Total PSRAM using caps:");
//  Serial.println(heap_caps_get_free_size(MALLOC_CAP_SPIRAM));
  Serial.print("Total PSRAM:");
  Serial.println(ESP.getPsramSize());
  Serial.print("Free PSRAM: ");
  Serial.println(ESP.getFreePsram());
  
  }
  delay(2000);
  }
```

### Debug Messages:
```
4:41:22.873 -> [D][esp32-hal-psram.c:47] psramInit(): PSRAM enabled
14:41:22.920 -> Total heap:388228
14:41:22.920 -> Free heap: 360968
14:41:22.920 -> PSRAM INIT : 1
14:41:22.920 -> PSRAM Found : 1
14:41:22.920 -> Total PSRAM:0
14:41:22.920 -> Free PSRAM: 0

```

me-no-dev, 2020-11-26:
did you enable PSRAM in the board menu? You do not need to call `psramInit` that is done at boot if you have selected the proper options

sanketa-54, 2020-11-26:
ESP wrover dont have option for psram enable.
And its automatically detecting and enabling also.
Can u please mention options ..If I have left something

sanketa-54, 2020-11-30:
I'm using WROVER-B standalone chip

sanketa-54, 2020-12-03:
Need to add the following to use on Arduino IDE :
extern "C" {
#include <esp_spiram.h>
#include <esp_himem.h>
}

closed 2020-12-03 by sanketa-54

badre2911, 2022-11-18:
not work with 8MB psram, it show 4MB

lbernstone, 2022-11-18:
Please don't hijack closed issues.
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/himem.html#the-himem-allocation-api