espressif/arduino-esp32 #4585: Installation instructions Debian, Ubuntu

reppr, 2020-11-26:
The instructions explain how to prepare and git clone arduino-esp32 on a Debian system, but do not explain
how to tell  Arduino IDE to *use* this code and to show esp32 boards? This info is missing.

The method with adding board URLs in Preferences works well, but does not clone the git repo, just the code.

lbernstone, 2020-11-26:
1) Follow the [issue template](https://github.com/espressif/arduino-esp32/blob/master/docs/ISSUE_TEMPLATE.md). It is right there when you open a new issue.  At the very least, read it.
2) Install [arduino 1.8.13](https://www.arduino.cc/en/software) to ~/arduino-1.8.13.
3) Follow the [install instructions](https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/debian_ubuntu.md.  Do not change folders names.  Watch for errors.  Post any errors you see.
4) In arduino, check your sketchbook path in File->Preferences.
5) You should see `ESP32 Arduino (in sketchbook)` in your Tools->Boards menu flyout.

chegewara, 2020-11-26:
https://esp32.com/viewtopic.php?f=19&t=18302#p68197

reppr, 2020-11-28:
@lbernstone thank you for your informations

1. well, I *did* read it, but had the impression that most of it is not relevant here for installation instructions. Sorry if this decicion was a stupid one.

2. I used another computer to try it again. This time I used an Ubuntu installation and started by uninstalling/deleting all Arduino related stuff.

3. No errors (like last time).

4. *This* was the missing information: I had configured my preexisting sketchbook path in File->Preferences. This did not work when installing esp32-arduino below ~/Arduino/ following the installation instructions. The sketchbook path should point to ~/Arduino/
and not be changed by the user.

May I suggest to make that clear in the installation instructions?

5. Now everything works fine.

closed 2021-02-13 by stale[bot]