espressif/arduino-esp32 #4587: Missing NimBLE library in SDK

jylestrat, 2020-11-27:
Hi,
I'm using latest arduino-esp32 framework version in Visual Studo Code / PlatformIO for my Heltec Lora32 v2.1 module since yesterday _(2020/11/26)_
Declared like this in platformio.ini file :
> platform_packages =
>     platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git

Because of the use of new methods (which appears with the 1.0.4 version arduino-esp32 framework) in the demo Heltec sketches such as : ESP.getChipModel(), ESP.getChipCores() 
Methods which are not present in the default arduino Framework version automatically downloaded by PlatformIO when following parameters are used in platformio.ini : 
> platform = espressif32
> framework = arduino

That said,
There is an error when I'm building my own arduino sketch in PlatformIO :
**NimBLE library is missing.** 

NimBLE library is declared in following files : 
https://github.com/espressif/arduino-esp32/blob/master/platform.txt
**compiler.cpreprocessor.flags= ... "-I{compiler.sdk.path}/include/nimble"**
and
https://github.com/espressif/arduino-esp32/blob/master/tools/platformio-build.py
**join(FRAMEWORK_DIR, "tools", "sdk", "include", "nimble**")

But NimBLE library is not présent in following directories : 
arduino-esp32/tools/sdk/include and /tools/sdk/lib

Furthermore, but it is not a problem for me, the following file is missing in the lattest version of the framework : 
arduino-esp32/cores/esp32/core_version.h

lbernstone, 2020-11-27:
NimBLE is available as a 3rd party package at https://github.com/h2zero/NimBLE-Arduino

closed 2021-02-13 by stale[bot]

hansonap, 2021-03-04:
Why is it a 3rd party package? Why is it not included in  espressif/arduino-esp32? Or at least included in the documentation for the installation of PlatformIO

lbernstone, 2021-03-04:
It is not included in this repo because it is maintained by h2zero (aka a 3rd party).  How would that be part of the installation for platformIO?

ashi2967, 2021-03-04:
unsubscribe

Windows 10 版のメ`ル<https://go.microsoft.com/fwlink/?LinkId=550986>から送信

差出人: Alex Hanson<mailto:notifications@github.com>
送信日r: 2021年3月4日 10:03
宛先: espressif/arduino-esp32<mailto:arduino-esp32@noreply.github.com>
CC: Subscribed<mailto:subscribed@noreply.github.com>
件名: Re: [espressif/arduino-esp32] Missing NimBLE library in SDK (#4587)

Why is it a 3rd party package? Why is it not included in espressif/arduino-esp32? Or at least included in the documentation for the installation of PlatformIO

hansonap, 2021-03-04:
NimBLE is part of the esp-idf

[image]

arduino-esp32 is published by espressif

The installation instructions provided by espressif/arduino-esp32 for platformio results in an exception when trying to compile a simple blink example

Something broke down in this process. What was it and how do we keep others from encountering the same exception? - That's my logic at least.

What are your thoughts on what can be improved?

hansonap, 2021-03-04:
> 
> 
> It is not included in this repo because it is maintained by h2zero (aka a 3rd party). How would that be part of the installation for platformIO?

I see our wires were crossed when I said "Or at least included in the documentation for the installation of PlatformIO" I meant to say the [arduino-esp32 platformio installation instructions](https://github.com/espressif/arduino-esp32/blob/master/docs/platformio.md)

lbernstone, 2021-03-04:
Umm, start by not hijacking closed threads.  I don't have the patience to deal with your bait.
Espressif has just hired a number of people that will help out on documentation.

Miq1, 2021-03-04:
Steal or not - I got the same issue here when updating to 3.1.0 in PlatformIO. Not only NimBLE seems missing out of a sudden, but one ``Adafruit_STMPE610_ID377`` is flagged missing as well. 
I do not know if it is arduino-esp32 or the PlatformIO guys losing sync here, but it is annoying anyway.

hansonap, 2021-03-05:
> 
> 
> Umm, start by not hijacking closed threads. I don't have the patience to deal with your bait.
> Espressif has just hired a number of people that will help out on documentation.

Wow. I meant no disrespect to the work done so far - my only intention was to help identify an item that could be improved and hopefully save others from dealing with the same issue. I would be glad to create a pull request which resolved the issue but am still not sure of the root cause / proper fix (arduino-esp32 packaging issue, arduino-esp32 documentation issue, or platformio framework packaging issue) which is why I originally posted on this issue. 

On the topic of "hijacking" - given that there are no contribution instructions for the arduino-esp32 repository I must default to my experiences - an issue shouldn't be closed unless is has been resolved. This issue seems like the most appropriate place to continue discussions on this exceptions. But if it fails to reopen I'll open a duplicate issue.

I hope we can improve the arduino-esp32 platformio tool chain.

ElPercha, 2021-03-21:
Same issue here, do we know what to do?
I get just warnings (but I hate them) 
This started after Espressif platform update on W10 + Visual Studio + PlatformIO + Arduino with ESP32

[image]

Miq1, 2021-03-21:
If you already got the most recent platform update, you need to select each project that has the error on it (via the chooser in the bottom line of VSCode:
[image]
and do a "Rebuild Intellisense" there - Ctrl+Shift+R, then select it:
[image]

ElPercha, 2021-03-21:
@Miq1 Thank you.
Your recommendation didn't fix the issue.
I used to fix issues like this by deleting the ".vscode" folder. Probably not the most professional way but it always worked.
This seems to be a different issue

ElPercha, 2021-03-21:
My bad, disregard the previous comments. 
I updated the Espressif platform to 3.1.1 as you recommended and the issue is fixed.

Thank you again.

hansonap, 2021-03-22:
A work around is to delete the two lines referencing nimble. You won't be able to use the nimble version of BLE, but at least it will compile!

ElPercha, 2021-03-23:
It compiles because it is just a warning.
The lines can be deleted but they are included automatically in the
compilation process.
So, the 2 warnings remain the same